espressif/esp-idf #4232: Release and Debug configuration for ESP32

marcinwilkabl, 2019-10-23:
Hello,
for my project I would like to create DEBUG and RELEASE configuration. It is quite easy to do with menuconfig - I just need to change "optimization level" and necessary changes are applied in sdkconfig.

But I do not want to use menuconfig everytime I need to change this configuration. So I created two different sdkconfigs: sdkconfig_debug and sdkconfig_release.

I am using CMake and I would like to be able to choose between sdkconfig_debug and sdkconfig_release. Is it possible?

I read here: viewtopic.php?t=5806
that someone was using SDKCONFIG for Make.

I tried that with CMake but it is not working for me :(

Thank you for any ideas.
Best,
Marcin

PerMalmberg, 2019-10-23:
You're looking for [idf_build_process](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system.html). You probably have to set up a [custom CMake project ](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system.html#using-esp-idf-in-custom-cmake-projects)to get it working.

renzbagaporo, 2020-04-28:
@marcinwilkabl Apologies for the late reply. You may have come across this already,  but it is possible to override the `sdkconfig.defaults` file for standard projects using the `SDKCONFIG_DEFAULTS` variable, like so:

```
# The following lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)

set(SDKCONFIG_DEFAULTS sdkconfig_debug) # or sdkconfig_release
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(hello-world)
```

renzbagaporo, 2020-09-24:
Hi @marcinwilkabl, any update if this is an ok solution?

closed 2021-11-03 by Alvin1Zhang