espressif/esp-idf #4233: MQTTs wrong cipher suites

Noxet, 2019-10-23:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32D
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/3762565/sdkconfig.txt)

]
- IDF version (run ``git describe --tags`` to find it): 
    v4.1-dev-474-g2e6398aff
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
   xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description

When setting up an MQTT connection with TLS (PKI or PSK), ALL supported cipher suites are being transmitted in the "client hello" message.
The problem is that when I set up a PSK connection, the server side (mosquitto) selects the first fitting cipher suite, which happens to be `ECDHE-ECDSA-AES256-GCM-SHA384`. This does not work because the client is only set up to use PSK, and the connection fails.
When running the CoAP examples, this problem does not occur

### Expected Behavior

The client should only send relevant supported cipher suites, and not all.

### Actual Behavior

The client sends all cipher suites it supports, causing the connection to fail depending on the setup.

### Steps to repropduce
Change the example MQTT code to set up a PSK connection to a server (mosquitto).

### Code to reproduce this issue

```cpp

#include "esp_log.h"

struct psk_key_hint psk_key = {
	.key = (uint8_t *) "password",
	.key_size = 8,
	.hint = "esp32"
};

void app_main()
{
    	const esp_mqtt_client_config_t mqtt_cfg = {
		.uri = CONFIG_BROKER_URI,
		.event_handle = mqtt_event_handler,
		.psk_hint_key = &psk_key,
	};

        esp_mqtt_client_handle_t client = esp_mqtt_client_init(&mqtt_cfg);
	esp_mqtt_client_start(client);
}

```

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

Alvin1Zhang, 2019-10-25:
@Noxet Thanks for reporting, we will look into it. Thanks.

david-cermak, 2019-10-25:
Could not the reason for choosing the `ECDHE-ECDSA-AES256-GCM-SHA384` be your server having certificate based TLS enabled?

Just checked with my local mosquitto:
```
# See also the mosquitto-tls man page and the "Certificate based SSL/TLS
# support" section. Only one of certificate or PSK encryption support can be
# enabled for any listener.
```
Even if I enforced this ciphersuite PSK-AES256-GCM-SHA384 on both client and server side, handshake fails with `routines:tls_post_process_client_hello:no shared cipher` if certificate authentication enabled in mosquitto.

Could you please check if that is not your case and if yes disable certificate based TLS and enable PSK only? Which version of mosquito you use?

Noxet, 2019-10-30:
@david-cermak Yes certificate based TLS is enabled. However, I am using both certificates and PSK so I cannot disable certificates. On the other hand, if PSK is configured on the client side, it should not send (and claim support for) certificate based suites anyway.

david-cermak, 2019-11-01:
@Noxet You can test this setup on your side too, just disable everything (which could be disabled) in `ComponentConfig -> mbedTLS -> TLS Key Exchange Methods`. Now the advertised cipher suites should include only PSK based ones. I assume that the connection would still not work with the same error message `tls_post_process_client_hello:no shared cipher`, since certificate based authentication takes precedence in mosquitto and could not be used at the same time as PSK (as stated previously here as well as on mosquitto man pages)

I'm sorry but runtime update of cipher suite set based on connection method is not supported in `esp-tls`. You could however statically configure that in sdkconfig.

Noxet, 2019-11-08:
Got it, that did the trick. Thank you for the support

closed 2019-11-08 by Noxet