espressif/esp-idf #4234: region `iram0_0_seg' overflowed - Check size-components is impossible

filzek, 2019-10-23:
v4.1-dev-474-g2e6398aff
Esp-WroverB 8mb 8mb
idf.py

Problem is that: 

idf.py size-components
section `.iram0.text' will not fit in region `iram0_0_seg'
IRAM0 segment data does not fit.
region `iram0_0_seg' overflowed by 13360 bytes
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

This command shall run perfect and list all data and report the full table assets working or not working, and show the status fail or pass in the end.

filzek, 2019-10-23:
Suggestion is to have an option to make it runs forced and produce the sizes for each part in a report, so, developer can traces and try to find what is using iram/other memory

closed 2019-10-31 by filzek

filzek, 2019-11-18:
Again, problem happen after update the SDK, so, IRAM0 is the major problem to develop a firmware today

ESP-Marius, 2023-04-11:
On latest master we've added a kconfig option for using parts of SRAM1 as IRAM in https://github.com/espressif/esp-idf/commit/5cbd311ecf0b381a0711ef8a96980a67b30a135f which will help with this issue.

We've also focused on reducing IRAM usage/making placement optional for several components. For an full overview of all such option see [Optimizing-iram-usage](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/ram-usage.html#optimizing-iram-usage). All new options for saving IRAM we add will be listed in this chapter.