espressif/esp-idf #4236: spiffs_create_partition_image fails on Clean Project.

Nulifier, 2019-10-23:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROOM-32U]
- IDF version (run ``git describe --tags`` to find it): 
    v4.0-beta1-134-gef17d8bba
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     [Windows]
- Power Supply:         [USB]

## Problem Description

When I try to run `idf menuconfig` after a clean build and with no pre-existing sdkconfig file, the command does not complete.

### Expected Behavior

The command configures the project and brings up the menu.

### Actual Behavior

Cmake throws an error during the configure and the configure does not complete.
```
CMake Error at C:/node/libs/esp-idf/components/esptool_py/project_include.cmake:171 (file):
  file RELATIVE_PATH must be passed a full path to the file: FLASH_IN_PROJECT
Call Stack (most recent call first):
  C:/node/libs/esp-idf/components/spiffs/project_include.cmake:45 (esptool_py_flash_project_args)
  main/CMakeLists.txt:11 (spiffs_create_partition_image)
```

### Steps to repropduce
I created a minimal reproduction [here](https://github.com/Nulifier/esp-test/tree/spiffs-missing-partition)
1. `git clone --single-branch --branch spiffs-missing-partition https://github.com/Nulifier/esp-test.git`
2. `cd esp-test`
3. `idf.py menuconfig`

### Probable Cause
I did some poking around in `components/spiffs/project_include.cmake` and it looks like what is failing is it isn't getting an offset from `parttool.py`. I think what the problem is that the parttool.py cannot determine the offset of the storage partition as I can't run menuconfig to set the custom partitions.

I have tested that this problem is solvable by manually setting the custom partitions in sdkconfig and that is a workaround but there should be at least a better error message in the configuration when it cannot find the partition as this would also happen if the partition name was misspelled.

igrr, 2019-10-23:
In the case when a custom partition table is used, one would normally set it in sdkconfig.defaults file. This way the option will be applied when doing a clean build (without an sdkconfig file).

Nulifier, 2019-10-23:
That's good information to have and it resolves my initial error. This still does happen if you have the partition name misspelled. I think it would be beneficial to have a proper error message returned from parttool.py when the partition is missing as the current message would have you thinking that the path for the files you want to add to the partition is wrong.

I did some looking into where this error could be added and I noticed that when a partition cannot be found in parttool.py, the function just returns without throwing an error. See this [line](https://github.com/espressif/esp-idf/blob/2e6398affaeeac2f7ce40457a881f2dda57ad11f/components/partition_table/parttool.py#L194).

Would the solution be to raise a KeyError on this line instead of returning? I could create a pull request if thats a good solution.

closed 2020-01-14 by espressif-bot