espressif/esp-idf #4238: Use after free in serial modbus master library. (mbc_serial_master)

sabbay, 2019-10-24:
I came across this when I wanted to restart modbus interface, because after some time it starts to malfunction and it was supposed to be a quick fix.
While trying to restart modbus master as shown below with function **modbus_restart**
we get heap corruption in **mbc_serial_master_create** because in **mbc_serial_master_destroy** mbm_interface_ptr isn`t being set to NULL after free. 

Adding mbm_interface_ptr = NULL; to the end of mbc_serial_master_destroy seems to fix this bug.

```c
esp_err_t modbus_init(void){
    mb_communication_info_t comm = {
            .port = MB_PORTNUM,
            .mode = MB_MODE_RTU,
            .baudrate = MB_BAUDRATE,
            .parity = MB_PARITY
    };
    void* master_handler = NULL;

    esp_err_t err = mbc_master_init(MB_PORT_SERIAL_MASTER, &master_handler);
    SENSE_MB_CHECK((master_handler != NULL), ESP_ERR_INVALID_STATE,
                                "mb controller initialization fail.");
    SENSE_MB_CHECK((err == ESP_OK), ESP_ERR_INVALID_STATE,
                            "mb controller initialization fail, returns(0x%x).",
                            (uint32_t)err);
    err = mbc_master_setup((void*)&comm);
    SENSE_MB_CHECK((err == ESP_OK), ESP_ERR_INVALID_STATE,
                            "mb controller setup fail, returns(0x%x).",
                            (uint32_t)err);
    err = mbc_master_start();
    SENSE_MB_CHECK((err == ESP_OK), ESP_ERR_INVALID_STATE,
                            "mb controller start fail, returns(0x%x).",
                            (uint32_t)err);
    // Set UART pin numbers
    err = uart_set_pin(MB_PORTNUM, CONFIG_MB_UART_TXD, CONFIG_MB_UART_RXD,
                                    CONFIG_MB_UART_RTS, UART_PIN_NO_CHANGE);
    SENSE_MB_CHECK((err == ESP_OK), ESP_ERR_INVALID_STATE,
            "mb serial set pin failure, uart_set_pin() returned (0x%x).", (uint32_t)err); 
    // Set driver mode to Half Duplex
    err = uart_set_mode(MB_PORTNUM, UART_MODE_RS485_HALF_DUPLEX);
    SENSE_MB_CHECK((err == ESP_OK), ESP_ERR_INVALID_STATE,
            "mb serial set mode failure, uart_set_mode() returned (0x%x).", (uint32_t)err);
    return err;
}

esp_err_t modbus_restart(void){
  SENSE_MB_CHECK((mbc_master_destroy() == ESP_OK), ESP_ERR_INVALID_STATE, "Failed to destroy mb handler structure.");
    return modbus_init();
}
```
```c
esp_err_t mbc_serial_master_create(mb_port_type_t port_type, void** handler)
{
    MB_MASTER_CHECK((port_type == MB_PORT_SERIAL_MASTER), 
                        ESP_ERR_INVALID_STATE, "mb incorrect port selected = %u.", 
                        (uint32_t)port_type);
    // Allocate space for master interface structure
    if (mbm_interface_ptr == NULL) {
        mbm_interface_ptr = malloc(sizeof(mb_master_interface_t));
    }
    MB_MASTER_ASSERT(mbm_interface_ptr != NULL);

```
```c
// Modbus controller destroy function
static esp_err_t mbc_serial_master_destroy(void)
{
    MB_MASTER_CHECK((mbm_interface_ptr != NULL),
                    ESP_ERR_INVALID_STATE,
                    "Master interface uninitialized.");
    mb_master_options_t* mbm_opts = &mbm_interface_ptr->opts;
    eMBErrorCode mb_error = MB_ENOERR;
    // Stop polling by clearing correspondent bit in the event group
    EventBits_t flag = xEventGroupClearBits(mbm_opts->mbm_event_group,
                                    (EventBits_t)MB_EVENT_STACK_STARTED);
    MB_MASTER_CHECK((flag & MB_EVENT_STACK_STARTED),
                ESP_ERR_INVALID_STATE, "mb stack stop event failure.");
    // Desable and then destroy the Modbus stack
    mb_error = eMBMasterDisable();
    MB_MASTER_CHECK((mb_error == MB_ENOERR), ESP_ERR_INVALID_STATE, "mb stack disable failure.");
    (void)vTaskDelete(mbm_opts->mbm_task_handle);
    (void)vEventGroupDelete(mbm_opts->mbm_event_group);
    mb_error = eMBMasterClose();
    MB_MASTER_CHECK((mb_error == MB_ENOERR), ESP_ERR_INVALID_STATE,
            "mb stack close failure returned (0x%x).", (uint32_t)mb_error);
    free(mbm_interface_ptr); // free the memory allocated for options
    return ESP_OK;
}
```

```
CORRUPT HEAP: multi_heap.c:308 detected at 0x3ffb98dc
abort() was called at PC 0x4008deff on core 0
0x4008deff: multi_heap_assert at /home/sabbay/esp32/esp-idf/components/heap/multi_heap_platform.h:54
 (inlined by) split_if_necessary at /home/sabbay/esp32/esp-idf/components/heap/multi_heap.c:307

ELF file SHA256: a2205ba82eacac51ac7bd6d7657b3d94022a5fac509e10ee15327f0491802d77

Backtrace: 0x4008bdad:0x3ffdd780 0x4008c135:0x3ffdd7a0 0x4008deff:0x3ffdd7c0 0x4008e22d:0x3ffdd7e0 0x40082465:0x3ffdd800 0x40082495:0x3ffdd820 0x40091745:0x3ffdd840 0x4009139e:0x3ffdd860 0x400d9543:0x3ffdd880 0x400d8198:0x3ffdd8b0 0x400d75d0:0x3ffdd8f0 0x400d770c:0x3ffdd940 0x400d7762:0x3ffdd970 0x400d6c46:0x3ffdd990 0x4008e645:0x3ffdd9b0
0x4008bdad: invoke_abort at /home/sabbay/esp32/esp-idf/components/esp32/panic.c:157

0x4008c135: abort at /home/sabbay/esp32/esp-idf/components/esp32/panic.c:174

0x4008deff: multi_heap_assert at /home/sabbay/esp32/esp-idf/components/heap/multi_heap_platform.h:54
 (inlined by) split_if_necessary at /home/sabbay/esp32/esp-idf/components/heap/multi_heap.c:307

0x4008e22d: multi_heap_malloc_impl at /home/sabbay/esp32/esp-idf/components/heap/multi_heap.c:455

0x40082465: heap_caps_malloc at /home/sabbay/esp32/esp-idf/components/heap/heap_caps.c:111

0x40082495: heap_caps_malloc_default at /home/sabbay/esp32/esp-idf/components/heap/heap_caps.c:140

0x40091745: malloc at /home/sabbay/esp32/esp-idf/components/newlib/heap.c:32

0x4009139e: xEventGroupCreate at /home/sabbay/esp32/esp-idf/components/freertos/event_groups.c:192

0x400d9543: mbc_serial_master_create at /home/sabbay/esp32/pppos_client/build/../components/freemodbus/serial_master/modbus_controller/mbc_serial_master.c:728

0x400d8198: mbc_master_init at /home/sabbay/esp32/pppos_client/build/../components/freemodbus/common/esp_modbus_master.c:36

0x400d75d0: modbus_init at /home/sabbay/esp32/pppos_client/build/../main/modbus_controller_common.c:100

0x400d770c: modbus_restart at /home/sabbay/esp32/pppos_client/build/../main/modbus_controller_common.c:128 (discriminator 2)

0x400d7762: modbus_controller_loop at /home/sabbay/esp32/pppos_client/build/../main/modbus_controller_common.c:30 (discriminator 9)

0x400d6c46: modbus_read_control at /home/sabbay/esp32/pppos_client/build/../main/pppos_client_main.c:164

0x4008e645: vPortTaskWrapper at /home/sabbay/esp32/esp-idf/components/freertos/port.c:143

```

Environment 
```
esp-idf: v4.1-dev-437-gd2ad0f077
xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
Linux
esp-wroom-32 devkitv1
```

marianoronchi, 2019-10-25:
I got the same error, in fact if you get the modbus master example and try to destroy and reinit the Modbus stack you'll get the error...

sabbay, 2019-10-25:
Thank You @alisitsyn for your fast response.
I have issue with MB_PORT_TIMER_ISR_FLAG not being declared.
Did you miss it? i cannot find it in source code

```bash
➜  esp-idf git:(master) ✗ grep -rn "MB_PORT_TIMER_ISR_FLAG" .
➜  esp-idf git:(master) ✗ 
```

sabbay, 2019-10-25:
@alisitsyn 

Is there a branch with those changes? If not I don`t mind waiting a bit for merge to master or other branch (unless it takes months :D). Also I don`t want to bother you to fix my code, it`s not a critical thing.
Thank you for taking care of those bugs for us! Really appreciate that!

alisitsyn, 2019-10-25:
@sabbay,
It is better to wait official merged fixes should not take a month. Thank you.

marianoronchi, 2019-10-25:
Hi, Thanks for your prompt answer, I tested the propossed code and seems to be working as expected... At least after a ModBus re-initialization still getting data, now I'll wait for the 0xFFFF errror and test if its solve the issue by now... When I have some time, I need to check why after a while of working ok the port appears to die.

@sabbay, if you want mi mbc_serial_master.c, mbc_serial_slave.c,porttimer_m.c, porttimer.c file just let my know (Mine are based on 4.0 branch)
Thanks.

Alvin1Zhang, 2020-05-27:
@sabbay Thanks for reporting and sorry for late reply, the fix has been available at https://github.com/espressif/esp-idf/commit/e24dda5dfbb61b172aea1aa1b80248c9cc23787c, feel free to reopen if the issue still happens. Thanks.

closed 2020-05-27 by Alvin1Zhang