espressif/esp-idf #4239: idf monitor ignores baudrate

will-emmerson, 2019-10-24:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): v3.2.2
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 5.2.0
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description
I am running the following command, then flashing the board with ctrl-t ctrl-a shortcut but it's not picking up the baudrate:
`idf.py monitor -p /dev/ttyUSB$0 -b 2000000`

I don't mean the console baudrate, I mean the flashing baudrate.

### Expected Behavior
It should flash using the 2000000 baudrate I've specified

### Actual Behavior
It flashes using 460800 instead

Same goes for flashing:
`idf.py flash monitor -p /dev/ttyUSB0 -b 2000000`

kumekay, 2019-11-01:
Hello @will-emmerson, thank you for the issue,
as a temporary solution, please put all your arguments before the list of subcommands,
i.e. `idf.py -p /dev/ttyUSB0 -b 2000000 flash monitor`  or `idf.py -p /dev/ttyUSB0 -b 2000000 monitor`

will-emmerson, 2019-11-01:
Thanks, it worked.

closed 2019-11-01 by will-emmerson