espressif/esp-idf #4241: esp_http_client functions causes ESP32 crashes

ccasenave-corexpert, 2019-10-24:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version : v3.2
- Build System:         [CMake]
- Compiler version : 1.22.0-80-g6c4433a
- Operating System:     [macOS]
- Power Supply:         [USB|Battery]

## Problem Description

In a download process I call some esp_http_client functions :

esp_http_client_init
esp_http_client_open
esp_http_client_fetch_headers
esp_http_client_read

When connected to a WIFI AP that is not so reliable, these functions makes ESP32 crash and I can't catch the error as the watchdog restart the ESP.

I'm ok with http functions can fail as connection is bad, but I would be able to catch and process the errors.
How can I catch these errors properly ?

### Expected Behavior

Http functions should return with error

### Actual Behavior

Http functions prints errors but crash the system

## Debug Logs

For exemple :

E (281379) esp-tls: read error :-76:
E (281379) TRANS_SSL: esp_tls_conn_read error, errno=No more processes

E (1495129) esp-tls: read error :-80:
E (1495139) TRANS_SSL: esp_tls_conn_read error, errno=Connection reset by peer

E (395599) esp-tls: Failed to open new connection
E (395599) TRANS_SSL: Failed to open a new connection

E (634909) esp-tls: Failed to connnect to host (errno 113)
E (634909) esp-tls: Failed to open new connection
E (634909) TRANS_SSL: Failed to open a new connection

G (875489) esp-tls: couldn’t get hostname for :s1-betty.ams3.digitaloceanspaces.com:
E (875489) esp-tls: Failed to open new connection
E (875489) TRANS_SSL: Failed to open a new connection

E (112078) esp-tls: mbedtls_ssl_setup returned -0x7f00
E (112088) esp-tls: Failed to open new connection
E (112088) TRANS_SSL: Failed to open a new connection

mahavirj, 2019-10-25:
@ccasenave-corexpert Can you please attach sample test code and/or crash dump (when ran through `make monitor` or `idf.py monitor`? Is it one of the default example from IDF or you have custom application?

lrodenasSistel, 2019-10-25:
I am also seeing this kind of errors at times, aprox once every 24h, but sometimes it repeats very often.
I was not getting watchdog resets, but sometimes it would break/corrupt everything and wifi was unusable afterwards.
I solved the breaking/corruption freeing some internal memory, moving the wifi allocation to PSRAM but leaving the mbedtls allocation to internal RAM.

I still get this errors at times but it fails, goes back to sleep and retries next minute without problem.
These are the errors I am seeing:

E (17772) HTTPS: Error perform http request ESP_ERR_HTTP_FETCH_HEADER

E (13138) esp-tls: read error :-29056:
E (13138) TRANS_SSL: esp_tls_conn_read error, errno=Bad file number

E (705388) esp-tls: Failed to connnect to host (errno 113)
E (705388) esp-tls: Failed to open new connection
E (705388) TRANS_SSL: Failed to open a new connection
E (705388) HTTP_CLIENT: Connection failed, sock < 0
E (705388) HTTPS: Error perform http request ESP_ERR_HTTP_CONNECT

E (3019788) esp-tls: mbedtls_ssl_handshake returned -0x4c
I (3019788) esp-tls: Certificate verified.
E (3019788) esp-tls: Failed to open new connection
E (3019788) TRANS_SSL: Failed to open a new connection
E (3019788) HTTP_CLIENT: Connection failed, sock < 0

I (16181) event: sta ip: 192.168.246.198, mask: 255.255.255.0, gw: 192.168.246.135
I (23511) wifi: bcn_timout,ap_probe_send_start
I (26021) wifi: ap_probe_send over, resett wifi status to disassoc
I (26021) wifi: state: unknown -> init (c800)

E (2292621) esp-tls: read error :-76:
E (2292621) TRANS_SSL: esp_tls_conn_read error, errno=No more processes

E (1426618) esp-tls: mbedtls_ssl_handshake returned -0x4c
I (1426618) esp-tls: Certificate verified.
E (1426618) esp-tls: Failed to open new connection
E (1426618) TRANS_SSL: Failed to open a new connection
E (1426618) HTTP_CLIENT: Connection failed, sock < 0

BTW I am using v3.3 and GNU Make with MSYS in Windows10.
As this was related somehow to memory issues, I thought this could be related to the PSRAM cache bug, so I was waiting for a release with a fix to try before opening a new issue.

shubhamkulkarni97, 2020-01-02:
> E (281379) esp-tls: read error :-76:
> E (281379) TRANS_SSL: esp_tls_conn_read error, errno=No more processes

This is timeout related issue. In case of slow network connections, you may try increasing timeout value in `esp_http_client_config_t`.

> E (112078) esp-tls: mbedtls_ssl_setup returned -0x7f00

The error value is `MBEDTLS_ERR_SSL_ALLOC_FAILED`, which suggests that `mbedtls_calloc` failed while `mbedtls_ssl_setup`. You might run out of stack in some cases, which leads to crash. To fix this issue, you may try increasing the stack size of the task.

Meanwhile, can you please attach sample code or crash logs and sdkconfig?

shubhamkulkarni97, 2020-01-13:
@ccasenave-corexpert Do you have any updates?

mahavirj, 2020-01-22:
Closing due to lack of response. Please feel free to re-open if you have any updates.

closed 2020-01-22 by mahavirj