espressif/esp-idf #4242: Persistent Provisioner (BLE mesh)

jkockx, 2019-10-24:
Hello
Some weeks ago I asked a question about making the provisioner of a BLE mesh network persistent. I got an answer that Espressif is working on this functionality, but I'm wondering when this functionality is to be expected because this is crucial to my application. Is this in terms of weeks or months?
Old Issue: https://github.com/espressif/esp-idf/issues/4141

Campou, 2019-10-25:
Hi @jkockx 

Currently we have implemented this feature and just test it, we will release this feature firstly in the branch `ble_mesh_release/esp-ble-mesh-v0.6.1` next week.

Thanks for your patience

Campou, 2019-10-28:
Hi @jkockx 

The Provisioner nvs functionality has been added in the branch `ble_mesh_release/esp-ble-mesh-v0.6.1`, please pull and have a try.

Thanks.

Alvin1Zhang, 2019-11-05:
@jkockx Thanks for reporting. Would you please help try and share the feedback? Thanks.

jkockx, 2019-11-09:
I'm sorry, I was away from this project for a while. I've pulled the `ble_mesh_release/esp-ble-mesh-v0.6.1`  branch but I'm not able to build my project anymore. Apparently, the export.sh script is missing in the main folder and because of this my system is unable set the correct path variables.
Should I apply another approach, different from the approach explained on the Getting Started page when working with this development branch?

Campou, 2019-11-11:
Hi @jkockx 

`ble_mesh_release/esp-ble-mesh-v0.6.1` is based on a specific commit `c064e008424d6f87cc78ca93ed572d4b2ba346d9`. If you are switching from `release/v4.0` or `master` branch to `ble_mesh_release/esp-ble-mesh-v0.6.1`, then the `export.sh` will be missing.

So if it's urgent that the Provisioner Persistant functionality needs to be used, please ref [here](https://docs.espressif.com/projects/esp-idf/en/latest/get-started-legacy/index.html) for some help.

If it's not urgent, the Provisioner Persistent functionlaity will be merged into the `master` and `release/v4.0` later. You can use the `release/v4.0` or `master` branch currently.

Thanks & Regards

jkockx, 2019-11-12:
So if I understand right, this branch is still using the legacy build system? How do I switch between the Toolchain setup that I've already setup and the setup you mentioned?

jkockx, 2019-11-14:
Or when do you think this functionality will be merged into the master branch? My question is again if we are talking about weeks or about months?

Campou, 2019-11-15:
Hi @jkockx 

Currenly this functionality is in review, and it will be merged into the master branch in a few weeks.

Thanks.

tidyjiang8, 2019-11-19:
@Campou 
Waiting this feature too...

Campou, 2020-05-27:
Hi @jkockx 

Sorry for the late reply. We have supported the persistent functionality of Provisioner, please have a try.
And this issue will be closed for now. If you have some other questions, please feel free to open another one. 

Thanks.

closed 2020-05-27 by Alvin1Zhang