espressif/esp-idf #4243: Nimble: Unable to stop advertising

zacwbond, 2019-10-24:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): 
    // v4.1-dev-369-g4dac7c7df
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 8.2
- Operating System:     Linux (via Docker)
- Power Supply:        external 5V
## Problem Description

I've been slowly converting an existing codebase from using Bluedroid to using Nimble.  Right now, I am working on implementing the Peripheral.  I am able to advertise, connect, handle writes/reads, send notifies, etc. all without a problem.  

My only problem is when I need to update some advertising data based on information sent by a connected central.  To do this, I begin by calling `ble_adv_stop()`, but this function invariably fails with a return code of `504` or `0x20C`, which means `BLE_HS_ERR_HCI_BASE` + `BLE_HS_ECONTROLLER`.  Apparently that means the controller sent an invalid event!?  

The only other thing I saw was an `BLE_HS_ENOMEM` if I try to enable advertising after the failed stop attempt.  That's odd, because I'd have expected `BLE_HS_EALREADY` instead.  The remaining heap at that moment was around 107K.  nRF Connect showed that advertising was still going on, as expected.

### Expected Behavior

`ble_adv_stop()` returns 0 and stops advertising.

### Actual Behavior

`ble_adv_stop()` always returns `BLE_HS_ECONTROLLER`.

### Steps to repropduce

1. Start a project that advertises.
2. Connect to the advertiser.
2. Stop advertising.

### Code to reproduce this issue

## Debug Logs

I don't have a useful debug log.  I'm not sure what I need to enable for Nimble to produce useful debug.  I'm happy to turn on anything you can suggest.

Nothing crashes or otherwise fails besides ble_adv_stop().  

## Other items if possible

dhrishi, 2019-10-25:
@zacwbond Once you are connected to a central, ESP32 peripheral stops advertising. So, you need not explicitly call ble_gap_adv_stop(). From the spec:

```
The Controller shall continue advertising until the Host issues an HCI_LE_Set_Advertising_Enable command with Advertising_Enable set to 0x00 (Advertising is disabled) or until a connection is created or until the Advertising is timed out due to high duty cycle Directed Advertising. 
```

The error you are seeing is 0x20c (or 524) which means: HCI_ERR_COMMAND_DISALLOWED (0x0c). The thing is that ESP32 controller returns with a disallowed error code when you try to stop advertising which is already stopped. You would not have seen this issue with Bluedroid because the Bluedroid host does not send the stop command to the controller if advertisement is already stopped but NimBLE host does send it.

The other problem you mentioned about restarting the advertisement after connection should not be related to the above issue. I think, you are getting BLE_HS_ENOMEM because the max number of connections set in menuconfig for NimBLE host are set to 1 by default. You can try to increase it through menuconfig: `make menuconfig -> Component config -> Bluetooth -> NimBLE Options -> Maximum number of concurrent connections` or pull the latest master branch which has its default value same as that of the controller: 3.

Please give it a try and let me know in case of anything.

zacwbond, 2019-10-29:
@dhrishi 

Thanks for the information.  I didn't notice that there was a default max connections.  Advertising seems to work well once I removed the extraneous stop.  I've managed to port almost our peripheral and our central from Bluedroid to Nimble, and nearly everything works.  

I've run into two additional problems I'm not sure how to solve.  I can open new issues if you like, but maybe I am just misunderstanding something.

1. We are using "just works" pairing with bonding between our central and a peripheral remote control.  The initial pairing works fine (at least as far as I an tell), as do subsequent re-connections by the same remote.  However, if I factory reset my remote control (that is, force it to erase its bonding information), I'm not sure how to get the central to re-pair correctly.  

```
E (423899) BLE Central: Exchanged MTU: 247
E (424198) BLE Central: event: BLE_GAP_EVENT_ENC_CHANGE
E (424199) BLE Central: encryption change event for 0x0000: status 0x206, encrypted 0 (key size 0), authenticated 0, bonded 0
E (54293) BLE Central: ble_gap_terminate() returned 0x212.
```

Error code `0x206` is `BLE_ERR_PINKEY_MISSING`, which I guess makes sense as the peer has deleted its key.  But I don't know what I'm supposed to do to recover so that the remote can pair again.  Here's my best guess for what I was supposed to do in the GAP handler, but it doesn't work--as you see in the log, the attempt to terminate returns `BLE_ERR_INV_HCI_CMD_PARMS` for some reason.  

```
case BLE_GAP_EVENT_ENC_CHANGE:
            BAssert(!ble_gap_conn_find(event->enc_change.conn_handle, &desc));

            ESP_LOGE(LOGGING_TAG,
                     "encryption change event for 0x%04X: status %u, encrypted "
                     "%u (key size %u), authenticated %u, bonded %u",
                     event->enc_change.conn_handle,
                     event->enc_change.status,
                     desc.sec_state.encrypted,
                     desc.sec_state.key_size,
                     desc.sec_state.authenticated,
                     desc.sec_state.bonded);

            if (event->enc_change.status
                == BLE_HS_ERR_HCI_BASE + BLE_ERR_PINKEY_MISSING)
            {
                ESP_LOGE(LOGGING_TAG, "Security failure.");

                // Something went wrong with pairing, so delete the peer from the store.
                int err{ble_store_util_delete_peer(&desc.peer_id_addr)};

                if (err)
                {
                    ESP_LOGE(LOGGING_TAG,
                             "ble_store_util_delete_peer() returned %u.",
                             err);
                }

                err = ble_gap_terminate(event->enc_change.conn_handle,
                                        BLE_ERR_CONN_TERM_LOCAL);

                if (err)
                {
                    ESP_LOGE(LOGGING_TAG, "ble_gap_terminate() returned %u.", err);
                }                
                return 0;
            }
```

2. With Bluedroid, I was able to remove bonding for specific devices, whether they are connected or not.  In Nimble, it seems like I should be able to do this by using `ble_gap_unpair()` to unpair and delete bonding for a given address, plus `ble_store_util_bonded_peers()` to get the list of bonded peers.  The trouble is that `ble_gap_unpair()` doesn't appear to delete anything from storage, so that I can "unpair" from a remote, but still find that remote's address in the list returned by `ble_store_util_bonded_peers()`.  

When a remote is connected, `ble_gap_unpair()` will close the connection but the remote will remain in the list.  On the other hand, if the remote is not connected, then `ble_gap_unpair()` will return `BLE_ERR_CMD_DISALLOWED`. 
 What am I missing?

prasad-alatkar, 2019-10-31:
> When a remote is connected, ble_gap_unpair() will close the connection but the remote will remain in the list

Yes @zacwbond , I have actually raised an upstream pull request to get the device un-paired using `ble_gap_unpair` irrespective of its connection status, I will update you on this ASAP. However, the simple workaround for this could be calling `ble_gap_terminate` before `ble_gap_unpair`. 

> On the other hand, if the remote is not connected, then ble_gap_unpair() will return BLE_ERR_CMD_DISALLOWED.

Can you please share logs for this ? I am suspecting you might be getting error code as `BLE_ERR_UNK_CONN_ID (0x202)`, I am working to get this resolved as well. 

However as a temporary workaround for this might be as below (from your code snippet)

```
case BLE_GAP_EVENT_ENC_CHANGE:
            BAssert(!ble_gap_conn_find(event->enc_change.conn_handle, &desc));

            ESP_LOGE(LOGGING_TAG,
                     "encryption change event for 0x%04X: status %u, encrypted "
                     "%u (key size %u), authenticated %u, bonded %u",
                     event->enc_change.conn_handle,
                     event->enc_change.status,
                     desc.sec_state.encrypted,
                     desc.sec_state.key_size,
                     desc.sec_state.authenticated,
                     desc.sec_state.bonded);

           if (event->enc_change.status
                == (BLE_HS_ERR_HCI_BASE + BLE_ERR_PINKEY_MISSING))
            {
                ESP_LOGE(LOGGING_TAG, "Security failure.");

                err = ble_gap_terminate(event->enc_change.conn_handle,
                                        BLE_ERR_CONN_TERM_LOCAL);

                if (err)
                {
                    ESP_LOGE(LOGGING_TAG, "ble_gap_terminate() returned %u.", err);
                }

                // Something went wrong with pairing, so delete the peer from the store.
                int err{ble_store_util_delete_peer(&desc.peer_id_addr)};

                if (err)
                {
                    ESP_LOGE(LOGGING_TAG,
                             "ble_store_util_delete_peer() returned %u.",
                             err);
                }
                
            }
   
    return 0;
```

zacwbond, 2019-11-01:
@prasad-alatkar Thanks for your recommendations!  

> Can you please share logs for this ? I am suspecting you might be getting error code as BLE_ERR_UNK_CONN_ID (0x202), I am working to get this resolved as well.

There's not that much logging for this, but here it is:

```
E (890954) Ble Driver: Removing bonding for E9:7E:E8:F5:89:F9
E (890956) Ble Driver: ble_gap_unpair() returned 524

E (890956) Ble Driver: Bonded devices:
E (890956) Ble Driver:    E9:7E:E8:F5:89:F9 1
```

Digging into it, that `BLE_ERR_CMD_DISALLOWED` return code is coming from the function `ble_hs_pvcy_remove_entry()`, which is called by `ble_gap_unpair()` before it gets to `ble_store_util_delete_peer()`.  I'm not sure what to do about that.

For my other problem (repairing after the peer has deleted its key), I gave your suggested code change a try, but ran into the BLE_ERR_INV_HCI_CMD_PARMS error again.  Turns out that in order to terminate, you need to provide the reason from the perspective of the peer, so I should have sent `BLE_ERR_REM_USER_CONN_TERM` instead of `BLE_ERR_CONN_TERM_LOCAL`.  If I use the right reason code, it works.

dhrishi, 2019-11-02:
> Turns out that in order to terminate, you need to provide the reason from the perspective of the peer, so I should have sent BLE_ERR_REM_USER_CONN_TERM instead of BLE_ERR_CONN_TERM_LOCAL.

That's right

prasad-alatkar, 2019-11-04:
> Digging into it, that BLE_ERR_CMD_DISALLOWED return code is coming from the function ble_hs_pvcy_remove_entry(), which is called by ble_gap_unpair() before it gets to ble_store_util_delete_peer(). I'm not sure what to do about that.

Yes @zacwbond, that is what I had observed and I have raised an upstream PR [PR!636](https://github.com/apache/mynewt-nimble/pull/636) for the same.  However for me the error code was `BLE_ERR_UNK_CONN_ID`, which by spec Vol 2, Part E, section 7.8.39 (**_When a Controller cannot remove a device from the resolving list because it is not found, it shall return the error code Unknown Connection Identifier (0x02)._**)

>  Turns out that in order to terminate, you need to provide the reason from the perspective of the peer, so I should have sent BLE_ERR_REM_USER_CONN_TERM instead of BLE_ERR_CONN_TERM_LOCAL. If I use the right reason code, it works.

Yes, I actually took it from your code snippet directly, my bad !!

zacwbond, 2020-01-29:
Hi @prasad-alatkar,  

I saw that your upstream PR had been merged, and that the change has been pulled into the ESP code as well.  But, I'm still unable to use `ble_gap_unpair()` in the way I want to because it calls `ble_hs_conn_find_by_addr()`, which has the assertion `BLE_HS_DBG_ASSERT(ble_hs_locked_by_cur_task())` and I am calling it from the unlocked context of the event handler.

prasad-alatkar, 2020-01-30:
@zacwbond , `ble_hs_conn_find_by_addr` needs to be called with `ble_hs_lock` . I will get this fixed by including upstream [PR!584](https://github.com/apache/mynewt-nimble/pull/584).

closed 2020-02-04 by espressif-bot