espressif/esp-idf #4244: examples/ethernet/basic not building

mzimmers, 2019-10-24:
Working with the ESP32-Ethernet-Kit_A_V1.1, trying to build the basic Ethernet example. The source code wants macros like CONFIG_EXAMPLE_USE_INTERNAL_ETHERNET defined, but no such macros exist in sdkconfig. Something appears out of sync; either the source code needs to be updated, or the sdkconfig file isn't being generated properly. Or, of course, it could all be a mistake on my part.

suda-morris, 2019-10-25:
can you try again in this way in the examples/ethernet/basic:
```bash
rm -rf build/
rm sdkconfig*
idf.py build flash monitor
```

mzimmers, 2019-10-25:
Hi - 

Thanks for the suggestion -- while that didn't fix it, it helped me discover the problem. I had originally copied the example when my IDF was V3.3. When I moved to V4.1 (for this example), I copied the source code from the examples directory, but not the Kconfig.projbuild file. Now I know better - always copy the entire source directory.

closed 2019-10-25 by mzimmers