espressif/esp-idf #4247: ld.exe: cannot find crt1-sim.o: No such file or directory

cslehel, 2019-10-25:
## Environment

- Development Kit: Lolin D32
- Module or chip used:  ESP32-WROOM-32
- IDF version v3.3
- Build System: CMake
- Compiler version:1.22.0-80-g6c4433a5-5.2.0
- Operating System: Windows 10
- Power Supply: USB

## Problem Description

Hello, i installed the ESP-IDF with the esp-idf-tools-setup-2.1.exe installer. 
After that i unpacked the hello_world example and i tried to run the following commands from the documentation: 

cd %userprofile%\esp\hello_world
idf.py menuconfig

And i am getting the following errors in the command line: 

CMake Error at C:/Users/Developer/esp/hello_world/main/CMakeLists.txt:1 (idf_component_register):
  Unknown CMake command "idf_component_register".
Call Stack (most recent call first):
  C:/Users/Developer/Desktop/esp-idf/tools/cmake/scripts/expand_requirements.cmake:107 (include)
  C:/Users/Developer/Desktop/esp-idf/tools/cmake/scripts/expand_requirements.cmake:217 (expand_component_requirements)

CMake Error at C:/Users/Developer/Desktop/esp-idf/CMakeLists.txt:39 (message):
  Failed to expand component requirements

-- Configuring incomplete, errors occurred!
See also "C:/Users/Developer/esp/hello_world/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/Developer/esp/hello_world/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1

And i have the following error in the CMakeOutput.log: 

Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: C:/Users/Developer/.espressif/tools/xtensa-esp32-elf/1.22.0-80-g6c4433a5-5.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe 
Build flags: -mlongcalls
Id flags:  

The output was:
1
c:/users/developer/.espressif/tools/xtensa-esp32-elf/1.22.0-80-g6c4433a5-5.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find crt1-sim.o: No such file or directory
c:/users/developer/.espressif/tools/xtensa-esp32-elf/1.22.0-80-g6c4433a5-5.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find _vectors.o: No such file or directory
c:/users/developer/.espressif/tools/xtensa-esp32-elf/1.22.0-80-g6c4433a5-5.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lsim
c:/users/developer/.espressif/tools/xtensa-esp32-elf/1.22.0-80-g6c4433a5-5.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lhandlers-sim
c:/users/developer/.espressif/tools/xtensa-esp32-elf/1.22.0-80-g6c4433a5-5.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lhal
collect2.exe: error: ld returned 1 exit status

### Expected Behavior
I was expecting something similar described in the documentation: 
https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#id9

Thank you.

negativekelvin, 2019-10-25:
Based on `Unknown CMake command "idf_component_register".` it looks like you are trying to build an example from esp-idf 4.0+ against esp-idf 3.3 which won't work.

cslehel, 2019-10-26:
Yes you are right, i was using the examples from the github not the ones installed. 

Thank you very much for your help.

closed 2019-10-26 by cslehel

AHafsy, 2019-12-23:
Hello,
I have the same problem: crt1, vectors, lsim, lhand and lhal are not found on the xtensa-esp32-elf/bin:ld path. But it's not because of `Unknown CMake command "idf_component_register"`. 
Here is my output when I run `idf.py menuconfig`
`Executing action: menuconfig
Running cmake in directory /home/esp8266/hello_world/build
Executing "cmake -G 'Unix Makefiles' -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=0 /home/esp8266/hello_world"...
Warn about uninitialized values.
-- Found Git: /usr/bin/git (found version "1.9.1") 
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/esp8266/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/esp8266/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/esp8266/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- broken
CMake Error at /usr/local/share/cmake-3.15/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/home/esp8266/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/esp8266/hello_world/build/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/usr/bin/make cmTC_8f7a2/fast && /usr/bin/make -f CMakeFiles/cmTC_8f7a2.dir/build.make CMakeFiles/cmTC_8f7a2.dir/build
    make[1]: Entering directory `/home/esp8266/hello_world/build/CMakeFiles/CMakeTmp'
    Building C object CMakeFiles/cmTC_8f7a2.dir/testCCompiler.c.obj
    /home/esp8266/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc   -mlongcalls -Wno-frame-address    -o CMakeFiles/cmTC_8f7a2.dir/testCCompiler.c.obj   -c /home/esp8266/hello_world/build/CMakeFiles/CMakeTmp/testCCompiler.c
    Linking C executable cmTC_8f7a2
    /usr/local/bin/cmake -E cmake_link_script CMakeFiles/cmTC_8f7a2.dir/link.txt --verbose=1
    /home/esp8266/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -mlongcalls -Wno-frame-address     CMakeFiles/cmTC_8f7a2.dir/testCCompiler.c.obj  -o cmTC_8f7a2 
    /home/esp8266/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find crt1-sim.o: No such file or directory
    /home/esp8266/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find _vectors.o: No such file or directory
    /home/esp8266/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lsim
    /home/esp8266/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lhandlers-sim
    /home/esp8266/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lhal
    collect2: error: ld returned 1 exit status
    make[1]: *** [cmTC_8f7a2] Error 1
    make[1]: Leaving directory `/home/esp8266/hello_world/build/CMakeFiles/CMakeTmp'
    make: *** [cmTC_8f7a2/fast] Error 2
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  /home/esp8266/esp-idf/tools/cmake/project.cmake:269 (__project)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "/home/esp8266/hello_world/build/CMakeFiles/CMakeOutput.log".
See also "/home/esp8266/hello_world/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1
`
Do you have a solution for this type of problem?
Thank you!

igrr, 2019-12-23:
@AHafsy  you seem to be using IDF 4.0 or 4.1 together with GCC 5.2 based toolchain. Please check the getting started guides for installing the correct version of the compiler: https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#step-3-set-up-the-tools.

AHafsy, 2019-12-23:
Thank you @igrr,
So, in my case I should use GCC 4.1 ?

igrr, 2019-12-23:
In IDF 4.0 and later, the correct toolchain version will be installed automatically if you follow the instructions linked in my previous comment.

At the moment the version which will be installed is `esp-2019r2`, it is based on GCC 8.2 and newlib 3.0.

If you wish to install the toolchain manually (instead of using `install.sh`), you can find the download links here: https://docs.espressif.com/projects/esp-idf/en/latest/get-started-legacy/linux-setup.html#toolchain-setup