espressif/esp-idf #4248: ID/size check in spi_flash_chip_generic_detect_size() is too strict

RobMeades, 2019-10-25:
We have a module of our own containing an ESP32 plus a Macronics MX25R1635F 2 Mbyte flash device.  This module worked fine under ESP-IDF v3.1.1.  Yesterday I update to the latest ESP-IDF `release/v4` with its very fast and whizzy new IDE (very happy with that).

However, I am forced to use the new SPI driver (setting the flag to use the pre-v4 SPI driver in `menuconfig` results in a compilation error) and, with the new SPI driver, our 2Mbyte flash chip is no longer supported.

The chip ID is `0xC22815`:
  - manufacturer: `0xC2`,
  - memory type: `0x28`,
  - memory density: `0x15`.

...whereas the check in `spi_flash_chip_generic_detect_size()` requires that the lower four bytes of the memory type field are zero:

```
/* Can't detect size unless the high byte of the product ID matches the same convention, which is usually 0x40 or
     * 0xC0 or similar. */
    if ((id & 0x0F00) != 0) {
        return ESP_ERR_FLASH_UNSUPPORTED_CHIP;
    }
```

If I comment out the check, the size comes out correctly and everything works fine.

For the moment I have forked this repo and commented out the check but it would be better if the genericness of the generic SPI flash driver could be expanded to allow such flash devices through.  Maybe and with `0x0700`?

negativekelvin, 2019-10-25:
Sidebar: love the form factor, too bad there is no way to fit a bigger flash chip in there since 2mb is very limiting.

jwestmoreland, 2020-03-23:
Hello All,

I'm seeing a similar issue with the board I have - it has the uBlox NINA-W102 on board.

I thought checking the 'don't check flash size on boot' would bypass all of this - but maybe that's just for the flash tool?

Thanks,
John W.

jwestmoreland, 2020-03-27:
Hello All,

Also - I've noticed when there is another problem with the build - this happens:

E (760) spi_flash: failed to get chip size
ESP32 (cpu_start.c) - flash_ret = 0x6005
assertion "flash_ret == ESP_OK" failed: file "C:/esp-tot/esp-idf/components/esp32/cpu_start.c", line 402, function: start_cpu0_default
abort() was called at PC 0x401af9a3 on core 0

In the old days we used to call these 'speed paths' - in this case - this is what the error is.

I can compile a different project using the same underlying esp-idf database, and this doesn't happen.  Same cpu_start.c.

FWIW,
John W.

jwestmoreland, 2020-03-30:
Hello Again Everyone,

While <I think> I got the above at times when I didn't have enough heap and/or I tried to start a task too soon on core 1... the above - for me and the specific target I'm working on - was actually this:

https://github.com/espressif/esp-idf/issues/5020

Anyway, HTH,
John W.

ginkgm, 2020-07-16:
@RobMeades 
The check id issue is solved in 79a5b0b5afbabeacfdcb7dc6c5069730ffd496ac, please wait until it's synced to Github...

ginkgm, 2020-07-16:
@negativekelvin We plan to support that, thanks for mentioning this and your waiting patiently!

RobMeades, 2020-07-21:
@ginkgm thanks!

closed 2020-07-21 by RobMeades

amptek-flau, 2021-04-22:
Hello All,

I am new to the u-blox NINA-W102.  I have a EVK and am trying my very first hello_world.  Took me a while to figure out how to flash to the NINA-W102.  However, the module kept rebooting with the following messages.  I see the spi_flash: failed to get chip size error, plus some other problems.  I don't have any clue on how to fix the problems.  Your suggestions are very much appreicated.

rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7068
load:0x40078000,len:13212
ho 0 tail 12 room 4
load:0x40080400,len:4568
0x40080400: _init at ??:?

entry 0x400806f4
I (31) boot: ESP-IDF v4.2-dirty 2nd stage bootloader
I (31) boot: compile time 16:03:58
I (31) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 2MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x05ad8 ( 2
3256) map
I (122) esp_image: segment 1: paddr=0x00015b00 vaddr=0x3ffb0000 size=0x02068 (
8296) load
I (126) esp_image: segment 2: paddr=0x00017b70 vaddr=0x40080000 size=0x00404 (
1028) load
0x40080000: _WindowOverflow4 at C:/Users/flau/esp-idf/components/freertos/xtensa
/xtensa_vectors.S:1730

I (132) esp_image: segment 3: paddr=0x00017f7c vaddr=0x40080404 size=0x0809c ( 3
2924) load
I (155) esp_image: segment 4: paddr=0x00020020 vaddr=0x400d0020 size=0x12fe4 ( 7
7796) map
0x400d0020: _stext at ??:?

I (185) esp_image: segment 5: paddr=0x0003300c vaddr=0x400884a0 size=0x01a54 (
6740) load
0x400884a0: DPORT_READ_PERI_REG at C:/Users/flau/esp-idf/components/soc/soc/esp3
2/include/soc/dport_access.h:172
 (inlined by) rtc_init at C:/Users/flau/esp-idf/components/soc/src/esp32/rtc_ini
t.c:47

I (194) boot: Loaded app from partition at offset 0x10000
I (194) boot: Disabling RNG early entropy source...
I (194) cpu_start: Pro cpu up.
I (198) cpu_start: Application information:
I (203) cpu_start: Project name:     hello-world
I (208) cpu_start: App version:      1
I (212) cpu_start: Compile time:     Apr 21 2021 21:41:39
I (218) cpu_start: ELF file SHA256:  7bd61b6317be8777...
I (224) cpu_start: ESP-IDF:          v4.2-dirty
I (230) cpu_start: Starting app cpu, entry point is 0x400815e8
0x400815e8: call_start_cpu1 at C:/Users/flau/esp-idf/components/esp32/cpu_start.
c:287

I (0) cpu_start: App cpu up.
I (240) heap_init: Initializing. RAM available for dynamic allocation:
I (247) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (253) heap_init: At 3FFB28A0 len 0002D760 (181 KiB): DRAM
I (259) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (266) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (272) heap_init: At 40089EF4 len 0001610C (88 KiB): IRAM
I (278) cpu_start: Pro cpu start user code
I (296) spi_flash: detected chip: mxic
E (297) spi_flash: failed to get chip size
assertion "flash_ret == ESP_OK" failed: file "C:/Users/flau/esp-idf/components/e
sp32/cpu_start.c", line 463, function: start_cpu0_default

abort() was called at PC 0x400d55af on core 0
0x400d55af: __assert_func at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-min
gw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8
)

Backtrace:0x40083f23:0x3ffe3b10 0x400845c1:0x3ffe3b30 0x40087b3e:0x3ffe3b50 0x40
0d55af:0x3ffe3bc0 0x4008156e:0x3ffe3bf0 0x400817ef:0x3ffe3c40 0x40079026:0x3ffe3
c80 |<-CORRUPTED
0x40083f23: panic_abort at C:/Users/flau/esp-idf/components/esp_system/panic.c:3
30

0x400845c1: esp_system_abort at C:/Users/flau/esp-idf/components/esp_system/syst
em_api.c:106

0x40087b3e: abort at C:/Users/flau/esp-idf/components/newlib/abort.c:46

0x400d55af: __assert_func at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-min
gw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8
)

0x4008156e: start_cpu0_default at C:/Users/flau/esp-idf/components/esp32/cpu_sta
rt.c:463 (discriminator 1)

0x400817ef: call_start_cpu0 at C:/Users/flau/esp-idf/components/esp32/cpu_start.
c:275

amptek-flau, 2021-04-22:
Just read Macronix MX25R1635F support (IDFGH-2988) #5020.  Checking the Use the legacy implementation before IDF v4.0 in the SDK Configurator solved the problem.  Thanks everyone.