espressif/esp-idf #4249: gatts 无法断开连接

MGDG, 2019-10-26:
你好，我在使用过程中发现有时候客户端断开后ESP32无法继续广播，定位发现是客户端断开连接后esp32无法产生 **ESP_GATTS_DISCONNECT_EVT** 事件，此时调用 **esp_ble_gatts_close**后可以返回**ESP_OK**，但是依然没有产生 **ESP_GATTS_DISCONNECT_EVT** 事件
使用以下方式关闭蓝牙后再重新启动又可以了
```cpp
static void gatts_deinit(void) {
    debug_w("gatts deinit");
    esp_err_t ret = esp_bluedroid_disable();
    if (ret != ESP_OK) {
        debug_e( "bluedroid disable err: %x", ret);
        return;
    }
    debug_i("bluedroid disable success");
    ret = esp_bluedroid_deinit();
    if (ret != ESP_OK) {
        debug_e("bluedroid deinit err: %x", ret);
        return;
    }
    debug_i("bluedroid deinit success");
    ret = esp_bt_controller_disable();
    if (ret != ESP_OK) {
        debug_e("bt controller disable err: %x", ret);
        return;
    }
    debug_i("bt controller disable success");
    ret = esp_bt_controller_deinit();
    if (ret != ESP_OK) {
        debug_e("bt controller deinit err: %x", ret);
        return;
    }
    debug_i("bt controller deinit success");
}
```

Alvin1Zhang, 2019-10-28:
@MGDG Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

MGDG, 2019-10-28:
## Environment

- IDF version : v3.2.2-42-g4a9f339-dirty
- mdf version : v1.0-beta1-dirty
- chip : ESP32-D0WD
- spiram: 64MBit

## log: 正常连接后客户端主动断开连接

```
W (1028558) [GATTS, 719]: GATTS_CONNECT, conn_id=0, remote 68:0B:02:98:87:04
I (1028648) [GATTS, 650]: GATTS_MTU, MTU=185
I (1029631) [main, 66]: sysInfo: staMac:4C11AEB6E2E4, fw:LING_MC_V1.0.5, freeMem:2005712, ip:0.0.0.0, time:2019/10/28-1-14:58:20, resetReason:1, parentBssid: 4c:11:ae:b6:e2:bd
I (1029907) [GATTS, 616]: GATT_WRITE, conn_id=0, trans_id=2, handle=43
I (1029909) [GATTS, 618]: GATT_WRITE, value len=2, value :
I (1029910) GATTS: 01 00 
I (1029911) [GATTS, 627]: notify enable
W (1030658) [GATTS, 725]: GATTS_DISCONNECT, reason:0x13
I (1030662) [GATTS, 807]: Adv start success
```

## log: 正常连接后超时断开
``我做了一个超过5000ms没有数据交互就调用esp_ble_gatts_close主动断开连接的机制，如果连续调用了esp_ble_gatts_close3次后（调用间隔500ms）仍然没有断开就关闭ble并重新初始化``
```
W (2092775) [GATTS, 719]: GATTS_CONNECT, conn_id=0, remote 67:70:6B:95:67:FA
I (2092866) [GATTS, 650]: GATTS_MTU, MTU=185
I (2094110) [GATTS, 616]: GATT_WRITE, conn_id=0, trans_id=2, handle=43
I (2094112) [GATTS, 618]: GATT_WRITE, value len=2, value :
I (2094113) GATTS: 01 00 
I (2094114) [GATTS, 627]: notify enable
I (2094631) [main, 66]: sysInfo: staMac:4C11AEB6E2E4, fw:LING_MC_V1.0.5, freeMem:2005504, ip:0.0.0.0, time:2019/10/28-1-15:16:05, resetReason:1, parentBssid: 4c:11:ae:b6:e2:bd
W (2099117) [GATTS, 446]: gattc timeout (5000ms), gatts_if:3, conn_id:0
W (2099119) [GATTS, 725]: GATTS_DISCONNECT, reason:0x16
I (2099124) [GATTS, 807]: Adv start success
```

## log: 客户端连接后断开，无法产生ESP_GATTS_DISCONNECT_EVT 

```
W (914110) [GATTS, 719]: GATTS_CONNECT, conn_id=0, remote 68:0B:02:98:87:04
I (914201) [GATTS, 650]: GATTS_MTU, MTU=185
I (914631) [main, 66]: sysInfo: staMac:4C11AEB6E2E4, fw:LING_MC_V1.0.5, freeMem:2005720, ip:0.0.0.0, time:2019/10/28-1-14:56:25, resetReason:1, parentBssid: 4c:11:ae:b6:e2:bd
W (919112) [GATTS, 446]: gattc timeout (5000ms), gatts_if:3, conn_id:0
W (919617) [GATTS, 446]: gattc timeout (5000ms), gatts_if:3, conn_id:0
I (919631) [main, 66]: sysInfo: staMac:4C11AEB6E2E4, fw:LING_MC_V1.0.5, freeMem:2006228, ip:0.0.0.0, time:2019/10/28-1-14:56:30, resetReason:1, parentBssid: 4c:11:ae:b6:e2:bd
W (920122) [GATTS, 446]: gattc timeout (5000ms), gatts_if:3, conn_id:0
W (920122) [GATTS, 179]: restart gatts
W (920173) [GATTS, 975]: gatts stopping
W (920178) [GATTS, 884]: gatts deinit
I (926680) [GATTS, 890]: bluedroid disable success
I (926681) [GATTS, 896]: bluedroid deinit success
I (926684) [GATTS, 902]: bt controller disable success
I (926684) [GATTS, 908]: bt controller deinit success
W (926685) [GATTS, 945]: gatts stop
W (926687) [GATTS, 977]: gatts stopped
W (926690) [GATTS, 962]: gatts running
I (926690) BTDM_INIT: BT controller compile version [5186949]
I (926696) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (926701) [main, 66]: sysInfo: staMac:4C11AEB6E2E4, fw:LING_MC_V1.0.5, freeMem:2044532, ip:0.0.0.0, time:2019/10/28-1-14:56:37, resetReason:1, parentBssid: 4c:11:ae:b6:e2:bd
I (926948) [GATTS, 567]: REGISTER_APP, status=0, app_id=0
W (926950) [GATTS, 572]: ble name: LL58594
I (926953) [GATTS, 655]: CREATE_SERVICE, status=0,  service_handle=40
I (926957) [GATTS, 701]: SERVICE_START, status=0, service_handle=40
I (926958) [GATTS, 674]: ADD_CHAR, status=0,  attr_handle=42, service_handle=40
I (926961) [GATTS, 683]: gatts char length = 1
I (926963) [GATTS, 685]: prf_char[0] = 0x12
I (926966) [GATTS, 807]: Adv start success
I (926968) [GATTS, 696]: ADD_DESCR, status=0, attr_handle=43, service_handle=40
```

### sdkconfig 
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/3777556/sdkconfig.zip)

### gatts代码
[gatts.zip](https://github.com/espressif/esp-idf/files/3777558/gatts.zip)

gengyuchao, 2019-10-29:
你好，感谢你的反馈，您能否在您的工程中先关闭 SPIRAM ，确认一下问题是否还会发生。在 menuconfig 中关闭 SPIRAM 的目录路径是 `> Component config > Bluetooth > Bluedroid Enable ` 的 “`BT/BLE will first malloc the memory from the PSRAM ` 选项

MGDG, 2019-10-29:
你好，按照你的建议关闭SPIRAM后貌似并没有改善，依然会出现无法断开的现象，并且带来了新的问题，在使用i2c时会频繁打印以下log，该问题可能和内存不够用有关系
```
E (136680) i2c: i2c command link malloc error
E (136680) i2c: i2c command link malloc error
E (136680) i2c: i2c command link malloc error
E (136690) i2c: i2c command link malloc error
E (136691) i2c: i2c command link malloc error
E (136691) i2c: i2c command link malloc error
E (136693) i2c: i2c command link malloc error
I (137145) [main, 66]: sysInfo: staMac:246F284C733C, fw:LING_MC_V1.0.5, freeMem:1943936, ip:192.168.6.160, time:2019/10/29-2-14:50:19, resetReason:1, parentBssid: 00:00:00:00:00:00
```
我在项目中同时使用了BLE & ETH & WIFI和MESH，是否和这个有关

gengyuchao, 2019-10-29:
新的问题可能是因为关掉了 SPIRAM 导致内存不足，根据你的描述，这个问题不是我之前遇到的 SPIRAM 问题导致的。我根据你上传的文件运行了你的 gatts_task ，链接了五十次但是没有复现这个问题，每次都能正常断开，是否还有其他触发问题的条件？

MGDG, 2019-10-30:
我屏蔽了代码中mesh相关的功能后测试50次都很正常，该问题是否和mesh功能有关系

gengyuchao, 2019-10-30:
目前还无法确定，你能否提供一个能够让我复现这个问题的方法或者代码，来让我了解更多有关这个问题出现的信息。谢谢。

gengyuchao, 2019-12-09:
你好，这个问题还存在吗？我对这个问题有了一些其他的发现。如果还存在，你可以尝试下面这个方法：把文件 `esp-idf/components/bt/host/bluedroid/stack/l2cap/include/l2c_int.h` 中的 `typedef struct t_l2c_linkcb` 结构体里的   `  tBT_TRANSPORT       transport;` 改为` int         transport;` 。

MGDG, 2019-12-11:
很抱歉没有及时回复你，按你的建议更改后目前可以正常使用，我原本想尝试屏蔽代码中的业务逻辑，提供一个可复现问题的代码给你们测试，但是一直没时间完成。
我发现这个问题似乎跟编译有关系，在我按你的建议更改之前，我完全不更改GATTS模块代码的前提下，我调整了另一个完全不相关模块中的结构体元素顺序，问题可能就不会复现

gengyuchao, 2019-12-11:
这个问题可能涉及到了一个已知的硬件bug，在访问外部 SPIRAM 时会有极小概率发生。在 ESP32 勘误表3.2节中有关于这个问题的说明。按照这个方法修改后可以解决这里的问题，完全避免这个问题需要等待工具链的更新。

zf8848, 2020-02-24:
我也碰到无法接收到 disconnect 问题，用的是 v3.3.1, 新的 v4.0 工具链解决这个问题了?

gengyuchao, 2020-02-24:
出现这个问题需要汇编语言编译时出现特定的汇编指令才会触发这个bug。改动任何一处代码都可能让这个bug消失。我还不清楚新工具链的事情。: )

jack0c, 2022-01-05:
IDFv3.3 以后的版本都已经解决了工具链的问题

closed 2022-01-05 by jack0c