espressif/esp-idf #4250: No rule to make target 'menuconfig'. Stop.

amirjak, 2019-10-26:
**I** download and extract the repository to the **components** folder of **my** ESP-IDF project.
But when I want to **make menuconfig** this happened.
[image]

**I** use **python 2.7 8** ( windows 10 ) 64 bit

dobairoland, 2019-10-27:
Hi @amirjak. You are not supposed to run `make` in the components directory. Please create your own project (based on the template or an example application) and set the `IDF_PATH` to your `~/esp-idf`.
`make` should be run in your project folder.

closed 2019-10-27 by amirjak

amirjak, 2019-10-27:
**Thank you very much**