espressif/esp-idf #4261: High current consumption after esp_bt_gap_set_scan_mode(ESP_BT_SCAN_MODE_NONE)

lollizze, 2019-10-29:
## Environment
- Development Kit: none
- Module or chip used:  ESP32
- IDF version (run ``git describe --tags`` to find it): 90af68bd3c1941d9f5513bee22a12de7250208d7
- Build System:  Make
- Compiler version:1.22.0-80-g6c4433a5
- Operating System:   Windows
- Power Supply:        external 3.3V

## Problem Description

the command to set scan mode to none seems to never put the radio to sleep.
If we issue the command 
`esp_bt_gap_set_scan_mode(ESP_BT_SCAN_MODE_CONNECTABLE_DISCOVERABLE);` and measure power consumption we can see that the average consumption is around 30mA with peaks at 100mA. But if we instead do `esp_bt_gap_set_scan_mode(ESP_BT_SCAN_MODE_NONE);`  the consumption goes to constant 100mA
### Expected Behavior
Power consumption should be the lowest possible
### Actual Behavior
 constant 100mA
### Steps to repropduce
It is really simple to reproduce, just change the esp_bt_gap_set_scan_mode in my example and measure the power consumption
[bt_discovery.zip](https://github.com/espressif/esp-idf/files/3784563/bt_discovery.zip)

Alvin1Zhang, 2019-10-30:
@lollizze Thanks for reporting, we will look into. Thanks.

blueMoodBHD, 2021-01-14:
@lollizze Sorry for late reply, I have test it and there is no problem. Does the issue still exist?

jack0c, 2023-04-12:
Close since no update long time

closed 2023-04-12 by jack0c