espressif/esp-idf #4262: BLE GATT connect causes both ESP_GATTS_CONNECT_EVT and  ESP_GATTC_CONNECT_EVT

domonoky, 2019-10-29:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version: v4
- Module or chip used:  ESP32-WROVER-B
- IDF version (run ``git describe --tags`` to find it): v4.0-beta1
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 5.2.0
- Operating System:     Windows
- Power Supply:         USB

## Problem Description
I am trying to create an app which is BLE client and server at the same time. It should connect to a BLE client and get some data from it and also present a BLE Interface for a phone app.
I tried this with IDF 3.2 (included in esp ADF ), the IDF3.3 and also IDF4.0-beta1. 

Every time the gattc (client) code finds the client and starts the connection. GATTC and GATTS get the connect event. 
Also if the smartphone connects to the BLE server we get a CONNECT event for both the client and the server.

For reproduction, i tried it with the gattc_gatts_coex example from the IDF4.0-beta1. The same issues happens here. See log from output:

I (0) cpu_start: Starting scheduler on APP CPU.
I (530) BTDM_INIT: BT controller compile version [a482cda]
I (530) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (630) phy: phy_version: 4102, 2fa7a43, Jul 15 2019, 13:06:06, 0, 0
I (940) GATTC_GATTS_COEX: REGISTER_APP_EVT, status 0, app_id 0
W (940) BT_BTM: BTM_BleWriteAdvData, Partial data write into ADV
I (950) GATTC_GATTS_COEX: CREATE_SERVICE_EVT, status 0,  service_handle 40
I (950) GATTC_GATTS_COEX: REGISTER_APP_EVT, status 0, app_id 1
I (960) GATTC_GATTS_COEX: REG_EVT
I (970) GATTC_GATTS_COEX: Advertising start successfully
I (970) GATTC_GATTS_COEX: SERVICE_START_EVT, status 0, service_handle 40
I (980) GATTC_GATTS_COEX: ADD_CHAR_EVT, status 0,  attr_handle 42, service_handle 40
I (990) GATTC_GATTS_COEX: CREATE_SERVICE_EVT, status 0,  service_handle 44
I (990) GATTC_GATTS_COEX: ESP_GAP_BLE_SCAN_PARAM_SET_COMPLETE_EVT, set scan sparameters complete
I (1010) GATTC_GATTS_COEX: ADD_DESCR_EVT, status 0, attr_handle 43, service_handle 40
I (1010) GATTC_GATTS_COEX: SERVICE_START_EVT, status 0, service_handle 44
I (1020) GATTC_GATTS_COEX: ADD_CHAR_EVT, status 0,  attr_handle 46, service_handle 44
I (1030) GATTC_GATTS_COEX: ESP_GAP_BLE_SCAN_START_COMPLETE_EVT, scan start success
I (1040) GATTC_GATTS_COEX: connect to the remote device PTRO0002
I (1040) GATTC_GATTS_COEX: ADD_DESCR_EVT, status 0, attr_handle 47, service_handle 44
I (1050) GATTC_GATTS_COEX: ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT, stop scan successfully
I (1090) GATTC_GATTS_COEX: ESP_GATTS_CONNECT_EVT, conn_id 0, remote cd:b9:e5:9a:2c:ef
I (1090) GATTC_GATTS_COEX: CONNECT_EVT, conn_id 0, remote cd:b9:e5:9a:2c:ef
I (1100) GATTC_GATTS_COEX: ESP_GATTS_MTU_EVT, MTU 64
I (1100) GATTC_GATTS_COEX: ESP_GATTS_MTU_EVT, MTU 64
I (1110) GATTC_GATTS_COEX: ESP_GATTC_CONNECT_EVT conn_id 0, if 5
I (1120) GATTC_GATTS_COEX: REMOTE BDA:
| (1120) GATTC_GATTS_COEX: cd b9 e5 9a 2c ef
I (1130) GATTC_GATTS_COEX: open success
I (1960) GATTC_GATTS_COEX: discover service complete conn_id 0
I (1960) GATTC_GATTS_COEX: Get service information from remote device
I (1960) GATTC_GATTS_COEX: ESP_GATTC_SEARCH_CMPL_EVT
I (1990) GATTC_GATTS_COEX: ESP_GATTC_CFG_MTU_EVT, Status 0, MTU 64, conn_id 0
I (6190) GATTC_GATTS_COEX: update connection params status = 0, min_int = 6, max_int = 6,conn_int = 6,latency = 0, timeout = 100

-> Smartphone connects:
I (128560) GATTC_GATTS_COEX: ESP_GATTS_CONNECT_EVT, conn_id 1, remote 58:00:8c:b3:25:ce
I (128560) GATTC_GATTS_COEX: CONNECT_EVT, conn_id 1, remote 58:00:8c:b3:25:ce
I (128560) GATTC_GATTS_COEX: ESP_GATTC_CONNECT_EVT conn_id 1, if 5
I (128570) GATTC_GATTS_COEX: REMOTE BDA:
I (128570) GATTC_GATTS_COEX: 58 00 8c b3 25 ce
I (129050) GATTC_GATTS_COEX: update connection params status = 0, min_int = 0, max_int = 0,conn_int = 6,latency = 0, timeout = 500
I (129430) GATTC_GATTS_COEX: update connection params status = 0, min_int = 0, max_int = 0,conn_int = 39,latency = 0, timeout = 500

### Expected Behavior
The connect should only cause a  ESP_GATTC_CONNECT_EVT event or ESP_GATTS_CONNECT_EVT depending if the client connects to a server or the smartphone connects to the server running on the esp.
### Actual Behavior
The client connect causes a  ESP_GATTC_CONNECT_EVT and a ESP_GATTS_CONNECT_EVT event.
The server connect causes a  ESP_GATTC_CONNECT_EVT and a ESP_GATTS_CONNECT_EVT event.
### Steps to repropduce

use the gattc_gattc_coex example from IDF 4.0-beta1 and modify it so it connects to a existing BLE server.
-> Observe the output.
Connect to it from a smartphone
-> Observe the output.

### Code to reproduce this issue
use the gattc_gattc_coex example from IDF 4.0-beta1

Alvin1Zhang, 2019-10-30:
@domonoky Thanks for reporting, we will look into. Thanks.

xiewenxiang, 2020-03-18:
Thank you very much for your suggestions. 

We will discuss whether it is more appropriate to do so.

Currently, you can do some checking at the application layer.

paddyze, 2020-04-02:
Has anyone made any progress with this? I'm getting the same issue, both events occur when a a remote client connects to the server. 
v4.1-dev-369-g4dac7c7df

ROYMURLI, 2020-04-16:
I am also facing this issue. Please let me know.
IDF- v4.1-dev-1086-g93a8603c5

domonoky, 2020-04-16:
You can work around this issue by checking for the bluetooth address in the application code. Your gattc code should know to which gatt server it currently wants to connect to (as you have to issue the connect command). So if you get a GATTC_CONNECT event you can check if it is the correct BLE address and if not, ignore the event. 
Similar for the GATTS_CONNECT event, if it is the ble address of the server your client code currently wants to connect -> ignore the event. 
You have to do the same thing for the disconnect events. 
Its not pretty, but it works :-)

sylque, 2024-11-06:
In complement to what @domonoky said, you can also rely on the `conn_id`:
1. When first receiving a GATTC_CONNECT or GATTS_CONNECT event, store the provided `conn_id` param.
2. Ignore subsequent GATTC_CONNECT or GATTS_CONNECT events if they relate to the same `conn_id`

It would be nice to have a clean solution, though.

sylque, 2024-11-25:
Is this issue being followed in any way at Espressif?

esp-cjh, 2024-11-26:
@sylque 
This is how Bluedroid's implementation works. It is not a issue. when a device acts as both a server and a client, both GATTS_CONNECT_EVT and GATTC_CONNECT_EVT will be reported whenever the device is connected. However, the params of events will include the peer device address and the conn_id, which can be used for identification.
The same applies when the connection is disconnected.

HeiderSati, 2024-12-11:
Hi @sylque:

I am having the same issue, the solution you mentioned does not work mainly because when I connect from my iPhone into a GATTS only, I get 2 lines saying the exact same thing as follows:

GATT-C: Connected ID: 00000000, Handle: 00000001, From MAC: 6C:EB:53:94:7E:17
GATT-S: Connected ID: 00000000, Handle: 00000001, From MAC: 6C:EB:53:94:7E:17

As you can see from the above, conn_id, Handle and MAC is the exact same for both. 

This makes no sense at all, it should be:
- GATT-C triggered *only* when I connect to GATT-C, or:
- GATT-S triggered *only* when I connect to GATT-S

I can't be both as it's incorrect.

If this is (as you say) the way it's meant to be, then surely there has to be somehow a parameter/attribute/value/etc that should tell us which one to ignore, I looked and debugged, cannot find anything on the object when the event is received. 

Personally I would suggest this is surely a bug and would be great if it could be solved by isolating the two,... if that's too late and you probably worry about maintaining compatibility if others developed based on that concept then please include something in the CONFIG object to change the behaviour to split the GATT-C from the GATT-S when a connection comes in to one of them. 

I hope I am making sense.

Many thanks
Heider

esp-cjh, 2024-12-11:
@HeiderSati
The cause of this issue is that you have registered both the GATTS app and the GATTC app. As a result, when the device is connected, ESP_GATTS_CONNECT_EVT and ESP_GATTC_CONNECT_EVT events will be reported.
If you only want to act as a GATT server, you only need to register the GATTS app. Please check your code to see if `esp_ble_gattc_app_register` is being called to register the GATTC app. If it is, please remove it.

HeiderSati, 2024-12-11:
Thank you esp-cjh

In my code I have:

1) esp_ble_gatts_app_register
2) esp_ble_gattc_app_register

I followed your advice and disabled (2) above, I now get only the C when connecting to C, so this looks promising. 

I will continue to test and get back to you if there are any issues with that. 

My system has 2 x profiles a (C) and and (S) and both should be connectable (and advertised) to be seen from a device, I have implemented the S adverts so far, will be doing C soon, I do hope this works out in the end.

I will keep you updated.

Many thanks again

Cheers,
Heider.