espressif/esp-idf #4263: Wifi cannot connect with BLK3 or Custom Mac Address

filzek, 2019-10-29:
mode:DIO, clock div:200,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
load:0x3fff0018,len:4
load:0x3fff001c,len:6916
load:0x40078000,len:14228
ho 0 tail 12 room 4
load:0x40080400,len:4296
entry 0x400806e4
I (71) boot: Chip Revision: 1
I (71) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot: ESP-IDF v4.1-dev-592-g758db1e00-dirty 2nd stage bootloader
I (40) boot: compile time 17:35:46
I (41) boot: Enabling RNG early entropy source...
I (47) boot: SPI Speed      : 40MHz
I (51) boot: SPI Mode       : DIO
I (55) boot: SPI Flash Size : 2MB
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x231c4 (143812) map
I (165) esp_image: segment 1: paddr=0x000331ec vaddr=0x3ffbdb60 size=0x0380c ( 14348) load
I (171) esp_image: segment 2: paddr=0x00036a00 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at C:/Users/filze/Desktop/esp-idf/components/freertos/xtensa_vectors.S:1778

I (172) esp_image: segment 3: paddr=0x00036e08 vaddr=0x40080400 size=0x09208 ( 37384) load
I (196) esp_image: segment 4: paddr=0x00040018 vaddr=0x400d0018 size=0xa2688 (665224) map
0x400d0018: _stext at ??:?

I (436) esp_image: segment 5: paddr=0x000e26a8 vaddr=0x40089608 size=0x0dd40 ( 56640) load
0x40089608: spi_flash_munmap at C:/Users/filze/Desktop/esp-idf/components/spi_flash/flash_mmap.c:261 (discriminator 3)

I (475) boot: Loaded app from partition at offset 0x10000
I (475) boot: Disabling RNG early entropy source...
I (475) cpu_start: Pro cpu up.
I (479) cpu_start: Application information:
I (484) cpu_start: Project name:     blufi_demo
I (489) cpu_start: App version:      v4.1-dev-592-g758db1e00-dirty
I (496) cpu_start: Compile time:     Oct 28 2019 17:35:15
I (502) cpu_start: ELF file SHA256:  becd0bece4db7038...
I (508) cpu_start: ESP-IDF:          v4.1-dev-592-g758db1e00-dirty
I (515) cpu_start: Starting app cpu, entry point is 0x4008128c
0x4008128c: call_start_cpu1 at C:/Users/filze/Desktop/esp-idf/components/esp32/cpu_start.c:276

I (0) cpu_start: App cpu up.
I (525) heap_init: Initializing. RAM available for dynamic allocation:
I (532) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (538) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (544) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (550) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (556) heap_init: At 3FFCC378 len 00013C88 (79 KiB): DRAM
I (562) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (569) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (575) heap_init: At 40097348 len 00008CB8 (35 KiB): IRAM
I (581) cpu_start: Pro cpu start user code
I (600) spi_flash: detected chip: generic
I (600) spi_flash: flash io: dio
W (601) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (611) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (622) WiseHome Switch Beta: Base MAC Address read from EFUSE BLK3
I (632) WiseHome Switch Beta: Using "0x49, 0x4f, 0x54, 0x0, 0x1, 0x31" as base MAC address
I (642) WIFI_STA MAC: 0x49, 0x4f, 0x54, 0x0, 0x1, 0x31
I (642) SoftAP MAC: 0x49, 0x4f, 0x54, 0x0, 0x1, 0x32
I (652) BT MAC: 0x49, 0x4f, 0x54, 0x0, 0x1, 0x33
I (652) Ethernet MAC: 0x49, 0x4f, 0x54, 0x0, 0x1, 0x34
I (692) wifi: wifi driver task: 3ffcf780, prio:23, stack:3584, core=0
I (802) wifi: wifi firmware version: 65d40b6
[... 124 lines cut ...]
    //Get MAC address for WiFi Station interface
    ESP_ERROR_CHECK(esp_read_mac(derived_mac_addr, ESP_MAC_WIFI_STA));
    ESP_LOGI("WIFI_STA MAC", "0x%x, 0x%x, 0x%x, 0x%x, 0x%x, 0x%x",
             derived_mac_addr[0], derived_mac_addr[1], derived_mac_addr[2],
             derived_mac_addr[3], derived_mac_addr[4], derived_mac_addr[5]);

    //Get MAC address for SoftAp interface
    ESP_ERROR_CHECK(esp_read_mac(derived_mac_addr, ESP_MAC_WIFI_SOFTAP));
    ESP_LOGI("SoftAP MAC", "0x%x, 0x%x, 0x%x, 0x%x, 0x%x, 0x%x",
             derived_mac_addr[0], derived_mac_addr[1], derived_mac_addr[2],
             derived_mac_addr[3], derived_mac_addr[4], derived_mac_addr[5]);

    //Get MAC address for Bluetooth
    ESP_ERROR_CHECK(esp_read_mac(derived_mac_addr, ESP_MAC_BT));
    ESP_LOGI("BT MAC", "0x%x, 0x%x, 0x%x, 0x%x, 0x%x, 0x%x",
             derived_mac_addr[0], derived_mac_addr[1], derived_mac_addr[2],
             derived_mac_addr[3], derived_mac_addr[4], derived_mac_addr[5]);

    //Get MAC address for Ethernet
    ESP_ERROR_CHECK(esp_read_mac(derived_mac_addr, ESP_MAC_ETH));
    ESP_LOGI("Ethernet MAC", "0x%x, 0x%x, 0x%x, 0x%x, 0x%x, 0x%x",
             derived_mac_addr[0], derived_mac_addr[1], derived_mac_addr[2],
             derived_mac_addr[3], derived_mac_addr[4], derived_mac_addr[5]);

Wifi simple doesnt work, there is a WIFI problem with BLK3 custom mac address

negativekelvin, 2019-10-30:
You can't set the multicast/group bit so your first byte could be 0x48 but not 0x49

projectgus, 2019-10-30:
Hi @filzek ,

@negativekelvin beat me to answering by 3 minutes, it seems. :)

The `esp_base_mac_addr_set` function should fail in this case (invalid MAC) with an error, we'll fix this.

Angus

filzek, 2019-10-30:
Okay, the efuse has a valid mac address, and the code seems valid to run, but wifi simple doesn't work.

@projectgus and @negativekelvin so the problem is with the wifi interface?

projectgus, 2019-10-30:
@filzek The MAC `49:4f:54:00:01:31` is a multicast MAC address. So it's not a valid unicast MAC address. This is why it doesn't work, the Wi-Fi library is trying to use it but the AP is correctly rejecting it.

To make it into a unicast MAC then you need to clear bit 1, changing it to `48:xx:xx:xx:xx:xx`

BTW, because `49:4f:54` isn't a valid OUI then to "do the right thing" you should also set bit 2 indicating it's a "locally administered" MAC, which would make it `4a:xx:xx:xx:xx:xx`.

If you have already saved this MAC in efuse then you can change it in code after you read it but before you call `esp_base_mac_addr_set`:

```
base_mac_addr[0] &= ~0x01; // make unicast
base_mac_addr[0] |= 0x02; // mark as locally administered
esp_base_mac_addr_set(base_mac_addr);
```

filzek, 2019-10-30:
Sure, I have no Idea why we didn't see it before, the lab testing was mean to be 49:4f:54 to hex to text IOT, but we forgot about multicast address.

Anyway, we found one brick in the wall, so, let's fix multicast address in interfaces setup.

closed 2019-10-30 by filzek