espressif/esp-idf #4264: Some examples have wrong I2S configuration, causing distortion on I2S DACs.

l0ud, 2019-10-29:
Problem: **i2s_adc_dac, a2dp_gatts_coex, a2dp_sink** examples do not set **I2S_COMM_FORMAT_I2S** bit in i2s configuration, even when they are setting **I2S_COMM_FORMAT_I2S_MSB** bit. This makes tx_msb_shift bit not being set in HW, causing i2s work in Philips mode, which is not supported by most i2s dacs and will cause distortions on higher volume.  

I observed it on a2dp_sink example and pcm5102 dac. Other i2s dacs are probably affected too.

Fix is simple, just find all occurrences of `I2S_COMM_FORMAT_I2S_MSB` and add `I2S_COMM_FORMAT_I2S | ` when it is missing. Or change i2s configuration logic to work correctly even when I2S_COMM_FORMAT_I2S bit is not set.

xiongyumail, 2019-11-06:
Thank you for your discovery. We have fixed this problem in the I2S source code. Merge in later.

HandcraftCavon, 2019-12-03:
Is this why I can't control volume with my iPhone, and it's sooooo loud that the speaker distorted? Can't wait to try it out!

cavonlee, 2019-12-03:
I tried, and the high volume part fix, thank you, but not the volume control part.

Alvin1Zhang, 2020-05-12:
@cavonlee Thanks for sharing the updates, the fix has been already on GitHub, feel free to reopen if the issue still happens. Thanks.

closed 2020-05-12 by Alvin1Zhang

mooalot, 2020-05-12:
> I tried, and the high volume part fix, thank you, but not the volume control part.

Yeah the volume control is not implemented in the code. I had the same problem, but I wrote some code adjust the volume using my phone. You have to modify the PCM data when it is received through the a2dp_sink callback functions.

Lukanite, 2020-07-07:
Just built the A2DP sink example to find this issue still exists on master (625bd5e), so I just wanted to leave a comment here if someone else runs into it. The suggested fix from @l0ud is now deprecated; what worked for my PCM5102 was setting .communication_format to I2S_COMM_FORMAT_STAND_I2S.

toybuilder, 2023-07-05:
Just adding a comment here to help clear up a possible misunderstanding --
The I2S Standard is the Philips standard.
See:
  A) Figure 1. I2S Audio Interface Timing Diagram (MAX98357A) versus Figure 2. Left-Justified Audio Interface Timing Diagram (MAX98357B) in the MAX98357A/MAX98357B datasheet
or
  B) Figure 13. Left Justified Audio Data Format versus Figure 14. I2S Audio Data Format in the PCM5102A datasheet.  (Figures 14 and 15 on the PCM5102 datasheet.)

The I2S alignment standard has the LRCLK preceding the MSB by one clock cycle.  
The MSB alignment standard has the LRCLK aligned with the MSB.

The PCM5102/5102A, in particular, support both modes depending on the setting of the FMT pin -- so it's important to confirm you are running in the correct mode for the chip/config in use.