espressif/esp-idf #4266: How to print std::string

dssatech, 2019-10-30:
## Environment
- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32U]

the following commands do NOT work on my pc !!!
What shell do these commands run in ?

- IDF version (run ``git describe --tags`` to find it):   ### Doesnt work on my pc ###
 - Build System:         [Make|CMake]     ### ???  Arduino 1.8.10 ###
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): ### Doesnt work on my pc ###
- Operating System:     [Windows10]
- Power Supply:         [USB]

## Problem Description

I used the std::string definitions provided in the "ESP32 BLE Arduino" library.
I am totally confused by your typedef  std::string

The ONLY way to print out the contents of a std::string  seems to be as follows :-
std::string  MyCustName = "MyName";
//
Serial.printf("Name= ");
for (uint8_t i = 0; i < device.length(); i++) {
	Serial.printf("%c",device[i]);
}
Serial.println();

Where can I find the details of the std::string functions ?

Can you show me the correct way to do std::string functions :- 
Ie 
How do I copy the std::string value to a "char *"  string ?
Eg  char * str = MyCustName ;

How do I set the std::string value from a "char *" string ?
Eg  std::string  MyCustName = str;

How do I concatenate a std::string value from a "char *" string?
Eg  std::string  MyCustName = MyCustName  + str;

How do I print the std::string value, or send to Serial port ?
Eg  MyCustName.print();   or Serial.print(MyCustName);

### Expected Behavior
I would assume these functions behave like stings, but they dont.

### Actual Behavior
Uncertain and wrong output.

### Steps to repropduce
implement below code.

### Code to reproduce this issue
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>
std::string  MyCustName;
char * Name = "MyName";
// Eg 1 - this prints rubbish
MyCustName = Name;
Serial.println(MyCustName);

// Eg 2 - this prints Ok
MyCustName = "MyName";
Serial.println(MyCustName);

## Debug Logs
the code compiles with no errors, but prints corrupted data, Ie not the contents of the string.

## Other items if possible

negativekelvin, 2019-10-30:
1. Don't post general programming questions here
2. For arduino specific issues use https://github.com/espressif/arduino-esp32
3. For how to questions use esp32.com
4. Google and stackoverflow are your friends

https://stackoverflow.com/questions/7416445/what-is-use-of-c-str-function-in-c

Alan5142, 2019-11-01:
Just use [c_str](https://en.cppreference.com/w/cpp/string/basic_string/c_str) and [printf's string format](https://en.cppreference.com/w/cpp/io/c/fprintf)

Alvin1Zhang, 2019-11-05:
@negativekelvin Thanks for the notes. @dssatech Thanks for reporting, for "how to" questions, you could submit a post on our forum at https://esp32.com. Thanks.

closed 2019-11-05 by Alvin1Zhang