espressif/esp-idf #4269: when is void Esp32MQTTClient_Close(void) or Esp32MQTTClient_Reset(void) going to work?

mpinkus, 2019-10-31:
is there a working version of mqtt anywhere? is anybody supporting any of these sdk's?

don't point me to another web page..respond with a concise example of restarting wifi and having the sntp connection surviving this most basic scenario...

Error: Time:Thu Oct 31 18:07:24 2019 File:C:\Users\\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\AzureIoT\src\az_iot\c-utility\pal\src\tlsio_openssl_compact.c Func:tlsio_openssl_destroy Line:213 tlsio_openssl_destroy called while not in TLSIO_STATE_CLOSED.

Info: Initializing SNTP

assertion "Operating mode must not be set while SNTP client is running" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/apps/sntp/sntp.c", line 600, function: sntp_setoperatingmode
abort() was called at PC 0x4012d35f on core 1

mpinkus, 2019-10-31:
ok, confused groups...assuming idf, how can the mqtt connection be reset in same situation, current behavior is it take x seconds to reconnect, but this is to long for a lost connection scenario...there must be a way to reset or reinit the mqtt connection when wifi is lost and returned

david-cermak, 2019-11-01:
Hi @mpinkus 

You can call `esp_mqtt_client_reconnect()` to force immediate reconnection whenever you receive a [disconnection event](https://github.com/espressif/esp-idf/blob/master/examples/protocols/mqtt/ssl/main/app_main.c#L67)

Unfortunately the mqtt library doesn't support configurable reconnection timeout as of now, it's currently hardcoded to 10 seconds (Might be a good idea to add this value to client config or sdkconfig)

david-cermak, 2020-01-17:
Hi @mpinkus 

Any update? the reconnection timeout is now a configuration parameter, added in this commit: https://github.com/espressif/esp-mqtt/commit/3da472fd37dd68604f0699d1a44a742cc3faaaee

Could you please close this issue or do share an update?

Alvin1Zhang, 2020-03-03:
@mpinkus Thanks for reporting. Will close due to short of feedback, feel free to reopen with more details provided. Thanks.

closed 2020-03-03 by Alvin1Zhang