espressif/esp-idf #4270: Update SDK fail xtensa-esp32s2-elf has no installed versions

filzek, 2019-10-31:
Adding ESP-IDF tools to PATH...
WARNING: tool xtensa-esp32s2-elf version esp32s2-dev-4-g3a626e9-8.2.0 is installed, but has reported version esp32s2-dev-4-g3a626e93-8.2.0
ERROR: tool xtensa-esp32s2-elf has no installed versions. Please run 'python.exe idf_tools.py install' to install it.

its cyclical,  even running export.bat or install.bat keep eternal as this, need to update install and setup files to fix it, but didnt work so far.

dobairoland, 2019-10-31:
The following patch should solve the issue until we push the fix to the `master` branch.

[win.zip](https://github.com/espressif/esp-idf/files/3795648/win.zip)

I'm sorry for the inconvenience.

dobairoland, 2019-11-01:
Just for future reference, commit bdb08de87555bd070905d416840d46405ead1ee0 will solve this issue.

lbaerd, 2019-11-05:
Can any confirmed that the patch file solve this issue? Thanks alot

pilnikov, 2019-11-05:
The error was corrected. Everything worked. But ... It is difficult to understand what and where to change. Maybe it was easier to lay out the corrected tools.json file?

tajallipour, 2019-11-05:
May I know how to apply the patch?

pilnikov, 2019-11-05:
you must replace the file /esp-idf/tools/tools.json 
[tools.zip](https://github.com/espressif/esp-idf/files/3807321/tools.zip)

tajallipour, 2019-11-05:
Perfect, worked.

> 
> 
> you must replace the file /esp-idf/tools/tools.json
> [tools.zip](https://github.com/espressif/esp-idf/files/3807321/tools.zip)

perfect, worked

dobairoland, 2019-11-05:
I updated my original comment with the command for applying the patch.

lbaerd, 2019-11-05:
Wow thanks. It works now...

HemanthSingh, 2019-11-05:
Hi I have the sme problem and this patch doesnt work on linux

Adding ESP-IDF tools to PATH...
WARNING: tool xtensa-esp32s2-elf version esp32s2-dev-4-g3a626e93-8.2.0 is installed, but has reported version esp32s2-dev-4-g3a626e9-8.2.0
ERROR: tool xtensa-esp32s2-elf has no installed versions. Please run '/mnt/c/Users/sownet/Desktop/esp-idf/install.sh' to install it

dobairoland, 2019-11-05:
@HemanthSingh This patch is for Windows and Mac. What is your architecture (amd64/i686)? Please give the full log of the install script.

igrr, 2019-11-05:
Macos/windows issue should be fixed by https://github.com/espressif/esp-idf/commit/bdb08de87555bd070905d416840d46405ead1ee0. Please open a new issue if you are still seeing the error.

Edit: Sorry, the master branch hasn't been synced yet. Keeping open.

HemanthSingh, 2019-11-05:
> @HemanthSingh This patch is for Windows and Mac. What is your architecture (amd64/i686)? Please give the full log of the install script.

Hi, Sorry this is full install log:
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp32-2019r1-8.2.0 (already installed)
WARNING: tool xtensa-esp32s2-elf version esp32s2-dev-4-g3a626e93-8.2.0 is installed, but has reported version esp32s2-dev-4-g3a626e9-8.2.0
Installing xtensa-esp32s2-elf@esp32s2-dev-4-g3a626e93-8.2.0
file xtensa-esp32s2-elf-gcc8_2_0-esp32s2-dev-4-g3a626e-linux-amd64.tar.gz is already downloaded
WARNING: destination path already exists, removing
Extracting /home/sownet/.espressif/dist/xtensa-esp32s2-elf-gcc8_2_0-esp32s2-dev-4-g3a626e-linux-amd64.tar.gz to /home/sownet/.espressif/tools/xtensa-esp32s2-elf/esp32s2-dev-4-g3a626e93-8.2.0
Skipping esp32ulp-elf@2.28.51.20170517 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20190708 (already installed)
Installing Python environment and packages
Creating a new Python environment in /home/sownet/.espressif/python_env/idf4.1_py2.7_env
Installing virtualenv
Collecting virtualenv .. ..

HemanthSingh, 2019-11-05:
> @HemanthSingh This patch is for Windows and Mac. What is your architecture (amd64/i686)? Please give the full log of the install script.

It is x86_64

keelung-yang, 2019-11-08:
> > @HemanthSingh This patch is for Windows and Mac. What is your architecture (amd64/i686)? Please give the full log of the install script.
> 
> It is x86_64

Me too on Win10

pilnikov, 2019-11-08:
> > > @HemanthSingh This patch is for Windows and Mac. What is your architecture (amd64/i686)? Please give the full log of the install script.
> > 
> > 
> > It is x86_64
> 
> Me too on Win10

you must replace the file /esp-idf/tools/tools.json
tools.zip

keelung-yang, 2019-11-08:
> > > > @HemanthSingh This patch is for Windows and Mac. What is your architecture (amd64/i686)? Please give the full log of the install script.
> > > 
> > > 
> > > It is x86_64
> > 
> > 
> > Me too on Win10
> 
> > you must replace the file /esp-idf/tools/tools.json
> > [tools.zip](https://github.com/espressif/esp-idf/files/3807321/tools.zip)

Worked! Thanks!

liydu, 2019-11-08:
Any workarounds or updates for Linux?

dobairoland, 2019-11-08:
For Linux users: Have you using the toolchain for `esp32s2beta` from the preview branch before it was merged into `master`? Otherwise I cannot explain that you experience these issues.

Please try to remove the installer from the `.espressif/dist` directory, so the installer would download it again. Another thing you can try is to remove the installed tool from `.espressif/tools` directory as well and after that run the installer again.

```bash
rm ~/.espressif/dist/xtensa-esp32s2-elf*
rm -r ~/.espressif/tools/xtensa-esp32s2-elf/
```

dobairoland, 2019-11-14:
The fix (bdb08de87555bd070905d416840d46405ead1ee0) is now available in the master branch.

closed 2019-11-14 by dobairoland

Maldus512, 2019-11-14:
> Adding ESP-IDF tools to PATH...
WARNING: tool xtensa-esp32s2-elf version esp32s2-dev-4-g3a626e93-8.2.0 is installed, but has reported version esp32s2-dev-4-g3a626e9-8.2.0
ERROR: tool xtensa-esp32s2-elf has no installed versions. Please run '/mnt/c/Users/sownet/Desktop/esp-idf/install.sh' to install it

I'm having the same issue on linux-arm64 (note that I added arm64 references in tools.json to download the armel version which works regardless). The installation completes successfully, but I find this error when sourcing `export.sh`

dobairoland, 2019-11-14:
@Maldus512 Note that there are two different blocks: one for `esp32s2-dev-4-g3a626e9-8.2.0` and one for `esp32s2-dev-4-g3a626e93-8.2.0`.

https://github.com/espressif/esp-idf/blob/bdb08de87555bd070905d416840d46405ead1ee0/tools/tools.json#L79-L97

https://github.com/espressif/esp-idf/blob/bdb08de87555bd070905d416840d46405ead1ee0/tools/tools.json#L98-L115

Also note that now the toolchain has been updated to `esp-2019r2-8.2.0` where should be no problem with the versions.

Maldus512, 2019-11-14:
> Also note that now the toolchain has been updated to esp-2019r2-8.2.0 where should be no problem with the versions.

True, I didn't notice that. With the latest master now there is no more version discrepancy, now exporting the environment fails complaining that openocd-esp32 is not installed

```
$ . export.sh
Adding ESP-IDF tools to PATH...
WARNING: directory for tool openocd-esp32 version v0.10.0-esp32-20190708 is present, but tool was not found
ERROR: tool openocd-esp32 has no installed versions. Please run '/home/maldus/Source/esp-idf/install.sh' to install it
```

`openocd-esp32` is indeed installed at `~/.espressif/tools/openocd-esp32/`

dobairoland, 2019-11-14:
@Maldus512 But can it be run?
```sh
~/.espressif/tools/openocd-esp32/v0.10.0-esp32-20190708/openocd-esp32/bin/openocd --version
```

igrr, 2019-11-14:
If it doesn't, you may need to install libusb (https://github.com/espressif/esp-idf/issues/4095)

Maldus512, 2019-11-15:
I realized only now it would not run because it relies on `/lib/ld-linux.so.3` which was missing from my system (I had `/lib/ld-linux-armhf.so.3`). After creating the symlink it works. Many thanks for the assistance!

mateusreyes, 2020-04-26:
Hi experts, same behaviral here when trying to install the SDK into debian:10.3-slim (container).

apt-get update && apt-get install git wget libncurses-dev flex bison gperf python python-pip python-setuptools python-serial python-click python-cryptography python-future python-pyparsing python-pyelftools cmake ninja-build ccache libffi-dev libssl-dev

mkdir /root/esp
cd /root/esp

# release/v4.2 openocd bug: "ERROR: tool openocd-esp32 has no installed versions." moving to master branch

git clone --recursive https://github.com/espressif/esp-idf.git

cd esp-idf

git submodule update --init --recursive

./install.sh

. ./export.sh

####### ERROR

Setting IDF_PATH to '/root/esp/esp-idf'
Adding ESP-IDF tools to PATH...
WARNING: tool openocd-esp32 version v0.10.0-esp32-20200420 is installed, but the tool failed to run
**ERROR: tool openocd-esp32 has no installed versions. Please run '/root/esp/esp-idf/install.sh' to install it.**

Any light on this please ? thanks a lot!!

igrr, 2020-04-26:
Can you please try installing libusb using your OS package manager?

mateusreyes, 2020-04-27:
> Can you please try installing libusb using your OS package manager?

Worked! thanks a lot @igrr

app-z, 2020-04-29:
Ubuntu 18.04
It still reproduce in latest esp-adf/esp-idf

I download openocd-esp32 in folder tools/

wget https://github.com/espressif/openocd-esp32/releases/download/v0.10.0-esp32-20190313/openocd-esp32-linux64-0.10.0-esp32-20190313.tar.gz
tar -xvzf openocd-esp32-linux64-0.10.0-esp32-20190313.tar.gz

But for run command:  tools/openocd-esp32/v0.10.0-esp32-20190708/openocd-esp32/bin/openocd --version

There was error: error while loading shared libraries: libusb-1.0.so.0: cannot open shared object file: No such file or directory

Its help me:
apt install libusb-1.0-0

Also it connected with that issue:
https://github.com/espressif/esp-idf/issues/3740