espressif/esp-idf #4271: Max TCB for gatt_if reached

wcauchois, 2019-10-31:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): unknown
- Module or chip used:  ESP32-SOLO-1
- IDF version (run ``git describe --tags`` to find it): 
    // `v3.3`
- Build System: CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System: macOS
- Power Supply: USB

## Problem Description

I have an application that's acting as a BLE GATT client and connecting to up to 3 peripherals. Sometimes I get the following log message:

```
E (614233) BT_GATT: Max TCB for gatt_if [5] reached.
```

Where `5` is one of the client interfaces. After this, GATTC opens on that interface fail repeatedly. Does anyone know what this error message means or how I can investigate this further?

Alvin1Zhang, 2019-11-01:
@wcauchois Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, and etc. would help us debug further. Thanks.

wcauchois, 2019-11-01:
Thanks for the reply @Alvin1Zhang. Here's my sdkconfig and some more comprehensive logs: https://gist.github.com/wcauchois/f0d0df5d26c19f24b895b1c5d1f84936

To give some color to the logs: my code is pretty similar to the [GATT client tutorial](https://github.com/espressif/esp-idf/blob/2e6398affaeeac2f7ce40457a881f2dda57ad11f/examples/bluetooth/bluedroid/ble/gatt_client/tutorial/Gatt_Client_Example_Walkthrough.md), wherein I maintain a profile_tab with the connection ID. To sanity check things, I have those `STATUS` logs which just print out the profile tab gatt IFs and conn IDs every second.

GYC-Espressif, 2020-01-12:
Hello. Sorry for the late reply. From your sdkconfig and logs, I found you try to connect more than four devices but config in menuconfig is just for four, so you can not connect the device and got the error log "Max TCB for gatt_if reached". If you want to connect more than four devices you need to change the menuconfig of both host option `→ Component config → Bluetooth → Bluedroid Options → BT/BLE MAX ACL CONNECTIONS(1~7)`and controller option `→ Component config → Bluetooth → Bluetooth controller → BLE Max Connections`.

Alvin1Zhang, 2020-04-26:
@wcauchois Thanks for reporting, will close due to short of feedback, feel free to reopen with more details. Thanks.

closed 2020-04-26 by Alvin1Zhang