espressif/esp-idf #4272: coap_mutex.h seems to be missing, included from coap_debug.c

TuxRobotics, 2019-10-31:
## Environment

- Development Kit: none
- Module or chip used:  ESP32-S
- IDF version: v4.1-dev-815-ga45e998
- Build System: CMake
- Compiler version: 8.2.0
- Operating System: Linux
- Power Supply: USB

## Problem Description
coap_mutex.h is required by coap_debug.c but does not seem to exist anywhere in the repository

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. fresh clone of master
2. build

Result:
coap_debug.c:469:10: error: unknown type name 'coap_mutex_t'
   static coap_mutex_t static_show_pdu_mutex = COAP_MUTEX_INITIALIZER;

mahavirj, 2019-11-01:
@TuxRobotics It should be available at `./components/coap/libcoap/include/coap2/coap_mutex.h`, did you update submodule pointer for `libcoap`?

mahavirj, 2019-11-05:
@TuxRobotics Do you have any update?

tajallipour, 2019-11-05:
the same here, cant find the coap_mutex.h

mahavirj, 2019-11-05:
@tajallipour 

Can you please confirm output of command `git submodule  | grep libcoap` showing: 

```
 98954eb30a2e728e172a6cd29430ae5bc999b585 components/coap/libcoap
```

Here is pointer to corresponding file from `libcoap` repository, https://github.com/obgm/libcoap/blob/develop/include/coap2/coap_mutex.h

tajallipour, 2019-11-05:
> @tajallipour
>
> Can you please confirm output of command `git submodule | grep libcoap` showing:
>
> ```
>  98954eb30a2e728e172a6cd29430ae5bc999b585 components/coap/libcoap
> ```
>
> Here is pointer to [...]

+cfec0d072c5b99ed3e54828ca50ea2f6b91e1f50 components/coap/libcoap (v4.2.0-rc4-35-gcfec0d0)

tajallipour, 2019-11-05:
Thank you, I just copied the file and compiler was able to finish the compile, with no issue. I am about to push it to esp32 chip.

mahavirj, 2019-11-05:
> +cfec0d072c5b99ed3e54828ca50ea2f6b91e1f50 components/coap/libcoap (v4.2.0-rc4-35-gcfec0d0)

@tajallipour You may update submodules using command `git submodule update --init --recursive`, that should help to fix this.

tajallipour, 2019-11-05:
i did submodule update and now mine is 98954eb30a2e728e172a6cd29430ae5bc999b585

TuxRobotics, 2019-11-07:
I removed the coap component a few days ago to keep moving forward, but a fresh clone seems to have resolved the issue.  My submodule is now at commit: 
`98954eb30a2e728e172a6cd29430ae5bc999b585 components/coap/libcoap (v4.2.0-rc4-114-g98954eb)`

closed 2019-11-08 by mahavirj