espressif/esp-idf #4278: Endless script loop when starting IDF command prompt

zinke-ct-video, 2019-11-04:
After updating my ESP-IDF via the commands "git pull" and "git submodule update --init --recursive" to commit a45e9985344575a80acb1dc9c12e7bec4d8af401, I can no longer open the IDF command prompt successfully.

The commant prompt prints upon opening:

> Using Python in C:\Program Files\Python37\
> Python 3.7.4
> Using Git in C:\Program Files\Git\bin
> git version 2.21.0.windows.1
> Setting IDF_PATH: C:\Users\user258\esp\esp-idf
> 
> Adding ESP-IDF tools to PATH...
> WARNING: tool xtensa-esp32s2-elf version esp32s2-dev-4-g3a626e9-8.2.0 is installed, but has reported version esp32s2-dev-4-g3a626e93-8.2.0
> **ERROR: tool xtensa-esp32s2-elf has no installed versions. Please run 'C:\Users\user258\esp\esp-idf\install.bat' to install it.**

So I executed the mentioned script, which gives me the following output:

C:\Users\user258\esp\esp-idf>C:\Users\user258\esp\esp-idf\install.bat
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, cmake, openocd-esp32, mconf, ninja, idf-exe, ccache
Skipping xtensa-esp32-elf@esp32-2019r1-8.2.0 (already installed)
WARNING: tool xtensa-esp32s2-elf version esp32s2-dev-4-g3a626e9-8.2.0 is installed, but has reported version esp32s2-dev-4-g3a626e93-8.2.0
Installing xtensa-esp32s2-elf@esp32s2-dev-4-g3a626e9-8.2.0
file xtensa-esp32s2-elf-gcc8_2_0-esp32s2-dev-4-g3a626e-win32.zip is already downloaded
WARNING: destination path already exists, removing
Extracting C:\Users\user258\.espressif\dist\xtensa-esp32s2-elf-gcc8_2_0-esp32s2-dev-4-g3a626e-win32.zip to C:\Users\user258\.espressif\tools\xtensa-esp32s2-elf\esp32s2-dev-4-g3a626e9-8.2.0
Skipping esp32ulp-elf@2.28.51.20170517 (already installed)
Skipping cmake@3.13.4 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20190708 (already installed)
Skipping mconf@v4.6.0.0-idf-20190628 (already installed)
Skipping ninja@1.9.0 (already installed)
Skipping idf-exe@1.0.1 (already installed)
Skipping ccache@3.7 (already installed)
Setting up Python environment
Installing Python packages from C:\Users\user258\esp\esp-idf\requirements.txt
Requirement already satisfied: setuptools in c:\users\user258\.espressif\python_env\idf4.1_py3.7_env\lib\site-packages (from -r C:\Users\user258\esp\esp-idf\requirements.txt (line 4)) (41.2.0)
Requirement already satisfied: click>=5.0 in c:\users\user258\.espressif\python_env\idf4.1_py3.7_env\lib\site-packages (from -r C:\Users\user258\esp\esp-idf\requirements.txt (line 8)) (7.0)
Requirement already satisfied: pyserial>=3.0 in c:\users\user258\.espressif\python_env\idf4.1_py3.7_env\lib\site-packages (from -r C:\Users\user258\esp\esp-idf\requirements.txt (line 9)) (3.4)
Requirement already satisfied: future>=0.15.2 in c:\users\user258\.espressif\python_env\idf4.1_py3.7_env\lib\site-packages (from -r C:\Users\user258\esp\esp-idf\requirements.txt (line 10)) (0.17.1)
Requirement already satisfied: cryptography>=2.1.4 in c:\users\user258\.espressif\python_env\idf4.1_py3.7_env\lib\site-packages (from -r C:\Users\user258\esp\esp-idf\requirements.txt (line 11)) (2.7)
Requirement already satisfied: pyparsing<2.4.0,>=2.0.3 in c:\users\user258\.espressif\python_env\idf4.1_py3.7_env\lib\site-packages (from -r C:\Users\user258\esp\esp-idf\requirements.txt (line 12)) (2.3.1)
Requirement already satisfied: pyelftools>=0.22 in c:\users\user258\.espressif\python_env\idf4.1_py3.7_env\lib\site-packages (from -r C:\Users\user258\esp\esp-idf\requirements.txt (line 13)) (0.25)
Requirement already satisfied: six>=1.4.1 in c:\users\user258\.espressif\python_env\idf4.1_py3.7_env\lib\site-packages (from cryptography>=2.1.4->-r C:\Users\user258\esp\esp-idf\requirements.txt (line 11)) (1.12.0)
Requirement already satisfied: asn1crypto>=0.21.0 in c:\users\user258\.espressif\python_env\idf4.1_py3.7_env\lib\site-packages (from cryptography>=2.1.4->-r C:\Users\user258\esp\esp-idf\requirements.txt (line 11)) (0.24.0)
Requirement already satisfied: cffi!=1.11.3,>=1.8 in c:\users\user258\.espressif\python_env\idf4.1_py3.7_env\lib\site-packages (from cryptography>=2.1.4->-r C:\Users\user258\esp\esp-idf\requirements.txt (line 11)) (1.12.3)
Requirement already satisfied: pycparser in c:\users\user258\.espressif\python_env\idf4.1_py3.7_env\lib\site-packages (from cffi!=1.11.3,>=1.8->cryptography>=2.1.4->-r C:\Users\user258\esp\esp-idf\requirements.txt (line 11)) (2.19)
WARNING: You are using pip version 19.2.3, however version 19.3.1 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.
All done! You can now run:
   export.bat

Then I typed "export.bat" and get the following error:

C:\Users\user258\esp\esp-idf>export.bat
Setting IDF_PATH: C:\Users\user258\esp\esp-idf

Adding ESP-IDF tools to PATH...
WARNING: tool xtensa-esp32s2-elf version esp32s2-dev-4-g3a626e9-8.2.0 is installed, but has reported version esp32s2-dev-4-g3a626e93-8.2.0
**ERROR: tool xtensa-esp32s2-elf has no installed versions. Please run 'C:\Users\user258\esp\esp-idf\install.bat'** to install it.

So we have an infinite loop here. Whan can I do about it?

dobairoland, 2019-11-04:
This is a duplicate of https://github.com/espressif/esp-idf/issues/4270.

Please see this post to resolve the issue: https://github.com/espressif/esp-idf/issues/4270#issuecomment-548549611

zinke-ct-video, 2019-11-04:
Ok, thank you. Then I close this issue now.

closed 2019-11-04 by zinke-ct-video