espressif/esp-idf #4279: 蓝牙mesh的返回状态的疑问

tykjdxjcxq, 2019-11-04:
活动名称ESP_BLE_MESH_MODEL_OPERATION_EVT操作码ESP_BLE_MESH_MODEL_OP_GEN_ONOFF_GET的回调函数

返回状态是用收到data数据中得到的这个 应该是在data是数据来源不应该是客户端吗
客户端发出的消息有再获得是不是写错了，还是在回调的时候已经读取gpio的值
static void gen_onoff_get_handler(esp_ble_mesh_model_t *model,
                                  esp_ble_mesh_msg_ctx_t *ctx,
                                  uint16_t length, uint8_t *data)
{
    struct _led_state *led = (struct _led_state *)model->user_data;
    uint8_t send_data;
    esp_err_t err;

    ESP_LOGI(TAG, "%s, addr 0x%04x onoff 0x%02x", __func__, model->element->element_addr, led->current);

    send_data = led->current;
    /* Send Generic OnOff Status as a response to Generic OnOff Get */
    err = esp_ble_mesh_server_model_send_msg(model, ctx, ESP_BLE_MESH_MODEL_OP_GEN_ONOFF_STATUS,
            sizeof(send_data), &send_data);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "%s: Generic OnOff Status failed", __func__);
        return;
    }
}

Campou, 2019-11-04:
Hi @tykjdxjcxq 

1. The **Generic OnOff Get** message from the Geneirc OnOff Client model has no parameters.
2. `led->current` is used to record the present onoff state value.
3. When the Generic OnOff Server model receives a **Generic OnOff Get** message, it will reply with **Generic OnOff Status** with the OnOff field set to its present onoff state value.

Thanks.

Alvin1Zhang, 2020-07-28:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more details/updates. Thanks.

closed 2020-07-28 by Alvin1Zhang