espressif/esp-idf #4281: No rule to make target 'menuconfig'. Stop.

saahilshroff, 2019-11-04:
Hi,
I have been following "https://exploreembedded.com/wiki/Setting_up_Bare_Metal_Development_Toolchain_for_ESP32" to be able to work on ESP-32 using Eclipse.
When I try running the "make menuconfig", it shows me the following error: "make: *** No rule to make target 'menuconfig'. Stop."

My IDF_PATH is "D:/SaahilESP/esp-idf".
Msys has been extracted in C drive.

How do I resolve this issue? Please help! TIA.

Alvin1Zhang, 2019-11-04:
@saahilshroff Thanks for reporting, and also saw you left a message https://github.com/espressif/esp-idf/issues/1703#issuecomment-549265126.

saahilshroff, 2019-11-04:
Correct! Could you please help me out?

saahilshroff, 2019-11-04:
@Alvin1Zhang, look at this: 
[image]

Dazza0, 2019-11-04:
* In your `msys32` directory, run `mingw32.exe` not `msys2.exe`
* Run `make menuconfig` in your project directory NOT in `esp-idf` directory
* If you get warnings about missing python dependencies, run `pip install -r $IDF_PATH/requirements.txt`

If you plan to use Make build system, I suggest you take a look at the [Legacy Getting Started Guide for GNU Make](https://docs.espressif.com/projects/esp-idf/en/latest/get-started-legacy/index.html). The version of ESP-IDF used in the link you've provided is release/v3.1 which is a relatively old version of ESP-IDF.

saahilshroff, 2019-11-04:
Hi @Dazza0 .

Thank you for replying. 
I could now successfully run the 'make menuconfig' and this is how I configured it. (The ESP-32 is connected to COM24)
[image]

Now, when I use "make" to compile this test code, the output on the screen: 
[image]

Similarly, when I use "make flash" and hold "BOOT" button while it flashes, the output on the screen:
[image]

The on-board LED doesn't glow. Could you please help me out with this as well?

[Note: The "blink" code is as follows:

#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/gpio.h"
#include "sdkconfig.h"

#define BLINK_GPIO 2

void blink_task(void *pvParameter)
{
     gpio_pad_select_gpio(BLINK_GPIO);
    gpio_set_direction(BLINK_GPIO, GPIO_MODE_OUTPUT);
    while(1) 
    {
        gpio_set_level(BLINK_GPIO, 0);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
  
        gpio_set_level(BLINK_GPIO, 1);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}

void app_main()
{
    xTaskCreate(&blink_task, "blink_task", 
    configMINIMAL_STACK_SIZE, NULL, 5, NULL);
}

]

Dazza0, 2019-11-04:
Won't be able to help unless I know which board/development kit you're using and which GPIOs are connected to which LED on the board. Have you tried changing `#define BLINK_GPIO 2` ?

saahilshroff, 2019-11-04:
I am using ESP-32.
Currently, I am just trying to blink the on-board LED, which is internally connected to GPIO 2.

Dazza0, 2019-11-04:
> I am using ESP-32
ESP32 is the chip in the board, not the board itself. I mean the actual development board you're using (e.g. WROVER Kit, Pico Kit). See [this link](https://www.espressif.com/en/products/hardware/development-boards) for all the official Espressif Boards.

It is possible that you may have also purchased a third party board (i.e., not made by Espressif). In that case, a link to the seller and their board schematic would he helpful.

saahilshroff, 2019-11-04:
Using ESP32-WROOM-32 (ESP32-DevKitC)

> ESP32 is the chip in the board, not the board itself. I mean the actual development board you're using (e.g. WROVER Kit, Pico Kit). See [this [...]

Dazza0, 2019-11-04:
> Using ESP32-WROOM-32 (ESP32-DevKitC)

There is no LED connected GPIO 2 on ESP32-DevkitC. Please refer to [the schematic](https://dl.espressif.com/dl/schematics/esp32_devkitc_v4-sch.pdf). The LED on  ESP32-DevkitC is connected to the power supply, not a GPIO.

saahilshroff, 2019-11-05:
Hi @Dazza0 .

I checked the schematic out and could not find an LED connected.
However, using Arduino, I was able to blink an onboard LED. Kindly refer below for the code and output. (I referred to _https://circuitdigest.com/microcontroller-projects/getting-started-with-esp32-with-arduino-ide_ when I was using Arduino IDE)

**CODE:** 
#define led 2

void setup() 
{
  pinMode(led,OUTPUT);
}
void loop() 
{
  digitalWrite(led,HIGH);
  delay(5000);
  digitalWrite(led,LOW);
  delay(500);
}

**OUTPUT** 
(It appears Green in the picture since Red&Blue LED are beside each other. Red indicates that the board is powered. Blue color signifies the onboard LED that I have been talking about)
[image]

Hence, "#define BLINK_GPIO 2" for the code that I am trying to flash on the board using the Make-Build system.

Dazza0, 2019-11-05:
That is not an ESP32-DevKitC. That is a third party board. Judging from the image, it looks like an HW-573 or NodeMCU. But if the works on Arduino, I guess there is indeed.

I've looked through the instructions in the [link you have provided](https://exploreembedded.com/wiki/Setting_up_Bare_Metal_Development_Toolchain_for_ESP32). Are you actually flashing you development board using `make flash` or `make flash monitor` commands?

If so and it's still not working, add a couple of `printf()` statements in your `blink_task()` to check if it's actually running.

saahilshroff, 2019-11-05:
Hey @Dazza0.

The "blink" project that I had downloaded as suggested by the [link that I was following](https://exploreembedded.com/wiki/Setting_up_Bare_Metal_Development_Toolchain_for_ESP32) didn't actually contain the code for blinking the LED on Pin-2.
Works fine now.
Kudos to you for the fast and patient replies. 
Closing this issue.
Thank you very much! :)

closed 2019-11-05 by saahilshroff

saahilshroff, 2019-11-07:
> Hi @Dazza0 .
>
> I checked the schematic out and could not find an LED connected.
> However, using Arduino, I was able to blink an onboard LED. Kindly refer below for the code and output. (I referred to [...]

Correction: The development board shown in the image is NodeMCU-32S.