espressif/esp-idf #4283: RS422 UART

lildent, 2019-11-04:
I've been looking through https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/uart.html

for support examples of using RS422 UART and I can not seem to find anything.

Has anyone else seen any examples?

Thanks in advance

negativekelvin, 2019-11-05:
Rs422 doesn't really need anything special it is just a uart with a differential phy.  There is an rs485 example here https://github.com/espressif/esp-idf/tree/master/examples/peripherals/uart/uart_echo_rs485

lildent, 2019-11-05:
thanks for that. am im able to use the ESP to calculate the differential or am i having to add an extra layer (chip) converter in the circuit ?

negativekelvin, 2019-11-05:
You need to use a transceiver chip

lildent, 2019-11-05:
thats awesome :) any recommendations?

alisitsyn, 2019-11-05:
Hi @lildent ,The RS422 is half duplex interface and has two differential lines instead of half duplex RS485 which has one differential line for send and transmit signal. You need specific line driver for this. For example SN65C1168E: http://www.ti.com/lit/ds/symlink/sn65c1168e-sep.pdf
In order to check interface you can use the uart_echo_rs485 example.

Alvin1Zhang, 2020-01-08:
@lildent Thanks for reporting. Would you please help check if any updates for the issue? Thanks.

lildent, 2020-01-08:
I manage to get a rs422 to TTL and got my issue working on the esp32 :) thanks

closed 2020-01-08 by lildent