espressif/esp-idf #4284: NimBLE - help with crash request

chegewara, 2019-11-04:
Hi,
i am working with AWSFreeRTOS and im having issue with NimBLE. When i am trying to release memory from BLE im getting this crash:
(CODE)
`ESP_ERROR_CHECK( esp_bt_controller_mem_release( ESP_BT_MODE_BLE ) );`
(CRASH)
```
ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x400d23a7
0x400d23a7: xBLEStackTeardown at /home/chegewara/upwork/leaksentinel2/build/../src/main.c:378 (discriminator 1)

file: "../src/main.c" line 378
func: xBLEStackTeardown
expression: esp_bt_controller_mem_release( ESP_BT_MODE_BLE )

Backtrace: 0x40091fec:0x3ffe3210 0x40092484:0x3ffe3230 0x400d23a7:0x3ffe3250 0x400d284e:0x3ffe3270 0x400d7cfb:0x3ffe3290 0x400d76d3:0x3ffe32b0 0x401484fa:0x3ffe32d0 0x40148f95:0x3ffe3320
0x40091fec: invoke_abort at /home/chegewara/upwork/leaksentinel2/build/../amazon-freertos/vendors/espressif/esp-idf/components/esp32/panic.c:676

0x40092484: _esp_error_check_failed at /home/chegewara/upwork/leaksentinel2/build/../amazon-freertos/vendors/espressif/esp-idf/components/esp32/panic.c:692

0x400d23a7: xBLEStackTeardown at /home/chegewara/upwork/leaksentinel2/build/../src/main.c:378 (discriminator 1)

0x400d284e: wifi_connectedCallback at /home/chegewara/upwork/leaksentinel2/build/../src/main.c:759

0x400d7cfb: _onNetworkStateChangeCallback at /home/chegewara/upwork/leaksentinel2/build/../amazon-freertos/demos/demo_runner/iot_demo_freertos.c:144

0x400d76d3: _dispatchNetworkStateChangeCB at /home/chegewara/upwork/leaksentinel2/build/../amazon-freertos/demos/network_manager/aws_iot_network_manager.c:926

0x401484fa: _taskPoolWorker at iot_taskpool.c:?

0x40148f95: _threadRoutineWrapper at iot_threads_freertos.c:?
```
I understand it is bluedroid function to release memory, but it is exist in AFR:
https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/aws_demos/application_code/main.c#L175-L194
which is, as i believe, ported by or at least consulted with espressif devs. 
@huguesBouvier

So my question is, how to unload NimBLE and release memory?

dhrishi, 2019-11-05:
@chegewara Please mention the sequence of steps/APIs that you are executing.

chegewara, 2019-11-05:
Its pretty simple, i have on wifi connect callback from which i am executing 
`ESP_ERROR_CHECK( esp_bt_controller_mem_release( ESP_BT_MODE_BLE ) );`
This can be reproduced with simple example from AFR master.

Also, its enough to release memory in this line
https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/aws_demos/application_code/main.c#L134

dhrishi, 2019-11-05:
@chegewara If I understand this correctly, you want to disable/deinit BLE once Wi-Fi connection is established. Before calling `ESP_ERROR_CHECK( esp_bt_controller_mem_release( ESP_BT_MODE_BLE ) );` did you call AFR handlers for that: `.pxDisable`, `.pxBtManagerCleanup`.

chegewara, 2019-11-05:
I did not, with release-1.5 and bluedroid i dont have too, or its my poor programming skills and lucky but it work with bluedroid.

Thanks @dhrishi 
i just thought i am missing something in NimBLE, and here is better support for it, or at least was once. Now bluetooth esp32 support is not so good.

dhrishi, 2019-11-05:
> I did not, with release-1.5 and bluedroid i dont have too, or its my poor programming skills and lucky but it work with bluedroid.

That's because, in case of Bluedroid in the function `xBLEStackTeardown` the host and controller is disabled and deinitialized and then memory is released. While in case of NimBLE you are directly trying to release memory before disabling and deinitializing controller (and hence error 0x103 = ESP_ERR_INVALID_STATE)

 Ideally, for Bluedroid, `xBLEStackTeardown` should be modified as internally its `.pxDisable`, `.pxBtManagerCleanup` handle the same (just like NimBLE)

chegewara, 2019-11-05:
Ok, now i found it:
```
BTStatus_t IotBle_Off( void )
{
....
    /* Currently Disabled due to a bug with ESP32 : https://github.com/espressif/esp-idf/issues/2070 */

    /* _BTInterface.p_BTInterface->pxDisable(); */
}
```
Thanks @dhrishi 
over a year and #2070 still open, sad

chegewara, 2019-11-05:
Ok, i uncommented `/* _BTInterface.p_BTInterface->pxDisable(); */`, now i am calling `IotBle_Off` and it works like intended.
Thank you @dhrishi 
If you think it can be closed, just let me know or feel free to close issue.

dhrishi, 2019-11-05:
@chegewara In addition to `pxDisable`, `pxBtManagerCleanup` should also be called as this does `esp_nimble_hci_and_controller_deinit`

closed 2019-11-05 by Alvin1Zhang