espressif/esp-idf #4286: CMake fails to read git revision when using git worktree

devanlai, 2019-11-05:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version: v4
- Module or chip used:  ESP32-WROOM-32D
- IDF version: `v4.1-dev-815-ga45e998`
- Build System:         CMake
- Compiler version: `xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0`
- Operating System:     Linux via Windows Subsystem for Linux
- Power Supply:         USB

## Problem Description

When using the [git worktree](https://git-scm.com/docs/git-worktree) feature to manage the project source code, the CMake build process is unable to find the current git revision hash/description to populate `PROJECT_VER`.

I believe that this is because in each worktree directory, `git-worktree` places a special `.git` file that, rather than being a directory of git objects, contains the path to the actual git object storage in the root worktree's `.git` directory.

Applying this patch written against an older copy of `GetGitRevisionDescription.cmake` to the copy shipped with `esp-idf` fixes the issue for me:
https://phabricator.kde.org/D9172

### Expected Behavior

Running `idf.py build` or otherwise triggering the build process should correctly detect the git revision even when the project is using `git-worktree` and use that to populate `PROJECT_VER`

### Actual Behavior

The build process fails to detect the git revision and falls back with "Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER."

### Steps to repropduce

1. Create a worktree from an existing project version controlled with git, e.g:
```
git clone ../worktree-test
```
2. Move to the worktree directory
```
cd ../worktree-test
```
3. Attempt to build the project with `idf.py build`
```
idf.py build
```

### Code to reproduce this issue

N/A

renzbagaporo, 2019-11-21:
A fix for this should be on the way.

renzbagaporo, 2019-11-27:
Can you retest @devanlai? The fix is supposed to be https://github.com/espressif/esp-idf/commit/9efbe1bb4709e917f6eca6d68469f60523cd216b.

devanlai, 2019-11-27:
I pulled master and I can confirm that it's fixed as of `v4.1-dev-1086-g93a8603`. Thanks!

closed 2019-11-27 by devanlai