espressif/esp-idf #4287: how to run eth2wifi(eth2STA) , need help ?

sushant-ht, 2019-11-05:
i am trying to use eth2ap as eth2sta, i modified the given [example ](https://github.com/espressif/esp-idf/tree/master/examples/ethernet/eth2ap)to eth2sta, i had successfully run eth2ap example and wanted to modify it for STA, i also read the eth2wifi [example ](https://github.com/espressif/esp-iot-solution/tree/master/examples/eth2wifi)in IOT-SOlutions, but i couldnt run it on my esp-idf version, hence i had need to modify the give example from eth2ap to eth2sta,

i believe i have done necessary changes rightly still i am unable to get it working,
following is modified code

````
/* eth2ap (Ethernet to Wi-Fi AP packet forwarding) Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <string.h>
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "esp_event.h"
#include "esp_log.h"
#include "esp_eth.h"
#include "esp_wifi.h"
#include "nvs_flash.h"
#include "esp_private/wifi.h"
#include "driver/gpio.h"
#include "sdkconfig.h"

static const char *TAG = "eth_example";
static esp_eth_handle_t s_eth_handle = NULL;
static xQueueHandle flow_control_queue = NULL;
static bool s_sta_is_connected = false;
static bool wifi_is_connected = false; // for STA
static bool s_ethernet_is_connected = false;
static uint8_t s_eth_mac[6];

#define FLOW_CONTROL_QUEUE_TIMEOUT_MS (100)
#define FLOW_CONTROL_QUEUE_LENGTH (40)
#define FLOW_CONTROL_WIFI_SEND_TIMEOUT_MS (100)

typedef struct {
    void *packet;
    uint16_t length;
} flow_control_msg_t;

void printHEX(char *buffer, int n)
{
	for (int i = 0; i < n; i++)
	{
		printf("%02X", buffer[i]);
	}
	printf("\n[i]\n");
}

// Forward packets from Wi-Fi to Ethernet
static esp_err_t pkt_wifi2eth(void *buffer, uint16_t len, void *eb)
{
	ESP_LOGI(TAG, "wifi-2-eth");
    if (s_ethernet_is_connected) {
		printf("[Wi-Fi to Ethernet]\n\r");
		printHEX(buffer, len);
        if (esp_eth_transmit(s_eth_handle, buffer, len) != ESP_OK) {
        }
    }
    esp_wifi_internal_free_rx_buffer(eb);
    return ESP_OK;
}

// Forward packets from Ethernet to Wi-Fi
// Note that, Ethernet works faster than Wi-Fi on ESP32,
[... 213 lines cut ...]
    gpio_pad_select_gpio(GPIO_NUM_16);
    
	gpio_set_direction(GPIO_NUM_16 , GPIO_MODE_OUTPUT );
	gpio_set_direction(GPIO_NUM_2 , GPIO_MODE_OUTPUT );
	gpio_set_direction(GPIO_NUM_14 , GPIO_MODE_OUTPUT );
    vTaskDelay(10 / portTICK_RATE_MS);
	printf("Set OSC EN  = 1");
	tmp = 1;
	while(++tmp < 10){
		printf("init GPIO\n\r");
		gpio_set_level(GPIO_NUM_14, 1);
		gpio_set_level(GPIO_NUM_16, 0);
		gpio_set_level(GPIO_NUM_2, 1);
		vTaskDelay(100 / portTICK_RATE_MS);
		gpio_set_level(GPIO_NUM_2, 0);
		gpio_set_level(GPIO_NUM_16, 1);
		vTaskDelay(100 / portTICK_RATE_MS);
		printf("get gpio14 %d\n\r", gpio_get_level(GPIO_NUM_14));
	}
    initialize_ethernet();
    initialize_wifi();
}
```

i have posted same issue in [forum] = ( https://esp32.com/viewtopic.php?f=2&t=12968 ) but have found no response

sushant-ht, 2019-11-06:
any help @esp admin

suda-morris, 2019-11-07:
I would like to suggest you try the example in iot-solution, since that one could best match what you need. The IDF example---eth2wifi, only shows another usage of Ethernet API without TCP/IP stack enabled. It's not for production, and also can't cover all scenario. If you found some bugs in the driver, you can post them here, otherwise the forum is the best place.

Alvin1Zhang, 2020-01-08:
@sushant-ht Thanks for reporting. Will close the issue. Thanks.

closed 2020-01-08 by Alvin1Zhang

vlmsx, 2021-05-12:
I did not see you running esp_eth_start()

vlmsx, 2021-05-12:
我没有看见你运行esp_eth_start()函数