espressif/esp-idf #4288: esp-modem.c semaphorTake : crash with TTGO T-CALL SIM800L

VXConsulting, 2019-11-05:
## Environment

- Development Kit:      ESP32-Dev-Kit - TTGO T-CAL SIM800L 
- Kit version (for DevKitC): [v1]
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): 3.3
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows]
- Power Supply:         [USB]

## Problem Description

Hello,

I debugged and I found where is the problem but I do not know how to solve it.

In esp_modem.c, line 215
`/* Send command via UART */
    ESP_LOGI("TAG", "UART WRITE RESULT : %d",uart_write_bytes(esp_dte->uart_port, command, strlen(command)));
    /* Check timeout */
    MODEM_CHECK(xSemaphoreTake(esp_dte->process_sem, pdMS_TO_TICKS(timeout)) == pdTRUE, "process command timeout", err);`

The command is sent to UART because the log shows the result >=0.
The next line xSemaphoreTake has an error and then, it crashes the program.

Regards,

### Actual Behavior

Crash
` (297) boot: Loaded app from partition at offset 0x10000
I (297) boot: Disabling RNG early entropy source...
I (298) cpu_start: Pro cpu up.
I (301) cpu_start: Application information:
I (306) cpu_start: Project name:     esp-idf
I (311) cpu_start: App version:      v3.3
I (316) cpu_start: Compile time:     Oct 31 2019 15:06:55
I (322) cpu_start: ELF file SHA256:  1f5f0b574ec01900...
I (328) cpu_start: ESP-IDF:          v3.3
I (332) cpu_start: Starting app cpu, entry point is 0x4008107c
0x4008107c: call_start_cpu1 at D:/esp/esp-idf/components/esp32/cpu_start.c:270

I (0) cpu_start: App cpu up.
I (343) heap_init: Initializing. RAM available for dynamic allocation:
I (350) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (356) heap_init: At 3FFB4148 len 0002BEB8 (175 KiB): DRAM
I (362) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (368) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (375) heap_init: At 400896AC len 00016954 (90 KiB): IRAM
I (381) cpu_start: Pro cpu `

### Code to reproduce this issue

```
modem_dce_t *dce = sim800_init(dte);
```

suda-morris, 2020-05-26:
Please check the connection between ESP32 UART and GSM modem first.

david-cermak, 2020-12-21:
@VXConsulting Any update on this issue? If the first initial sync fails, it's usually the wiring/connection problem. Could you please share an update if any, or close the issue?

Alvin1Zhang, 2021-01-18:
Thanks for reporting, feel free to reopen.

closed 2021-01-18 by Alvin1Zhang

QuentinFarizon, 2021-12-16:
I get the same error with the same development kit.

@suda-morris what do you mean by "check the connection between ESP32 UART and GSM modem"
And @david-cermak " it's usually the wiring/connection problem" ?

This is board that I bought (Lilygo TTGO TCALL 1.4 with sim800L)