espressif/esp-idf #4289: ESP32-WROVER-B (M213DH6464PC3Q0) : Failed to init external RAM! (IDFGH-2129)

sagarvegg, 2019-11-05:
esp32-wrover-b fail to run esp-idf example --> ~$IDF_PATH/examples/system/himem

INSTRUCTIONS
============

## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit): [v3]
- Module or chip used:  [ESP32-WROVER-B]
- IDF version (run ``git describe --tags`` to find it): v3.3
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description

esp-idf system example for psram test -->  ~$IDF_PATH/examples/system/himem don't work

### Expected Behavior

rst:0x1 (POWERON_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6124
load:0x40078000,len:10084
load:0x40080400,len:6552
entry 0x40080764
I (28) boot: ESP-IDF v3.2-dev-1455-ga51d5706f-dirty 2nd stage bootloader
I (29) boot: compile time 18:51:28
I (30) boot: Enabling RNG early entropy source...
I (35) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 4MB
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot: End of partition table
I (85) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0dde0 ( 56800) map
I (114) esp_image: segment 1: paddr=0x0001de08 vaddr=0x3ff80000 size=0x00000 (     0) load
I (114) esp_image: segment 2: paddr=0x0001de10 vaddr=0x3ff80000 size=0x00000 (     0) load
I (120) esp_image: segment 3: paddr=0x0001de18 vaddr=0x3ffb0000 size=0x01fb4 (  8116) load
I (132) esp_image: segment 4: paddr=0x0001fdd4 vaddr=0x3ffb1fb4 size=0x00000 (     0) load
I (138) esp_image: segment 5: paddr=0x0001fddc vaddr=0x40080000 size=0x00234 (   564) load
I (147) esp_image: segment 6: paddr=0x00020018 vaddr=0x400d0018 size=0x180d4 ( 98516) map
I (191) esp_image: segment 7: paddr=0x000380f4 vaddr=0x40080234 size=0x001cc (   460) load
I (191) esp_image: segment 8: paddr=0x000382c8 vaddr=0x40080400 size=0x0e14c ( 57676) load
I (221) esp_image: segment 9: paddr=0x0004641c vaddr=0x400c0000 size=0x00000 (     0) load
I (222) esp_image: segment 10: paddr=0x00046424 vaddr=0x50000000 size=0x00000 (     0) load
I (228) esp_image: segment 11: paddr=0x0004642c vaddr=0x50000000 size=0x00000 (     0) load
I (245) boot: Loaded app from partition at offset 0x10000
I (246) boot: Disabling RNG early entropy source...
I (250) spiram: Found 64MBit SPI RAM device
I (254) spiram: SPI RAM mode: flash 40m sram 40m
I (259) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (266) cpu_start: Pro cpu up.
I (270) cpu_start: Starting app cpu, entry point is 0x40081174
I (0) cpu_start: App cpu up.
I (735) spiram: SPI SRAM memory test OK
I (735) heap_init: Initializing. RAM available for dynamic allocation:
I (735) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (741) heap_init: At 3FFB3448 len 0002CBB8 (178 KiB): DRAM
I (748) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (754) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (760) heap_init: At 4008E54C len 00011AB4 (70 KiB): IRAM
[... 278 lines cut ...]
I (121) esp_image: segment 2: paddr=0x00023114 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/sagar/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (124) esp_image: segment 3: paddr=0x0002351c vaddr=0x40080400 size=0x0ca88 ( 51848) load
I (154) esp_image: segment 4: paddr=0x0002ffac vaddr=0x00000000 size=0x00064 (   100) 
I (155) esp_image: segment 5: paddr=0x00030018 vaddr=0x400d0018 size=0x189a4 (100772) map
0x400d0018: _flash_cache_start at ??:?

I (204) boot: Loaded app from partition at offset 0x10000
I (204) boot: Disabling RNG early entropy source...
I (204) psram: This chip is ESP32-D0WD
E (210) cpu_start: Failed to init external RAM!
abort() was called at PC 0x40081496 on core 0
0x40081496: call_start_cpu0 at /home/sagar/esp/esp-idf/components/esp32/cpu_start.c:174 (discriminator 1)

ELF file SHA256: 33d23b69a2c8eb6be8cfaa99c0c1f7e32909681276738a2d97bdaac8fa5f2231

Backtrace: 0x40086494:0x3ffe3bb0 0x400866d9:0x3ffe3bd0 0x40081496:0x3ffe3bf0 0x4007940b:0x3ffe3c30 0x400794bd:0x3ffe3c60 0x400794db:0x3ffe3ca0 0x400797c1:0x3ffe3cc0 0x40080796:0x3ffe3df0 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20
0x40086494: invoke_abort at /home/sagar/esp/esp-idf/components/esp32/panic.c:715

0x400866d9: abort at /home/sagar/esp/esp-idf/components/esp32/panic.c:715

0x40081496: call_start_cpu0 at /home/sagar/esp/esp-idf/components/esp32/cpu_start.c:174 (discriminator 1)

```

sagarvegg, 2019-11-05:
if anyone find solution regrading to this thing of psram than please respond fast

costaud, 2019-11-06:
@sagarvegg  Did you found this error on only one module, or on all your ESP32-WROVER-B modules ?

sagarvegg, 2019-11-06:
@costaud  I found this issue on each 4mb and 8mb ram ESP32-Wrover-B modules...

negativekelvin, 2019-11-06:
Are you using wrover boards with nothing connected or what is your schematic?

sagarvegg, 2019-11-06:
Hello,
I got the order from aliexpress.
We have uncovered top part of the esp32-wrover and check whether ram is genuine or not
RAM is indeed espressif's PSRAM64.
The thing is we tried this thing in several module the result is same for all.
The ram is being found but the himem test to check the ram is okay or not is being failed.

Himem has 4352KiB of memory, 4352KiB of which is free. Testing the free memory...
check_mem_seed: 0 has 0x00000000 expected 0x777f7370
Error in block 0
assertion "test_region(memfree, 0xaaaa)" failed: file "../main/himem_test_main.c", line 95, function: app_main
abort() was called at PC 0x400d8d4f on core 0

akshar001, 2019-11-06:
Hello @negativekelvin and @costaud.
We have ordered 10 boards with the same module but all of them having a same problem.

costaud, 2019-11-06:
What kind of boards ?  wrover-kit ?

costaud, 2019-11-06:
Do I understand correctly that you have tested himem on wrover-kit before and it works properly (that's where you got the correct expected log) but this time you ordered 10 more boards, but found it failed himem test on those 10 boards ?

akshar001, 2019-11-06:
@costaud No we were using the 4MB SRAM ESP32-WROVER till now. We were in need of higher SRAM for one of our application. So we ordered 10pcs with 8MB PSRAM. And we are running himem example code right now. But it's not working. 
[image]

negativekelvin, 2019-11-06:
So it is a custom pcb? What is the schematic? Have you tried QIO mode or esp-idf master branch has same problem?

akshar001, 2019-11-07:
@negativekelvin not custom pcb. Same board that is in the photo. From ESPRESSIF.
Qio Mode tried.
ESP-IDF MASTER BRANCH tried.
The problem is in 8MB PSRAM64 running with esp.
We have replaced it with 4MB PSRAM and it's working fine now.
So is it a software issue or hardware issue.
The PSRAM64 is genuine as well. 
[image]

negativekelvin, 2019-11-07:
Module is from espressif but green pcb is custom, right?  Interesting results.  You replaced it with a 3.3v  32mbit psram?

sagarvegg, 2019-11-07:
It's not green pcb,it's not custom it's from espressif. it's black pcb @negativekelvin .
And no it's not 3.3V PSRAM it's 1.8V ESP-PSRAM32 that's working

igrr, 2019-11-07:
> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

MTDI pin seems to be strapped low, which sets the VDDSDIO voltage to 1.8V. Isn't this 64MB PSRAM a 3.3V part?

negativekelvin, 2019-11-07:
@igrr no, strapped high for 1.8v

akshar001, 2019-11-07:
No @igrr its 1.8V ram

akshar001, 2019-11-07:
Flash is running fine at 1.8V.

costaud, 2019-11-07:
In menuconfig, did you set the psram type to auto-detect or 32M or 64M ?  @akshar001

akshar001, 2019-11-07:
@costaud we have tried in both ways auto and PSRAM64M

negativekelvin, 2019-11-07:
Wrover-b ships with 3.3v flash and psram so if you are setting efuse to 1.8v that is probably the issue

akshar001, 2019-11-07:
Right now the board we have both 1.8V flash and PSRAM64. The thing is when we changed PSRAM64 with PSRAM32 the PSRAM is working fine?

costaud, 2019-11-07:
Would you please also check the type/id of the flash ?

costaud, 2019-11-07:
@sagarvegg Would you please also provide the chip ID or MAC of those modules ?

akshar001, 2019-11-07:
Chip id :- ESP32D0WDQ6 (revision 1)
Mac:- 30:AE:A4:60:B4:40
Flash Size:- 4MB
Manufacturer :- C8
Device:- 6016

akshar001, 2019-11-07:
@costaud @negativekelvin we thought the same way that components are not correct. But when we change the psram64 with psram32 its working

negativekelvin, 2019-11-07:
I'm confused because your first picture shows a wrover-b metal shield but your second picture shows the old style wrover module with a psram64 (1.8v).  And your logs show strapping pins are set for 3.3v.  So are you setting efuse to 1.8v?

costaud, 2019-11-07:
Have you check the  power to the flash and psram, is it 3.3v or 1.8v ?
Would you please also upload a picture of the whole module without the shield ?

akshar001, 2019-11-07:
Sure.
Yes power has been checked.
It’s 1.8V

costaud, 2019-11-07:
If it's 1.8v, then that's really wired. Need the picture of the whole module(without the shield) to further confirm.

akshar001, 2019-11-07:
[image]

akshar001, 2019-11-07:
[image]

negativekelvin, 2019-11-07:
That picture shows a wrover-b with psram64h which is 3.3v and does not match the other picture so now I am more confused

akshar001, 2019-11-07:
@costaud @negativekelvin if the strapping pin has a problem then i dont understand why its running in param32 with same flash same board

costaud, 2019-11-07:
@negativekelvin  Agree, I'm also confused by the pictures.

akshar001, 2019-11-07:
Okay in this board the ram is showing as psramh.
Let me send you another board picture which has param64 only

Alvin1Zhang, 2019-11-08:
@costaud @negativekelvin  Thanks for the notes. @sagarvegg Thanks for reporting, would you help share if any updated board picture which has PSRAM64 only? Thanks.

akshar001, 2019-11-08:
@costaud @negativekelvin @Alvin1Zhang Sorry for misunderstanding. The picutres that i have shared with you guys PSRAM64H one. Has 3.3V flash memory. As MTDI pin in the log that i have shared with you guys is LOW as well. The thing is still RAM is not working. The program is getting crashed?

akshar001, 2019-11-08:
I can share all the details and log that you guys will need just tell me

negativekelvin, 2019-11-08:
Share schematic of the base board and efuse settings

akshar001, 2019-11-12:
hello @negativekelvin the boards are ordered from aliexpress. So i do not have the schematic but i will send you efuse settings now.

wegunterjr, 2019-11-12:
what were the sdkconfig.h settings for this board?

Alvin1Zhang, 2020-01-08:
@akshar001 Thanks for reporting. Would you please help share if any updates for this issue? Thanks.

costaud, 2020-05-26:
We'll close this issue for now, feel free to reopen if you get further information.

closed 2020-05-26 by costaud