espressif/esp-idf #4291: Guru Meditation Error: Core  1 panic'ed (LoadProhibited) on esp_http_client_set_post_field

sagar448, 2019-11-06:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): 
    v3.3 release
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System:     [Windows]
- Power Supply:         [USB]

## Problem Description
Reading the temperature data from a thermal camera. It produces an array of 768 float numbers. Now I want to send this to my backend via HTTP post. It works perfectly for the first 6 posts. (It reads and posts the data every 2 seconds/1 second) After that it gives off a GURU meditation load prohibited error on the line "esp_http_client_set_post_field(client, pData, strlen(pData));"

I don't know if its the amount of data being sent or the heap is running out?

### Expected Behavior
While loop to just post the data every 2 seconds, indefinitely 

### Actual Behavior

### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
//Standard Libraries
#include <string.h>
#include <stdlib.h>

//Other ESP-IDF Components
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "cJSON.h"
#include "driver/gpio.h"
#include "driver/i2c.h"
#include "esp_http_client.h"
#include "MLX90640_I2C_Driver.h"
#include "esp_log.h"

static const char *TAG = "Spacr";
static const char *secretKey = "FAwh5#w5c?wtLCvZ";

const int requestSend = BIT1;
static EventGroupHandle_t requestDone;

extern const uint8_t name_pem_start[] asm("_binary_name_pem_start");
extern const uint8_t name_pem_end[] asm("_binary_name_pem_end");

const char *deviceName = (const char*) name_pem_start;

esp_err_t _http_event_handler(esp_http_client_event_t *evt)
{
    switch(evt->event_id) {
        case HTTP_EVENT_ERROR:
            ESP_LOGD(TAG, "HTTP_EVENT_ERROR");
            break;
        case HTTP_EVENT_ON_CONNECTED:
            ESP_LOGD(TAG, "HTTP_EVENT_ON_CONNECTED");
            break;
        case HTTP_EVENT_HEADER_SENT:
            ESP_LOGD(TAG, "HTTP_EVENT_HEADER_SENT");
[... 806 lines cut ...]

0x400d2573: task_run at C:\Users\sagar\Desktop\Spacr-Occu\build/../main/detect_people.c:116 (discriminator 1)

0x400d21d3: detectPeople at C:\Users\sagar\Desktop\Spacr-Occu\build/../main/main.c:72

0x400887a1: vPortTaskWrapper at C:/Users/sagar/Desktop/esp-idf/components/freertos/port.c:403

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6452
load:0x40078000,len:11484
load:0x40080400,len:6700
entry 0x40080768
I (29) boot: ESP-IDF v3.3-beta1-889-gbeb34b539 2nd stage bootloader
I (29) boot: compile time 21:00:23
```

Any help would be greatly appreciated, thank you! 
@mahavirj

sagar448, 2019-11-06:
@negativekelvin @projectgus 
Hey guys any help? Apologies for the pestering but I cannot move forward without solving this issue.

projectgus, 2019-11-07:
Hi @sagar448 ,

Without seeing your code it's not possible to debug, but there doesn't seem to be any indication this is a bug in ESP-IDF.

Here are some pointers to help you debug this:

* LoadProhibitedError with EXCVADDR=0x00000000 means that the code tried to dereference a NULL pointer.
* The PC address 0x400014fd is in the ROM implementation of strlen(), so most likely the `pData` variable is NULL for some reason.
* You'll need to debug your code to figure out the rest, I think.
* If you think you might be running out of memory due to a memory leak or similar (ie malloc() has returned NULL somewhere and the code didn't check this) then you can use the heap information functions shown here: https://docs.espressif.com/projects/esp-idf/en/v3.3/api-reference/system/heap_debug.html#heap-information - this page also has information on finding memory leaks, if you determine that you have indeed run out of memory.

sagar448, 2019-11-07:
Thank you so much @projectgus !! I had a feeling it was a memory leak, but I didn't know why strlen was returning null. Anyway, problem solved, I traced the memory leak to the cJSON_PrintUnformatted(root) function and wow no where in the documentation I read that the pointer needed to be freed. 

Solution for anyone else:

char *cjsonUn = cJSON_PrintUnformatted(root);
do_some_stuff();
free(cjsonUn);

Thanks again! :)

closed 2019-11-07 by sagar448