espressif/esp-idf #4292: intenger overflow in pdMS_TO_TICKS

az-z, 2019-11-06:
Hello,
 the ticket is for 2 issues.

a. The version of projdefs.h ( IDF v 3.3) is out of syn from the FREERTOS's [one](https://sourceforge.net/p/freertos/code/HEAD/tree/trunk/FreeRTOS/Source/include/projdefs.h)
2. The pdMS_to_ticks overflow the integer when the time is in **hours**. 
documented [here](https://www.esp32.com/viewtopic.php?f=19&t=12983&p=51412#p51412)

projectgus, 2019-11-06:
Hi @az-z ,

As discussed on the thread, changing the type here won't prevent an overflow if using this macro to convert >1.2 hours into ticks. The better solution in this case is to divide by pdMS_TO_TICKS.

However we can still update the macro to cast the operand to TickType_t.

az-z, 2019-11-06:
@projectgus ,
 thank you for your feedback.
what if:
#define pdMS_TO_TICKS( xTimeInMs ) (  ( TickType_t ) ( xTimeInMs ) * ( configTICK_RATE_HZ  / ( TickType_t ) 1000 ) )

negativekelvin, 2019-11-06:
( configTICK_RATE_HZ / ( TickType_t ) 1000 ) = 0 for configTICK_RATE_HZ < 1000

az-z, 2019-11-06:
combine into one?
#if configTICK_RATE_HZ < 1000
#define pdMS_TO_TICKS( xTimeInMs ) ( ( TickType_t ) ( xTimeInMs ) * ( configTICK_RATE_HZ / ( TickType_t ) 1000 ) )
#else 
#define pdMS_TO_TICKS( xTimeInMs ) ( ( TickType_t ) ( xTimeInMs ) * ( configTICK_RATE_HZ / ( TickType_t ) 1000 ) )

negativekelvin, 2019-11-06:
That only works if tick rates greater than 1000 are also multiple of 1000. Usually max is 1000 anyway so it is not helpful.   
#define pdMS_TO_TICKS( xTimeInMs ) ( ( TickType_t ) ( xTimeInMs ) / ( ( TickType_t ) 1000 / configTICK_RATE_HZ ) ) works best for configTICK_RATE_HZ ≤ 1000 as long as 1000 % configTICK_RATE_HZ = 0

az-z, 2019-11-06:
Then what would be a good solution?

negativekelvin, 2019-11-06:
As was suggested, don't use it.  https://github.com/espressif/esp-idf/search?q=portTICK_period_MS&unscoped_q=portTICK_period_MS

az-z, 2019-11-06:
not sure what you meant with the url... the search for pdm_to_ticks returns a number of [results](https://github.com/espressif/esp-idf/search?p=2&q=pdMS_TO_TICKS&unscoped_q=pdMS_TO_TICKS) too.

negativekelvin, 2019-11-06:
Yes they all show how to calculate a delay without using pdMS_TO_TICKS

az-z, 2019-11-06:
the issue with overflow has been raised with the freertos team [before ](https://sourceforge.net/p/freertos/discussion/382005/thread/317c7160/).

Their answer was :

> You should not need to edit any of the core FreeRTOS source files, just define pdMS_TO_TICKS() however you want in FreeRTOSConfig.h. The default definition appears as below, so will only be used if pdMS_TO_TICKS is not defined elsewhere.

#ifndef pdMS_TO_TICKS
    #define pdMS_TO_TICKS( xTimeInMs ) ( ( TickType_t ) ( ( ( TickType_t ) ( xTimeInMs ) * ( TickType_t ) configTICK_RATE_HZ ) / ( TickType_t ) 1000 ) )
#endif

shouldn't the projdefs.h be updated with the proper version from upstream?
thank you.

az-z, 2019-11-06:
> Yes they all show how to calculate a delay without using pdMS_TO_TICKS. You are welcome to try fixing it in a PR and upstreaming it to freertos if you want but avoiding it is easier.

got it. yeah, they didn't seem to be interested to fix it before. (see my post above)

negativekelvin, 2019-11-07:
Right, to fix it you have to avoid the multiply and avoid fractional intermediate division results. It becomes a bit complicated to support arbitrary tick rates but that is already a problem for portTICK_period_MS calculation.

az-z, 2019-11-08:
do i need to do anything to update the  projdefs.h from the upstream?

projectgus, 2020-04-30:
The definition of portTICK_RATE_MS (and other macros) from the newer upstream version FreeRTOS 10 is now in our preview branch and will be updated in master when this branch is merged and we update the master branch to FreeRTOS 10:
https://github.com/espressif/esp-idf/blob/feature/freertos_upstream_plus_tlsf_allocator_preview/components/freertos/include/freertos/projdefs.h#L41

I'm going to close this issue now though, as not using the FreeRTOS 10 projdefs.h is not a bug in the current version (which is [based on FreeRTOS 8](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos_additions.html)).

closed 2020-04-30 by projectgus

jimparis, 2020-05-28:
@projectgus I don't understand your answer.  The only change I see between the FreeRTOS 8 and FreeRTOS 10 macros are that they added a bunch of TickType_t casts which have no effect on the esp32 case (since TickType_t is uint32_t, and normal type promotions rules do the same thing as those added casts).

It doesn't address the real issue, which is that you're taking a uint32_t, multiplying, then dividing.  The multiply overflows even though the result would not.  An intermediate cast to uint64_t could fix it, or special case values like HZ=1000, etc?  It seems weird to say "Don't use this macro" when the macro could be fixed.

Rodrigodd, 2025-02-25:
I ran into the same problem and fixed it by defining the macro in a file passed as a `-include` argument:

```cmake
# after idf_component_register
target_compile_options(${COMPONENT_LIB} PRIVATE "-include" "freertos_fix.h")
```

freertos_fix.h:

```c
#include "FreeRTOSConfig.h"

static_assert((1000 / configTICK_RATE_HZ ) * configTICK_RATE_HZ == 1000, "configTICK_RATE_HZ must be a divisor of 1000");

// define pdMS_TO_TICKS to avoid overflows
#define pdMS_TO_TICKS( xTimeInMs )    ( ( TickType_t ) (  ( TickType_t ) ( xTimeInMs ) / ( ( TickType_t ) 1000U / ( TickType_t ) configTICK_RATE_HZ  ) ) )

// define pdTICKS_TO_MS to avoid overflows
#define pdTICKS_TO_MS( xTicks )    ( ( TickType_t ) ( uint64_t ) ( xTicks ) * ( 1000 / configTICK_RATE_HZ ) )
```

The `-include` option ensures that the file is included in every C source file, which is not ideal but is better than manually including it before every FreeRTOS header.

I also just noticed that `pdTICKS_TO_MS` casts the intermediate value to `uint64_t` before casting it back to `TickType_t`, at least in the latest version of ESP-IDF. Strange that the same approach is not applied to `pdMS_TO_TICKS`.