espressif/esp-idf #4293: LCB sometimes not released after ESP_GATTC_DISCONNECT_EVT

AlpinProgrammer, 2019-11-06:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version: v2
- Module or chip used: ESP32-WROOM-32
- IDF version: v3.3 (stable)
- Build System: Make
- Compiler version: 1.22.0-80-g6c4433a
- Operating System: Linux
- Power Supply: USB

## Problem Description

Hi all ! I come to you because I encountered an error from the stack in a certain situation.
I need to control several devices in BLE in order to connect and write to one characteristic (the handle has been stored beforehand). Once the operation is done, the device stays connected.

My problem occurs **sometimes**, when I need to control another device:
1.  I disconnect the former device using *esp_err_t esp_ble_gattc_close(esp_gatt_if_t gattc_if, uint16_t conn_id)*

2. I get the *ESP_GATTC_DISCONNECT_EVT*

3. I try to connect my new device directly calling *esp_err_t esp_ble_gattc_open(esp_gatt_if_t gattc_if, esp_bd_addr_t remote_bda, esp_ble_addr_type_t remote_addr_type, bool is_direct)* because I know its MAC address and the gattc_if.

4. I get the following error trace:

W (17262) BT_L2CAP: L2CA_ConnectFixedChnl(0x0004) - no LCB
E (17302) BT_GATT: gatt_connect failed
E (17312) BT_APPL: Connection open failure
E (17362) GATT: open failed, status 133

## What I found

After some research I deduced that the problem came from the "no LCB" problem. When I dug in esp-idf, I realised that **l2cu_release_lcb** function is supposed to be called when disconnecting. I noticed when I encountered the problem, that the *l2cu_release_lcb* function had not been called.

## What I need

What can cause this type of problem ? Do I need to be careful about some best practices that might avoid this kind of problem ?
I have a limited understanding of low level layers in BLE...
I can't spot anything that is against rules in my code for the moment. I'll try to provide a test code as soon as it's ready.

Thanks for your help :)

gengyuchao, 2019-11-26:
We recommend scanning the adv device before opening. If the adv is not scanned, please confirm that the peer broadcast has started before calling esp_ble_gattc_open otherwise the program will return an error.

AlpinProgrammer, 2019-11-26:
Thanks for your answer !
Indeed, when we request an open without scan, the server is advertising.

Also, I recently found out that when we don't get the l2cu_release_lcb, it is eventually called after a time t=connection supervision timeout

Alvin1Zhang, 2019-11-27:
@AlpinProgrammer Thanks for the updates. Has your issue been resolved? If it gets resolved, would you please help close the issue? Thanks.

Alvin1Zhang, 2019-12-04:
@AlpinProgrammer Thanks for reporting, feel free to reopen if you still have the issue, and please help file another issue if that is a different issue. Thanks.

closed 2019-12-04 by Alvin1Zhang

AlpinProgrammer, 2020-01-16:
Hi @Alvin1Zhang ! So far, I still have the issue on v3.3.
I found a workaround by reducing the connection supervision timeout because as I mentioned earlier, l2cu_release_lcb is eventually called after a time t=connection supervision timeout.

If I try on v3.3.1 I will let you know if I still have the issue with that version.
Thanks

Alvin1Zhang, 2020-01-16:
@AlpinProgrammer Thanks for sharing the updates.

AlpinProgrammer, 2020-02-20:
Hi @Alvin1Zhang , I just tried my code on esp-idf v3.3.1.

Unfortunately, it still doesn't work, I have the same trace:
`W (167316) BT_L2CAP: L2CA_ConnectFixedChnl(0x0004) - no LCB`
`E (167326) BT_GATT: gatt_connect failed`
`E (167326) BT_APPL: Connection open failure`
`I (167336) GATTC: lio_ble_gattc_open_failed`
`E (167336) GATT: open failed, status 133`

Besides, my workaround (_reducing the connection supervision timeout_) proves to be a bad idea since the BLE server devices connected on the ESP32 keep getting disconnected.