espressif/esp-idf #4294: modbus master on IDF V3.3

Pratikm78, 2019-11-06:
I am using v3.3 stable release of the esp-idf however the modbus component only supports slave in that version of the idf. I want to programming using the modbus master component which is part of the newer versions of the idf. 

I have to use v3.3 because a sdk that I am using is only supported by the v3.3 and "make" build not the idf.py found in the newer versions. 

I have copied the modbus component from the v4.0 to v3.3 however i can not build the code

alisitsyn, 2019-11-07:
Hi Pratikm78,
The v4.0 contains significant changes compare to v.3.3. Did you copy the whole freemodbus folder from v4.0 into your_project/components/freemodbus? The v4.0 example can be compiled with make, but there can be conflicts with changes in some other components. Could you provide the compilation logs? I would recommend you to use latest revision of ESP_IDF.

Alvin1Zhang, 2020-01-08:
@Pratikm78 Thanks for reporting. Would you please help share if any updates for the issue? Could you help provide the compilation logs? Thanks.

Alvin1Zhang, 2020-03-30:
@Pratikm78 Thanks for reporting, will close due to short of feedback, feel free to reopen with more details. Thanks.

closed 2020-03-30 by Alvin1Zhang