espressif/esp-idf #4296: Suggestion: Asio is outdated, consider upgrading & add WolfSSL support?

huming2207, 2019-11-06:
## Environment

- IDF version: master, https://github.com/espressif/esp-idf/commit/a45e9985344575a80acb1dc9c12e7bec4d8af401

(other irrelevant stuff removed)

## Problem Description

Hi there,

I've posted this issue on the forum before but no one replied. So I think I might have posted that on the wrong place. Here's the original topic: https://www.esp32.com/viewtopic.php?f=13&t=12973

Basically, nowadays Asio supports WolfSSL as the SSL library backend (ref: https://www.wolfssl.com/wolfssl-support-asio-boost-asio-c-libraries/) since earlier this year. Meanwhile, ESP-IDF also has WolfSSL included. However, the Asio library in ESP-IDF hasn't been updated for a year and it does not have SSL/TLS support. 

Could you guys please upgrade this library? Thanks in advance!

Regards,
Jackson

david-cermak, 2019-11-12:
Hi @huming2207 

Thank you for this suggestion. Indeed the esp-asio hasn't been updated for a long time. Will update to the latest master soon.
As for the TLS support, this may take some time however, since ESP ASIO port is really basic with no SSL support and it's preferred to use esp-tls rather than WolfSSL directly. With latest ASIO it should still be possible to use WolfSSL setting define `ASIO_USE_WOLFSSL` on project level.

huming2207, 2019-11-12:
Hi @david-cermak ,

Thanks for your reply. I agree. If `esp-tls` can be used as a backed of `Asio` on ESP-IDF, it should be better. 

Meanwhile, don't forget the WolfSSL has a dual-license model. It may cause some problems with some commercial projects.

Jackson.

closed 2020-01-10 by espressif-bot

schmidma, 2020-04-01:
Is there a new status regarding `asio` ssl support in `esp-idf`?

As I can see `asio` requests file system support from `wolfSSL` when using `asio` and `wolfSSL` with `ASIO_USE_WOLFSSL`.
E.g. `asio::ssl::context::load_verify_file(...)` requests usage of `SSL_CTX_load_verify_locations(...)` which is implemented by `wolfSSL` in `wolfSSL_CTX_load_verify_locations(...)` but disabled in the esp version with `NO_FILESYSTEM`.

Also when porting `asio` to use `esp-tls` there seems to be no `asio`-native option to disable file system related implementations.

Are there any plans to support the ssl set of `asio`?