espressif/esp-idf #4297: mac install.sh error

isDerek, 2019-11-07:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROVER-B]
- IDF version (run ``git describe --tags`` to find it): 
    v4.1-dev-815-ga45e99853
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    /xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     [macOS]
- Power Supply:         [USB]

## Problem Description

1. run ./install.sh
```
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp32-2019r1-8.2.0 (already installed)
WARNING: directory for tool xtensa-esp32s2-elf version esp32s2-dev-4-g3a626e93-8.2.0 is present, but tool was not found
Installing xtensa-esp32s2-elf@esp32s2-dev-4-g3a626e93-8.2.0
file xtensa-esp32s2-elf-gcc8_2_0-esp32s2-dev-4-g3a626e-macos.tar.gz  is already downloaded
WARNING: destination path already exists, removing
Extracting /Users/derek/.espressif/dist/xtensa-esp32s2-elf-gcc8_2_0-esp32s2-dev-4-g3a626e-macos.tar.gz  to /Users/derek/.espressif/tools/xtensa-esp32s2-elf/esp32s2-dev-4-g3a626e93-8.2.0
Traceback (most recent call last):
  File "/Users/derek/code/work/esp/esp-idf/tools/idf_tools.py", line 1344, in <module>
    main(sys.argv[1:])
  File "/Users/derek/code/work/esp/esp-idf/tools/idf_tools.py", line 1340, in main
    action_func(args)
  File "/Users/derek/code/work/esp/esp-idf/tools/idf_tools.py", line 1135, in action_install
    tool_obj.install(tool_version)
  File "/Users/derek/code/work/esp/esp-idf/tools/idf_tools.py", line 595, in install
    unpack(archive_path, dest_dir)
  File "/Users/derek/code/work/esp/esp-idf/tools/idf_tools.py", line 255, in unpack
    raise NotImplementedError('Unsupported archive type')
NotImplementedError: Unsupported archive type

```
run source ./export.sh
```
Adding ESP-IDF tools to PATH...
WARNING: directory for tool xtensa-esp32s2-elf version esp32s2-dev-4-g3a626e93-8.2.0 is present, but tool was not found
ERROR: tool xtensa-esp32s2-elf has no installed versions. Please run '/Users/derek/code/work/esp/esp-idf/install.sh' to install it.
ERROR: tool esp32ulp-elf has no installed versions. Please run '/Users/derek/code/work/esp/esp-idf/install.sh' to install it.
ERROR: tool openocd-esp32 has no installed versions. Please run '/Users/derek/code/work/esp/esp-idf/install.sh' to install it.
```
cd helloworld and make -j4
```
Toolchain path: /Users/derek/code/work/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: esp32-2019r1
Compiler version: 8.2.0
Python requirements from /Users/derek/code/work/esp/esp-idf/requirements.txt are satisfied.

App "hello-world" version: v4.1-dev-815-ga45e99853-dirty
Generating esp32.project.ld
linker script generation failed for /Users/derek/code/work/esp/esp-idf/components/esp32/ld/esp32.project.ld.in
ERROR: failed to parse /Users/derek/code/work/esp/esp-idf/components/esp_event/linker.lf
required keys ['entries'] for fragment not found (at char 41), (line:2, col:5)
make: *** [/Users/derek/code/work/esp/esp-idf/examples/get-started/hello_world/build/esp32/esp32.project.ld] Error 1
```
run idf.py build
```
linker script generation failed for /Users/derek/code/work/esp/esp-idf/components/esp32/ld/esp32.project.ld.in
ERROR: failed to parse /Users/derek/code/work/esp/esp-idf/components/esp_event/linker.lf
required keys ['entries'] for fragment not found (at char 41), (line:2, col:5)
[2/7] Performing build step for 'bootloader'
[0/1] Re-running CMake...
-- Unexpected file in components directory: /Users/derek/code/work/esp/esp-idf/components/.DS_Store
-- Project version: v4.1-dev-815-ga45e99853-dirty
-- Building ESP-IDF components for target esp32
-- Adding linker script /Users/derek/code/work/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script /Users/derek/code/work/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /Users/derek/code/work/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /Users/derek/code/work/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /Users/derek/code/work/esp/esp-idf/components/bootloader/subproject/main/ld/esp32/bootloader.ld
-- Adding linker script /Users/derek/code/work/esp/esp-idf/components/bootloader/subproject/main/ld/esp32/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp32 esp_common esp_rom esptool_py freertos log main micro-ecc partition_table soc spi_flash xtensa
-- Component paths: /Users/derek/code/work/esp/esp-idf/components/bootloader /Users/derek/code/work/esp/esp-idf/components/bootloader_support /Users/derek/code/work/esp/esp-idf/components/efuse /Users/derek/code/work/esp/esp-idf/components/esp32 /Users/derek/code/work/esp/esp-idf/components/esp_common /Users/derek/code/work/esp/esp-idf/components/esp_rom /Users/derek/code/work/esp/esp-idf/components/esptool_py /Users/derek/code/work/esp/esp-idf/components/freertos /Users/derek/code/work/esp/esp-idf/components/log /Users/derek/code/work/esp/esp-idf/components/bootloader/subproject/main /Users/derek/code/work/esp/esp-idf/components/bootloader/subproject/components/micro-ecc /Users/derek/code/work/esp/esp-idf/components/partition_table /Users/derek/code/work/esp/esp-idf/components/soc /Users/derek/code/work/esp/esp-idf/components/spi_flash /Users/derek/code/work/esp/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/derek/code/work/esp/esp-idf/examples/get-started/hello_world/build/bootloader
[1/1] Generating binary image from built executable
esptool.py v2.9-dev
Generated /Users/derek/code/work/esp/esp-idf/examples/get-started/hello_world/build/bootloader/bootloader.bin
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

dobairoland, 2019-11-07:
Probably this solves this issue: https://github.com/espressif/esp-idf/issues/4270#issuecomment-548549611

isDerek, 2019-11-07:
fixed the NotImplementedError and then run ./install.sh
```
Installing Python packages from /Users/derek/code/work/esp/esp-idf/requirements.txt
/Users/derek/.espressif/python_env/idf4.1_py2.7_env/bin/python: No module named pip
Traceback (most recent call last):
  File "/Users/derek/code/work/esp/esp-idf/tools/idf_tools.py", line 1344, in <module>
    main(sys.argv[1:])
  File "/Users/derek/code/work/esp/esp-idf/tools/idf_tools.py", line 1340, in main
    action_func(args)
  File "/Users/derek/code/work/esp/esp-idf/tools/idf_tools.py", line 1163, in action_install_python_env
    subprocess.check_call(run_args, stdout=sys.stdout, stderr=sys.stderr)
  File "/usr/local/Cellar/python@2/2.7.15_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 190, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/Users/derek/.espressif/python_env/idf4.1_py2.7_env/bin/python', '-m', 'pip', 'install', '--no-warn-script-location', '-r', '/Users/derek/code/work/esp/esp-idf/requirements.txt']' returned non-zero exit status 1``

dobairoland, 2019-11-07:
@isDerek You don't have `pip` installed. Please see https://docs.espressif.com/projects/esp-idf/en/latest/get-started/macos-setup.html#install-prerequisites how to install it.

isDerek, 2019-11-07:
cd helloworld demo and run idf.py build
```
[1/1] Generating binary image from built executable
FAILED: .bin_timestamp 
cd /Users/zhujunkun/esp/esp-idf/examples/get-started/hello_world/build/bootloader && /Users/zhujunkun/.espressif/python_env/idf4.1_py2.7_env/bin/python /Users/zhujunkun/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 elf2image --flash_mode dio --flash_freq 40m --flash_size 2MB -o /Users/zhujunkun/esp/esp-idf/examples/get-started/hello_world/build/bootloader/bootloader.bin bootloader.elf && /usr/local/Cellar/cmake/3.14.5/bin/cmake -E echo "Generated /Users/zhujunkun/esp/esp-idf/examples/get-started/hello_world/build/bootloader/bootloader.bin" && /usr/local/Cellar/cmake/3.14.5/bin/cmake -E md5sum /Users/zhujunkun/esp/esp-idf/examples/get-started/hello_world/build/bootloader/bootloader.bin > /Users/zhujunkun/esp/esp-idf/examples/get-started/hello_world/build/bootloader/.bin_timestamp
/Users/zhujunkun/.espressif/python_env/idf4.1_py2.7_env/bin/python: can't open file '/Users/zhujunkun/esp/esp-idf/components/esptool_py/esptool/esptool.py': [Errno 2] No such file or directory
ninja: build stopped: subcommand failed.
[6/398] Building C object esp-idf/lwip/CMakeFiles/__idf_lwip.dir/lwip/src/core/ipv6/icmp6.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```
git submoudle update is over

dobairoland, 2019-11-07:
Is there a file `/Users/zhujunkun/esp/esp-idf/components/esptool_py/esptool/esptool.py` in you computer? If not then the submodule wasn't initialized.

It can be done by `git submodule update --init --recursive` run from IDF directory.

isDerek, 2019-11-07:
it has done by git submodule update --init --recursive, but the esptool.py  is not created in esptool.

isDerek, 2019-11-07:
perfect work! Thanks.

closed 2019-11-07 by isDerek