espressif/esp-idf #4298: ESP32 ETHERNET PHY Reset Timeout, init mac failed Error

aiden-kim-j, 2019-11-07:
Hello.
I have made a PCB based on the circuit diagram shown in ESP32-ETHERNET-KIT V1.0 developed by your company and I am in the middle of TEST. But there's a problem working on it.
[image]

I think the problem is that the ESP32-WROVER-B product does not come with the GPIO0 signal, the 50Mhz frequency.
I've tried TEST with 25Mhz CRYSTAL on the outside, but so is this. Can you help me?

Information
- Firmware : Ethernet-Example-basic
- Tool : Mingw32
- PHY : IP101GRI
- Schematic : ESP32-ETHERNET-KIT V1.0

suda-morris, 2019-11-08:
If you have referenced our official board design, obviously, your GPIO0 didn't get the RMII clock(50MHz) from outside. So the mac can't be initialized successfully.

Alvin1Zhang, 2019-11-08:
@aiden-kim-j Thanks for reporting.

closed 2019-11-08 by Alvin1Zhang