espressif/esp-idf #4299: Extra space in tool name in tools.json

TommyPec, 2019-11-07:
in tools/tools.json
            "url": " https://dl.espressif.com/dl/toolchains/preview/xtensa-esp32s2-elf-gcc8_2_0-esp32s2-dev-4-g3a626e-macos.tar.gz "

The leading space in the file name causes an error:
```
/Users/whatever/.espressif/dist/xtensa-esp32s2-elf-gcc8_2_0-esp32s2-dev-4-g3a626e-macos.tar.gz 
Extracting /Users/pecos/.espressif/dist/xtensa-esp32s2-elf-gcc8_2_0-esp32s2-dev-4-g3a626e-macos.tar.gz  to /Users/pecos/.espressif/tools/xtensa-esp32s2-elf/esp32s2-dev-4-g3a626e9-8.2.0
Traceback (most recent call last):
  File "/Users/pecos/esp/esp-idf/tools/idf_tools.py", line 1345, in <module>
    main(sys.argv[1:])
  File "/Users/pecos/esp/esp-idf/tools/idf_tools.py", line 1341, in main
    action_func(args)
  File "/Users/pecos/esp/esp-idf/tools/idf_tools.py", line 1136, in action_install
    tool_obj.install(tool_version)
  File "/Users/pecos/esp/esp-idf/tools/idf_tools.py", line 596, in install
    unpack(archive_path, dest_dir)
  File "/Users/pecos/esp/esp-idf/tools/idf_tools.py", line 256, in unpack
    raise NotImplementedError('Unsupported archive type')
NotImplementedError: Unsupported archive type
```

dobairoland, 2019-11-07:
Duplicate of https://github.com/espressif/esp-idf/issues/4297 and https://github.com/espressif/esp-idf/pull/4274. The solution is here https://github.com/espressif/esp-idf/commit/bdb08de87555bd070905d416840d46405ead1ee0

dobairoland, 2019-11-14:
The fix (bdb08de87555bd070905d416840d46405ead1ee0) is now available in the master branch.

closed 2019-11-14 by dobairoland