espressif/esp-idf #4300: Setting I2C clock speed of 1K..2K activates a entirely different clock speed (>5000)

pantaluna, 2019-11-07:
- [ Y ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ N ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ Y ] I have searched the issue tracker for a similar issue and not found a similar issue.

## Environment

- Development Kit: LOLIN D32
- Module or chip used:  ESP32-WROOM-32
- IDF version: v3.3
- Build System: Make
- Compiler version: xtensa-esp32-elf-gcc.exe (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System: Windows
- Power Supply: USB

## Problem Description

A continuation of https://esp32.com/viewtopic.php?f=13&t=13016

The ESP-IDF docs do not specify a minimum I2C clock speed so I assumed that a low speed of for example 1K would be possible. But it is not.

This is the source doc:
```
uint32_t clk_speed;     /*!< I2C clock frequency for master mode, (no higher than 1MHz for now) */
```

### Expected Behavior

Setting a I2C clock speed < 5000 using ESP-IDF should enable approx. that clock speed (5% deviation is fine).

### Actual Behavior

Setting a clock speed of 1000 results in a real clock speed of 5511. See below of other examples.
```
I ran a few tests using a logic analyzer.
Test using separate short 10 centimer wires for VCC GND SCL SDA.
The actual clock speed is inconsistent when specifying values lower than 5K:
	STATUS	I2C CLK_SPEED	SCL MeasuredSpeed
	=======================================
	Error	   100		  5800 !
	Error	  1000		  5511 !
	Error	  2000		 10000 !
	Error	  4000		  3989 !
	Ok	  5000		  4983
	Ok	 20000		 19750
	Ok	100000		 94120
	Ok	400000		320000
```

### Code to reproduce this issue

```cpp
    if (param_ptr_config->manage_i2c_driver == true) {
        // Config
        i2c_config_t i2c_conf = { 0 };
        i2c_conf.mode = I2C_MODE_MASTER;
        i2c_conf.master.clk_speed = param_ptr_config->i2c_clk_speed;
        i2c_conf.scl_io_num = param_ptr_config->i2c_scl_gpio_num;
        i2c_conf.sda_io_num = param_ptr_config->i2c_sda_gpio_num;
        i2c_conf.scl_pullup_en = param_ptr_config->i2c_scl_pullup_en;
        i2c_conf.sda_pullup_en = param_ptr_config->i2c_sda_pullup_en;
        f_retval = i2c_param_config(param_ptr_config->i2c_port_num, &i2c_conf);
        if (f_retval != ESP_OK) {
            ESP_LOGE(TAG, "ABORT. i2c_param_config() error (%i)", f_retval);
            // LABEL
            goto cleanup;
        }
        // ...
    }
```

stickbreaker, 2019-11-08:
The I2C clock is sourced from the APB clock.  The APB clock is either 80MHz or CPU clock, whichever is lower.  The maximum I2C cycle is 32766 APB clocks(16383 for High, 16383 for Low) **PLUS bus transition time** .  

  A typical SCL cycle proceeds in this order. 
* SCL is a pulled `LOW` 
* SCL is monitored UNTIL a logic `LOW` is detected. (timing is dependent on bus capacitance)
* then the peripheral starts counting `scl_low_period` APB clock cycles,
* SCL drive is released, the pullup resistor pulls SCL `HIGH`.
* SCL is monitored until it exceed the `HIGH` logic level (timing is dependent on bus capacitance, and Pullup current.
* after `HIGH` level is detected on SCL the peripheral starts the `scl_high_period` countdown.
 
This sequence is reason for your minimum speed, and why the higher speeds are slower than your selected speed.

Chuck.

negativekelvin, 2019-11-08:
It is still a bug that there are missing bounds checks and there are also instances where a 10-bit value is set to a 14-bit value instead of UINT10MAX.

pantaluna, 2019-11-08:
Thanks.
I have done some more tests and it seems that the lowest I2C clock speed seems to be around 2.5K so I will stick to 3K for this project. It works fine on a 1 meter special TP cable.
```
STATUS  ESP-IDF CLK_SPEED  SCLMeasuredSpeed
===========================================
ERR          50                  2924 !
ERR         100                  5800 !
ERR        1000                  5511 !
ERR        2000                 10990 !
ERR        2250                 28270 !
Ok         2500                  2496 ***
Ok         3000                  2995
Ok         4000                  3989
Ok         5000                  4983
Ok        20000                 19750
Ok       100000                 94120
Ok       400000                320000

```

stickbreaker, 2019-11-08:
@pantaluna Yep, theoretical minimum with CPU clock at 80MHz or above is 2441Hz minus bus transition factor.

  If you needed it to go slower, you would have to reduce CPU clock frequency(but, WiFi and BlueTooth will not function when CPU freq is less than 80MHz), with 10MHz CPU frequency the minimum i2c frequency drops to 305Hz.

As @negativekelvin mentioned, the frequency config code is not error checking correctly, it is trying to store a 14bit value into a register that only holds 10bits.

Chuck.

mythbuster5, 2025-01-21:
This issue is out of date, and on new esp-idf version the clock should be accurate (except esp32 because digital design is not very mature.) I will close it. Feel free to reopen if there is no further update!

closed 2025-01-21 by mythbuster5

mythbuster5, 2025-01-22:
Because esp32 lack some i2c timing register, it has been improved on s2 and c3