espressif/esp-idf #4301: error: 'MB_SLAVE_ID_SHORT' undeclared

keelung-yang, 2019-11-08:
on latest master, if CONFIG_FMB_CONTROLLER_SLAVE_ID_SUPPORT defined

MB_SLAVE_ID_SHORT should be CONFIG_FMB_CONTROLLER_SLAVE_ID

```shell
D:/dev/esp-idf/components/freemodbus/serial_slave/modbus_controller/mbc_serial_slave.c: In function 'mbc_serial_slave_start':
D:/dev/esp-idf/components/freemodbus/serial_slave/modbus_controller/mbc_serial_slave.c:99:28: error: 'MB_SLAVE_ID_SHORT' undeclared (first use in this function); did you mean 'MB_SLAVE_ASSERT'?
     status = eMBSetSlaveID(MB_SLAVE_ID_SHORT, TRUE, (UCHAR*)mb_slave_id, sizeof(mb_slave_id));
                            ^~~~~~~~~~~~~~~~~
                            MB_SLAVE_ASSERT
D:/dev/esp-idf/components/freemodbus/serial_slave/modbus_controller/mbc_serial_slave.c:99:28: note: each undeclared identifier is reported only once for each function it appears in
D:/dev/esp-idf/components/freemodbus/serial_slave/modbus_controller/mbc_serial_slave.c:99:61: error: 'mb_slave_id' undeclared (first use in this function); did you mean 'mbc_slave_init'?
     status = eMBSetSlaveID(MB_SLAVE_ID_SHORT, TRUE, (UCHAR*)mb_slave_id, sizeof(mb_slave_id));
                                                             ^~~~~~~~~~~
                                                             mbc_slave_init
[643/850] Building C object esp-idf/freemodbus/CMakeFiles/__idf_freemodbus.dir/modbus/functions/mbutils.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

```

Alvin1Zhang, 2019-11-08:
@keelung-yang Thanks for reporting, we will look into. Thanks.

alisitsyn, 2019-11-08:
Hi @keelung-yang,
Thank you for this finding. This came after incorrect conflict resolving during rebase. It will be corrected ASAP.

keelung-yang, 2019-11-22:
It's my mistake, MB_SLAVE_ID_SHORT and mb_slave_id defined in another module.

closed 2019-11-22 by keelung-yang

alisitsyn, 2019-11-22:
It is defined in esp_modbus_slave.c module but it was also used in mbc_serial_slave.c by mistake (rebase conflict resolution). Your finding is correct and initialization of slave id was removed from port setup.

gigwegbe, 2020-08-14:
@alisitsyn @keelung-yang I am currently stuck with this error. Please how can I resolve it? Thanks
```
/george/esp/esp-idf/components/esp_common /home/george/esp/esp-idf/components/esp_rom /home/george/esp/esp-idf/components/esptool_py /home/george/esp/esp-idf/components/log /home/george/esp/esp-idf/components/bootloader/subproject/main /home/george/esp/esp-idf/components/bootloader/subproject/components/micro-ecc /home/george/esp/esp-idf/components/partition_table /home/george/esp/esp-idf/components/soc /home/george/esp/esp-idf/components/spi_flash /home/george/esp/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: /home/george/Documents/MLANDAI/tensorflow/tensorflow/lite/micro/tools/make/gen/esp_xtensa-esp32/prj/micro_speech/esp-idf/build/bootloader
[330/920] Performing build step for 'bootloader'
[1/2] Linking C executable bootloader.elf
[2/2] Generating binary image from built executable
esptool.py v2.8
Generated /home/george/Documents/MLANDAI/tensorflow/tensorflow/lite/micro/tools/make/gen/esp_xtensa-esp32/prj/micro_speech/esp-idf/build/bootloader/bootloader.bin
[655/920] Building C object esp-idf/freemodbus/CMakeFiles/__idf_freemodbus.dir/serial_slave/modbus_controller/mbc_serial_slave.c.obj
FAILED: esp-idf/freemodbus/CMakeFiles/__idf_freemodbus.dir/serial_slave/modbus_controller/mbc_serial_slave.c.obj 
/home/george/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc  -Iconfig -I/home/george/esp/esp-idf/components/freemodbus/common/include -I/home/george/esp/esp-idf/components/freemodbus/common -I/home/george/esp/esp-idf/components/freemodbus/port -I/home/george/esp/esp-idf/components/freemodbus/modbus -I/home/george/esp/esp-idf/components/freemodbus/modbus/ascii -I/home/george/esp/esp-idf/components/freemodbus/modbus/functions -I/home/george/esp/esp-idf/components/freemodbus/modbus/rtu -I/home/george/esp/esp-idf/components/freemodbus/modbus/tcp -I/home/george/esp/esp-idf/components/freemodbus/modbus/include -I/home/george/esp/esp-idf/components/freemodbus/serial_slave/port -I/home/george/esp/esp-idf/components/freemodbus/serial_slave/modbus_controller -I/home/george/esp/esp-idf/components/freemodbus/serial_master/port -I/home/george/esp/esp-idf/components/freemodbus/serial_master/modbus_controller -I/home/george/esp/esp-idf/components/newlib/platform_include -I/home/george/esp/esp-idf/components/freertos/include -I/home/george/esp/esp-idf/components/heap/include -I/home/george/esp/esp-idf/components/log/include -I/home/george/esp/esp-idf/components/soc/esp32/include -I/home/george/esp/esp-idf/components/soc/include -I/home/george/esp/esp-idf/components/esp_rom/include -I/home/george/esp/esp-idf/components/esp_common/include -I/home/george/esp/esp-idf/components/xtensa/include -I/home/george/esp/esp-idf/components/xtensa/esp32/include -I/home/george/esp/esp-idf/components/esp32/include -I/home/george/esp/esp-idf/components/driver/include -I/home/george/esp/esp-idf/components/esp_ringbuf/include -I/home/george/esp/esp-idf/components/esp_event/include -I/home/george/esp/esp-idf/components/tcpip_adapter/include -I/home/george/esp/esp-idf/components/lwip/include/apps -I/home/george/esp/esp-idf/components/lwip/include/apps/sntp -I/home/george/esp/esp-idf/components/lwip/lwip/src/include -I/home/george/esp/esp-idf/components/lwip/port/esp32/include -I/home/george/esp/esp-idf/components/lwip/port/esp32/include/arch -I/home/george/esp/esp-idf/components/vfs/include -I/home/george/esp/esp-idf/components/esp_wifi/include -I/home/george/esp/esp-idf/components/esp_wifi/esp32/include -I/home/george/esp/esp-idf/components/esp_eth/include -I/home/george/esp/esp-idf/components/efuse/include -I/home/george/esp/esp-idf/components/efuse/esp32/include -I/home/george/esp/esp-idf/components/app_trace/include -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.0.1\" -DGCC_NOT_5_2_0 -DESP_PLATFORM -MD -MT esp-idf/freemodbus/CMakeFiles/__idf_freemodbus.dir/serial_slave/modbus_controller/mbc_serial_slave.c.obj -MF esp-idf/freemodbus/CMakeFiles/__idf_freemodbus.dir/serial_slave/modbus_controller/mbc_serial_slave.c.obj.d -o esp-idf/freemodbus/CMakeFiles/__idf_freemodbus.dir/serial_slave/modbus_controller/mbc_serial_slave.c.obj   -c /home/george/esp/esp-idf/components/freemodbus/serial_slave/modbus_controller/mbc_serial_slave.c
/home/george/esp/esp-idf/components/freemodbus/serial_slave/modbus_controller/mbc_serial_slave.c: In function 'mbc_serial_slave_start':
/home/george/esp/esp-idf/components/freemodbus/serial_slave/modbus_controller/mbc_serial_slave.c:99:28: error: 'MB_SLAVE_ID_SHORT' undeclared (first use in this function); did you mean 'MB_SLAVE_ASSERT'?
     status = eMBSetSlaveID(MB_SLAVE_ID_SHORT, TRUE, (UCHAR*)mb_slave_id, sizeof(mb_slave_id));
                            ^~~~~~~~~~~~~~~~~
                            MB_SLAVE_ASSERT
/home/george/esp/esp-idf/components/freemodbus/serial_slave/modbus_controller/mbc_serial_slave.c:99:28: note: each undeclared identifier is reported only once for each function it appears in
[... cut ...] b_slave_id' undeclared (first use in this function); did you mean 'mbc_slave_init'?
     status = eMBSetSlaveID(MB_SLAVE_ID_SHORT, TRUE, (UCHAR*)mb_slave_id, sizeof(mb_slave_id));
                                                             ^~~~~~~~~~~
                                                             mbc_slave_init
[664/920] Building CXX object esp-idf/asio/CMakeFiles/__idf_asio.dir/asio/asio/src/asio.cpp.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

```

alisitsyn, 2020-08-14:
@gigwegbe,

You are using intermediate commit of v4.0 and there is still issue with MB_SLAVE_ID_SHORT.
There are two options to avoid it:
1. disable CONFIG_FMB_CONTROLLER_SLAVE_ID_SUPPORT.
2. Apply the changes as below:
freemodbus/common/esp_modbus_slave.c:
```
#ifdef CONFIG_FMB_CONTROLLER_SLAVE_ID_SUPPORT

#define MB_ID_BYTE0(id) ((uint8_t)(id))
#define MB_ID_BYTE1(id) ((uint8_t)(((uint16_t)(id) >> 8) & 0xFF))
#define MB_ID_BYTE2(id) ((uint8_t)(((uint32_t)(id) >> 16) & 0xFF))
#define MB_ID_BYTE3(id) ((uint8_t)(((uint32_t)(id) >> 24) & 0xFF))

#define MB_CONTROLLER_SLAVE_ID (CONFIG_FMB_CONTROLLER_SLAVE_ID)
#define MB_SLAVE_ID_SHORT      (MB_ID_BYTE3(MB_CONTROLLER_SLAVE_ID))

// Slave ID constant
static uint8_t mb_slave_id[] = { MB_ID_BYTE0(MB_CONTROLLER_SLAVE_ID),
                                MB_ID_BYTE1(MB_CONTROLLER_SLAVE_ID),
                                MB_ID_BYTE2(MB_CONTROLLER_SLAVE_ID) };

#endif

/**
 * Start Modbus controller start function
 */
esp_err_t mbc_slave_start(void)
{
    esp_err_t error = ESP_OK;
    MB_SLAVE_CHECK((slave_interface_ptr != NULL),
                    ESP_ERR_INVALID_STATE,
                    "Slave interface is not correctly initialized.");
    MB_SLAVE_CHECK((slave_interface_ptr->start != NULL), 
                    ESP_ERR_INVALID_STATE,
                    "Slave interface is not correctly initialized.");
#ifdef CONFIG_FMB_CONTROLLER_SLAVE_ID_SUPPORT
    // Set the slave ID if the KConfig option is selected
    eMBErrorCode status = eMBSetSlaveID(MB_SLAVE_ID_SHORT, TRUE, (UCHAR*)mb_slave_id, sizeof(mb_slave_id));
    MB_SLAVE_CHECK((status == MB_ENOERR), ESP_ERR_INVALID_STATE, "mb stack set slave ID failure.");
#endif
    error = slave_interface_ptr->start();
    MB_SLAVE_CHECK((error == ESP_OK), 
                    error, "SERIAL slave start failure error=(0x%x).", error);
    return error;
} 
```
serial_slave/modbus_controller/mbc_serial_slave.c:
```
// Start Modbus controller start function
static esp_err_t mbc_serial_slave_start(void)
{
    MB_SLAVE_CHECK((mbs_interface_ptr != NULL),
                    ESP_ERR_INVALID_STATE,
                    "Slave interface is not correctly initialized.");
    mb_slave_options_t* mbs_opts = &mbs_interface_ptr->opts;
    eMBErrorCode status = MB_EIO;
    // Initialize Modbus stack using mbcontroller parameters
    status = eMBInit((eMBMode)mbs_opts->mbs_comm.mode,
                         (UCHAR)mbs_opts->mbs_comm.slave_addr,
                         (UCHAR)mbs_opts->mbs_comm.port,
                         (ULONG)mbs_opts->mbs_comm.baudrate,
                         (eMBParity)mbs_opts->mbs_comm.parity);
    MB_SLAVE_CHECK((status == MB_ENOERR), ESP_ERR_INVALID_STATE,
            "mb stack initialization failure, eMBInit() returns (0x%x).", status);
    status = eMBEnable();
    MB_SLAVE_CHECK((status == MB_ENOERR), ESP_ERR_INVALID_STATE,
            "mb stack set slave ID failure, eMBEnable() returned (0x%x).", (uint32_t)status);
    // Set the mbcontroller start flag
    EventBits_t flag = xEventGroupSetBits(mbs_opts->mbs_event_group,
                                            (EventBits_t)MB_EVENT_STACK_STARTED);
    MB_SLAVE_CHECK((flag & MB_EVENT_STACK_STARTED),
                ESP_ERR_INVALID_STATE, "mb stack start event set error.");
    return ESP_OK;
}
```
Refer to simple_modbus_fixesv40.zip project which includes fixes here: https://esp32.com/viewtopic.php?f=13&t=16685&hilit=modbus