espressif/esp-idf #4302: [Doc] Is there any CHANGELOG.rst or Migrating Guide?

keelung-yang, 2019-11-08:
I'm migrating our project from 3.x to 4.x, but there is no document about it, at least it's not easyly found as in most projects on GitHub.

Alvin1Zhang, 2019-11-08:
@keelung-yang Thanks for reporting, would you please help take a look at the [release notes](https://github.com/espressif/esp-idf/releases)? Thanks.

keelung-yang, 2019-11-08:
> @keelung-yang Thanks for reporting, would you please help take a look at the [release notes](https://github.com/espressif/esp-idf/releases)? Thanks.

@Alvin1Zhang I checked, it helps although far away from a migrating guide. But I almost finished my migrating work. Thanks!

Alvin1Zhang, 2019-11-08:
Thanks for the reporting and feedback, could this ticket be closed? Thanks.

closed 2019-11-08 by keelung-yang

chegewara, 2019-11-08:
https://docs.espressif.com/projects/esp-idf/en/stable/api-guides/build-system-cmake.html#migrating-from-esp-idf-gnu-make-system

keelung-yang, 2019-11-08:
> https://docs.espressif.com/projects/esp-idf/en/stable/api-guides/build-system-cmake.html#migrating-from-esp-idf-gnu-make-system

Thanks for your url. It did help. Seems migrating guides are seperated by topics.