espressif/esp-idf #4307: gattc_multi_connect example doesn't work

paulelong, 2019-11-08:
#4100  Environment

- Development Kit:      HiLetgo ESP_WROOM-32 ESP32 ESP-32s Development Board
- Kit version (for WroverKit/PicoKit/DevKitC): NA
 Module or chip used:  ESP-WROOM-32 ESP32 ESP-32S (this is what amazon says)
- IDF version v4.1-dev-815-ga45e99853
- Build System:        idf.py build
- Compiler versionxtensa-esp32-elf-gcc.exe (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     Windows
- Power Supply:         USB Hub

## Problem Description

I'm motivated to help however I can.

I'm using the gattc_multi_connect example and using it to connect to an HM10 and an HM18.  Occasionally it fails to connect with one of two outcomes listed below under failures I observe.  If I comment out a few lines that print some status (in ESP_GAP_BLE_SCAN_RESULT_EVT), it fails much more frequently.  This leads me to believe there is a threading issue.  The only modifications I made are:

1 so that it connects to 2 devices, not 3.
```cpp
if (conn_device_a && conn_device_b /*&& conn_device_c*/ && !stop_scan_done){
```

And 2, I also changed the Service and Char UUID's to match what the HM BLE devices use.  And then changed the names, which in my case are DSD TECH, which is the default for HM devices, and HarmCtl, which I used AT+NAME to modify.
```cpp
#define REMOTE_SERVICE_UUID        0xFFE0
#define REMOTE_NOTIFY_CHAR_UUID    0xFFE1
...
static const char remote_device_name[3][20] = {"DSD TECH", "HarmCtl", "ESP_GATTS_DEMO_c"};

```

To make the problem happen very frequently (fails 9 of 10 times).  I comment out these lines of code below.  If I don't comment them out, it fails less frequently (maybe 1 every 10 times).
```cpp
            //esp_log_buffer_hex(GATTC_TAG, scan_result->scan_rst.bda, 6);
            //ESP_LOGI(GATTC_TAG, "Searched Adv Data Len %d, Scan Response Len %d", scan_result->scan_rst.adv_data_len, scan_result->scan_rst.scan_rsp_len);
            adv_name = esp_ble_resolve_adv_data(scan_result->scan_rst.ble_adv,
                                                ESP_BLE_AD_TYPE_NAME_CMPL, &adv_name_len);
            //ESP_LOGI(GATTC_TAG, "Searched Device Name Len %d", adv_name_len);
            //esp_log_buffer_char(GATTC_TAG, adv_name, adv_name_len);
            //ESP_LOGI(GATTC_TAG, "\n");
```

The failures I observe are:
1. Usually after one device is connected, the debug logging I've enabled indicates a disconnect.  I've included the debug output below.

```
I (12910) GATTC_MULTIPLE_DEMO: A UUID16: ffe0
I (13310) GATTC_MULTIPLE_DEMO: A write descr success
I (13520) GATTC_MULTIPLE_DEMO: A write char success
I (13520) GATTC_MULTIPLE_DEMO: Scan start success
W (18210) BT_APPL: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=259 reason=0x0008
W (18210) BT_APPL: bta_gattc_conn_cback() - cif=4 connected=0 conn_id=260 reason=0x0008
W (18220) BT_APPL: bta_gattc_conn_cback() - cif=5 connected=0 conn_id=261 reason=0x0008
W (18220) BT_APPL: bta_gattc_conn_cback() - cif=6 connected=0 conn_id=262 reason=0x0008
I (18230) GATTC_MULTIPLE_DEMO: B device disconnect
I (18240) GATTC_MULTIPLE_DEMO: B Unhandled Event 5
E (18240) BT_BTM: BTM_BleScan scan already active
W (18250) BT_APPL:  bta_dm_ble_scan start scan failed. status=0x6

E (18260) GATTC_MULTIPLE_DEMO: Scan start failed
```

2. I also see an GURU Mediation Error.  
```
I (1630) GATTC_MULTIPLE_DEMO: B Unhandled Event 46
I (1660) GATTC_MULTIPLE_DEMO: B Status 0, MTU 251, conn_id 0
I (1660) GATTC_MULTIPLE_DEMO: B SEARCH RES: conn_id = 0 is primary service 1
I (1660) GATTC_MULTIPLE_DEMO: B start handle 11 end handle 65535 current handle value 11
I (1670) GATTC_MULTIPLE_DEMO: B UUID16: ffe0
I (1700) GATTC_MULTIPLE_DEMO: B write descr success
I (1730) GATTC_MULTIPLE_DEMO: Scan start success
I (6100) GATTC_MULTIPLE_DEMO: Searched device DSD TECH
I (6110) GATTC_MULTIPLE_DEMO: Stop scan successfully
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40087337  PS      : 0x00060834  A0      : 0x8008a0e1  A1      : 0x3ffbe240
0x40087337: r_ea_elt_insert at ??:?

A2      : 0x3ffb32c8  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x3ffb80ec  A7      : 0x3ffb3224  A8      : 0x80087264  A9      : 0x3ffbe220  
A10     : 0x3ffb3224  A11     : 0x00000002  A12     : 0x000004e2  A13     : 0x00000000  
A14     : 0x000020c5  A15     : 0x04000000  SAR     : 0x00000017  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000
Core 0 was running in ISR context:
EPC1    : 0x4008ff3b  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40087337
0x4008ff3b: portENTER_CRITICAL_NESTED at E:/users/Paul/Documents/src/esp-idf/components/freertos/include/freertos/portmacro.h:328
 (inlined by) xPortInIsrContext at E:/users/Paul/Documents/src/esp-idf/components/freertos/port.c:338

0x40087337: r_ea_elt_insert at ??:?

ELF file SHA256: 8b8bac867b1f3c2b07ef6373e31b929ed45c96baac9ebcc16381e7729ebdd870

Backtrace: 0x40087334:0x3ffbe240 0x4008a0de:0x3ffbe270 0x40089e79:0x3ffbe290 0x4008af41:0x3ffbe2b0 0x4008bd2b:0x3ffbe2d0 0x40082a45:0x3ffbe2f0 0x4000bfed:0x3ffbd940 |<-CORRUPTED
0x40087334: r_ea_elt_insert at ??:?

0x4008a0de: r_lld_evt_end at ??:?

0x40089e79: r_lld_evt_end_isr at ??:?

0x4008af41: r_rwble_isr at ??:?

0x4008bd2b: r_rwbtdm_isr_wrapper at intc.c:?

0x40082a45: _xt_lowint1 at E:/users/Paul/Documents/src/esp-idf/components/freertos/xtensa_vectors.S:1153

Core 1 register dump:
PC      : 0x401534f6  PS      : 0x00060834  A0      : 0x800d3cb9  A1      : 0x3ffcbf00
0x401534f6: esp_pm_impl_waiti at E:/users/Paul/Documents/src/esp-idf/components/esp32/pm_esp32.c:486

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x80000001  
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x800d36e2  A9      : 0x3ffcbed0
A10     : 0x00000000  A11     : 0x00060623  A12     : 0x00060620  A13     : 0x00000001
A14     : 0x00060620  A15     : 0x3ffcd310  SAR     : 0x00000000  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

ELF file SHA256: 8b8bac867b1f3c2b07ef6373e31b929ed45c96baac9ebcc16381e7729ebdd870

Backtrace: 0x401534f3:0x3ffcbf00 0x400d3cb6:0x3ffcbf20 0x40091971:0x3ffcbf40 0x4008fe41:0x3ffcbf60
0x401534f3: esp_pm_impl_waiti at E:/users/Paul/Documents/src/esp-idf/components/esp32/pm_esp32.c:484

0x400d3cb6: esp_vApplicationIdleHook at E:/users/Paul/Documents/src/esp-idf/components/esp_common/src/freertos_hooks.c:63

0x40091971: prvIdleTask at E:/users/Paul/Documents/src/esp-idf/components/freertos/tasks.c:3381 (discriminator 1)

0x4008fe41: vPortTaskWrapper at E:/users/Paul/Documents/src/esp-idf/components/freertos/port.c:143

Rebooting...
```
### Expected Behavior

I would expect it to connect every time to both devices.  And while it works decently with the original code, a small change make it unstable.  As I'm deriving my code from the example, I've found it impossible so far to get a stable connection.  I believe there is a threading issue which makes the example work fairly consistently as written, but if you introduce a deviation, like the one I  presented above you get consistent issues.  At this point I'm unable to make any progress with development.

### Actual Behavior

With the small, non functional changes I made, the two BLE devices fail to connect most of the time.  I have to re-run the ESP32 program (by running idf.py -p com7 monitor) 10-20 times before it gets connected.  As I'm trying to develop using this example as a base, even after running 10-20 times to get a stable connection, I run into other issues that I can't tell if they are related or not.

### Steps to repropduce

1. Power an HM10 and HM18 and configure so they advertise (which is the default mode).
2. Make the code changes mentioned above.
3. Run the program and hopefully notice the issues I've mentioned above.

### Additional problems

Somewhat related is that the example and the tutorial aren't in sync.  It's hard to understand the code flow, especially since it's a set of state changes in the first place.  But the explanation in the tutorial doesn't match.  Perhaps as part of this resolution we could help make the tutorial better.  The current issues I see are .
1. The diagram indicates that devices are scanned for first, and then the connections are made.  But this doesn't follow the code, which connects once it finds a device.  
2. The tutorial shows that ESP_GATTC_NOTIFY_EVT does a esp_bel_gattc_write_char, but that's not what the real code shows.
3. In ESP_GATTC_WRITE_DESCR_EVT, it creates an array of uint8_t data from 0-35 and writes it.  Why does it do this?  Is it required and can it be explained?

gengyuchao, 2019-11-22:
The disconnection reason 0x08 on the client is connection timeout. It is generally caused by the environment or caused by the peer device.  The log you provide is normal. The peer device disconnected abnormally, so the client cannot find the peer device and report a timed out warning. I want to know the disconnect reason about your server device. If it is a problem with the peer device, you can try to change the device. And I haven't any crashed yet, can you provide more details? Thank you.

paulelong, 2019-12-03:
I've worked around the disconnect, however this magnifies the crash issues because the more times it tries to connect, the more times it might crash.

To reproduce the crash, I don't do anything more than run a few times using 2 devices.  What I have found recently is that for some reason a Ble-Uno, which has a bluetooth BLE built-in causes a crash much more frequently, but the crash is different, so I'm adding that below.

The precise hardware I'm using to reproduce is using 2 of the 3 combinations below:

1. Ble-uno
2. DSD-Tech HM18 configured with a Bela.io board (which is a pocket begal with a sheild).  Nothing too special, as the HM18 is connect to a UART, and no software needs to be running.  My guess it would happen if I just powered up the HM18.
3. DSD-Tech HM10 connected to a Nano.  Again, I think just powering it up is enough.

Please let me know what other details would be helpful.

The other crash I see is:

```
I (1714) GATTC_MULTIPLE_DEMO: Scan start success
I (5574) GATTC_MULTIPLE_DEMO: Searched device Ble-Uno
I (5584) GATTC_MULTIPLE_DEMO: Stop scan successfully
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40014f24  PS      : 0x00060c34  A0      : 0x80088de8  A1      : 0x3ffbe230  
A2      : 0x3ffb3324  A3      : 0x68db8bad  A4      : 0x000004e2  A5      : 0x00000000  
A6      : 0x00001cab  A7      : 0x04000000  A8      : 0xfffffffd  A9      : 0x00000005  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000578  A13     : 0x00000578  
A14     : 0x04000000  A15     : 0x259a004e  SAR     : 0x00000013  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff
Core 0 was running in ISR context:
EPC1    : 0x40092e8c  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40014f24
0x40092e8c: portENTER_CRITICAL_NESTED at E:/users/Paul/Documents/src/esp-idf/components/freertos/include/freertos/portmacro.h:328
 (inlined by) xPortInIsrContext at E:/users/Paul/Documents/src/esp-idf/components/freertos/port.c:338

ELF file SHA256: e0cc62dd855347aa2acb9e2d660521643c67820492d685aa574c70e20c19ed1c

Backtrace: 0x40014f21:0x3ffbe230 |<-CORRUPTED

Core 1 register dump:
PC      : 0x40140435  PS      : 0x00060e34  A0      : 0x800d3e55  A1      : 0x3ffccb20
0x40140435: esp_pm_impl_waiti at E:/users/Paul/Documents/src/esp-idf/components/esp32/pm_esp32.c:486

A2      : 0x00000000  A3      : 0x259a004e  A4      : 0x00000001  A5      : 0x80000001  
A6      : 0x400d240c  A7      : 0x00000000  A8      : 0x3ffcb434  A9      : 0x259a004e
0x400d240c: dport_access_init_core at E:/users/Paul/Documents/src/esp-idf/components/esp32/dport_access.c:159

A10     : 0x00000000  A11     : 0x00060e23  A12     : 0x00060e20  A13     : 0x00000001  
A14     : 0x00060e20  A15     : 0x00060023  SAR     : 0x00000000  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff

ELF file SHA256: e0cc62dd855347aa2acb9e2d660521643c67820492d685aa574c70e20c19ed1c

Backtrace: 0x40140432:0x3ffccb20 0x400d3e52:0x3ffccb50 0x400950f7:0x3ffccb80 0x40092caa:0x3ffccbb0
0x40140432: esp_pm_impl_waiti at E:/users/Paul/Documents/src/esp-idf/components/esp32/pm_esp32.c:484

0x400d3e52: esp_vApplicationIdleHook at E:/users/Paul/Documents/src/esp-idf/components/esp_common/src/freertos_hooks.c:63

0x400950f7: prvIdleTask at E:/users/Paul/Documents/src/esp-idf/components/freertos/tasks.c:3381 (discriminator 1)

0x40092caa: vPortTaskWrapper at E:/users/Paul/Documents/src/esp-idf/components/freertos/port.c:143

I (6025) esp_core_dump_flash: Save core dump to flash...
```

gengyuchao, 2019-12-07:
This doesn't look like a ble problem. What did you do after scanning the device?

paulelong, 2019-12-08:
I've upload some code that reproduces the problem, https://github.com/paulelong/gattc_multi_connect.  The difference from the published example is minimal, but now I believe the main cause is sdkconfig.  I suppose the next step I could take is to narrow down which setting(s) cause the crash, but I'm not sure why configuration should change the behavior.  Can you reproduce the issue using my code?

gengyuchao, 2019-12-09:
Hello, because I don't have the Ble-uno or HM. I have tried you code on my esp32 device. But I still can not find any crash, it is work fine. Can you help open and provide you full log ?  That may be more helpful. And for the example you provide I want to know if your server device has a corresponding service and Notify char. And whether your service is writable. Because in this example, the client will exchange the MTU after connecting to the server. After the exchange of the MTU is successful, it will search the specified service, write data to the corresponding char, and register Notify.
I don’t know where the problem is or the connection is unstable and disconnected after successful connection.
And you can also try to change the server device's interval to a larger number. Maybe it can stable the connection.

paulelong, 2019-12-09:
I've attached the log of one reboot cycle.  I've also attached pictures which uses an android app to display the DSD TECH Service information.  As far as I understand the 0xFFE0 services characteristic 0xFFE1 is Read, Notify, Write without Response.

I can try other interval parameters., but I have a question on how to do that.  From what I understand the peripherals/server, in my case the HM10 and HM18 and BLE Uno provide a Link Layer connection interval min/max values.  But in my case the ESP32 (client/master), has the final say on the interval.  Is this something I have to code directly, collecting the min/max values from each peripheral and setting them, or can I just configure each peripheral with a larger min and max, and the ESP32 will figure it out for me?

Also I wanted to be sure that you tried to reproduce the problem with two peripherals and the MOD1 branch which has more logging enabled?
[GatExLog.txt](https://github.com/espressif/esp-idf/files/3940373/GatExLog.txt)
[image]

gengyuchao, 2019-12-10:
Hello, from you log, I found that your version between IDF and controller does not match. So Please try to use `git submodule update --init` first. And hope you problem can be solved. :)

paulelong, 2019-12-13:
I did an update as you mentioned and I still have the issue.  The result of the git command didn't seem to do anything significant. 

```
E:\users\paul\documents\src\esp-idf
PS> git submodule update --init
Submodule 'components/esp_wifi/lib_esp32s2beta' (https://github.com/espressif/esp32-wifi-lib.git) registered for path 'components/esp_wifi/lib_esp32s2beta'
Cloning into 'E:/users/Paul/documents/src/esp-idf/components/esp_wifi/lib_esp32s2beta'...
Submodule path 'components/esp_wifi/lib_esp32s2beta': checked out '8ea1c651254560c59f816c9d16aa71322dcd15e5'
```

Looking at the the line in the log "I (32) boot: ESP-IDF v4.1-dev-815-ga45e99853-dirty 2nd stage bootloader", it doesn't look different than before.  Is this the version you are looking at?  What should it be?

I've attached the new log.
[GatExLog2.txt](https://github.com/espressif/esp-idf/files/3961127/GatExLog2.txt)

paulelong, 2019-12-13:
I've successfully updated, but the problem still occurs :(
[GatExLog3.txt](https://github.com/espressif/esp-idf/files/3961644/GatExLog3.txt)
.   I've included a new log which shows the version: v4.1-dev-815-ga45e99853-dirty.    BTW: How does one learn about new versions?

I can try still try different intervals on the peripherals, but even if that worked, the system shouldn't crash, right?  That kind of solution doesn't sound very robust, though I understand if it's just a tshooting step.  In that regard I've got a new Beetle with BLE, so I'm going to see if that works differently as another data point

gengyuchao, 2019-12-15:
Hello, `ESP-IDF v4.1-dev-815-ga45e998534` should be with `BT controller compile version [6decaaf]`.
From the GatExLog3.txt ,I find `ESP-IDF v4.1-dev-1086-g93a8603c5 2nd stage bootloader` with `I (705) BTDM_INIT: BT controller compile version [340d4d4]` , controller and IDF are matched. 
As you said, no matter what the interval is, IDF should not crash. That is what the device performs in my environment. And I will try to analyze the possible cause based on the IDF-93a8603c5 log you provided.
It would be of great help if both your master and slave can use ESP32 and reproduce this problem.

paulelong, 2019-12-16:
I"m not sure I'll be able to buy two more esp32 to reproduce the issue.  Is it better to test with other types of BLE because that would be a common use case, and potentially related to the issue?  BTW, do you work for Espressif?

Perhaps another clue is that once the connections to both devices are complete, it becomes much more stable.  The actual project includes a 128x128 OLED screen which receives and sends BLE to the other two BLE devices.  The last time I tested my project, once it's past the connection phase, it has stayed running for at least 45 minutes.

Are there steps I can do on my end to help you understand more about my issue?

gengyuchao, 2019-12-17:
Yes, it would be better to use more other types of BLE devices. Because I only have ESP32 devices, I can only use it for testing, and they did not show the crash you said. I'm also interested in trying out other equipment, after my work is done. :)
Also I have some suggestions you can try.
1. Increase interval to see if it will increase the success rate of the connection. Whether the crash will still occur.
2. There is a `Timeout for RTC watchdog (ms)` option in `Bootloader config` of menuconfig. You can try to double this value to see whether the crash will still occur. I also want to know your CPU frequency.

paulelong, 2019-12-17:
I tried everything above and it still crashes.  I've left details below about the interval settings I used.  I'm not an expert on BLE, so it would help if you validate that I'm following your instructions.  I did the following, in order.

I've increased the interval of both BLE devices.  First I increased the Minimum and Maximum communication interval on the Emakefun Nano BLE using the AT+MINI_INTERVAL and AT+MAX_INTERVAL commands.  The defaults are 6ms for both.  I increased to 20 and 40 respectively.  Then I ran my program and it caused the reported crash each time.

Then I increased the "Minimum and Maximum Link Layer connection interval, AT+COMI and AT_COMA on the HM18.  The defaults are 7.5ms and 20ms and I increased to 40ms and 4000ms respectively.  I ran the program again, and it also crash each time.

Finally I increased two more HM18 settings called Link Layer connection slave latency and MIN Link Layer supervision connection MAX.  The defaults were 0 and 6000ms.  I set the MIN to 4, the documentation doesn't say what these represent.  The MAX was already at the highest.  Again I tested, and again it crashed consistently.

I changed the watchdog timeout from 9000 to 18000.  That didn't change the outcome.

The frequency was set to 160MHz, I tried at 240Mhz and 80Mhz, but the app still crashed with every setting I used.

Please let me know what the next step is, thanks.

gengyuchao, 2019-12-18:
That confuses me. Are they have the same crash log? Because of the log before I find it crashes in the 'watchdog'. But we increased the watchdog time, it still crashes. That means it is not a performance problem. I have no other ideas, I will continue to check this issue with the log.

paulelong, 2019-12-18:
Made some good progress.  By going through the differences in the two SDKConfigs, I was able to determine that turning off the  "BLE full scan feature supported" option makes the problem go away.  I made this change in my main project and now all 3 BLE devices can connect at once.   So that option either makes the problem happen more frequently.  I can be sure it's completely gone, but I will monitor over the next few days.  

To  answer your questions, the stack traces were the same.  The only difference is that the stack is corrupted, perhaps because i turned off stack/head protection options that were previously enabled in hopes of finding other issues.  I've attached a log that many of the stack traces as I one-by-one remove an SDK option that is not set in the working master repro.

[GatExLog4.txt](https://github.com/espressif/esp-idf/files/3980225/GatExLog4.txt)

I'm not sure how the WD timer is related to the issue, but I could increase it more if you want to try that.

GYC-Espressif, 2019-12-19:
Thank you for your effort. We have known some issues with the full scan function. And we have fixed some of them. But now we can not reproduce the crash you have. Could you please give me the sdkconfig file you used, so we can try to find the actual problem.

paulelong, 2019-12-19:
The project is on github, as I mentioned earlier (https://github.com/paulelong/gattc_multi_connect.).  Isn't that what you are are reproducing with?

Currently the branch "partition" is in a reproducible state.  The "mod1" branch is working, and so is "master".  I've attached a static copy of sdkconfig as well (renamed to .txt so github will accept).

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/3983931/sdkconfig.txt)

GYC-Espressif, 2020-04-26:
Hello, Just for your information, we have refactor the `full scan` function after commit  `7a7c4a5486d85242a6` in the master branch, that may solve many crash problem related. Thanks for your contribution again.

Alvin1Zhang, 2020-05-27:
@paulelong Thanks for reporting. Would you please help share if any more updates about the issue? Has your issue been resolved? Thanks.

Alvin1Zhang, 2020-06-09:
Thanks for reporting and updates, will close the ticket. Feel free to reopen if the issue still happens.

paulelong, 2020-09-07:
I've just updated to 4.1 and the problem still happens consistently.  I think the example above should still reproduce the problem.

paulelong, 2020-09-07:
@Alvin1Zhang , I guess I can't reopen the issue myself?

paulelong, 2020-09-09:
@Alvin1Zhang , sorry I realized this is not related to the issue I'm seeing consistently.  I will open up another issue for that, though I thought I already did.  You can close this again :)

closed 2020-09-10 by Alvin1Zhang