espressif/esp-idf #4310: ESP32 stuck at checksum error after secure boot enabled

akshar001, 2019-11-10:
Hello,
This might not be an issue rather the problem for secure boot enable process.
I have esp-idf version 3.3dirty.
ESP32-WROVER-B (8MB PSRAM).

I was trying to enable secure boot in the process of production devices.
https://docs.espressif.com/projects/esp-idf/en/v3.3/security/secure-boot.html
i have followed the same steps as this document has suggested.

but the thing is when i try to another program it's actually being uploaded.

Efuse summary.
Connecting........_
Security fuses:
FLASH_CRYPT_CNT        Flash encryption mode counter                     = 0 R/W (0x0)
FLASH_CRYPT_CONFIG     Flash encryption config (key tweak bits)          = 0 R/W (0x0)
CONSOLE_DEBUG_DISABLE  Disable ROM BASIC interpreter fallback            = 1 R/W (0x1)
ABS_DONE_0             secure boot enabled for bootloader                = 0 R/W (0x0)
ABS_DONE_1             secure boot abstract 1 locked                     = 0 R/W (0x0)
JTAG_DISABLE           Disable JTAG                                      = 0 R/W (0x0)
DISABLE_DL_ENCRYPT     Disable flash encryption in UART bootloader       = 0 R/W (0x0)
DISABLE_DL_DECRYPT     Disable flash decryption in UART bootloader       = 0 R/W (0x0)
DISABLE_DL_CACHE       Disable flash cache in UART bootloader            = 0 R/W (0x0)
BLK1                   Flash encryption key                              
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W 
BLK2                   Secure boot key                                   
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 -/- 
BLK3                   Variable Block 3                                  
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W 

Efuse fuses:
WR_DIS                 Efuse write disable mask                          = 256 R/W (0x100)
RD_DIS                 Efuse read disablemask                            = 2 R/W (0x2)
CODING_SCHEME          Efuse variable block length scheme                = 0 R/W (0x0)
KEY_STATUS             Usage of efuse block 3 (reserved)                 = 0 R/W (0x0)

Config fuses:
XPD_SDIO_FORCE         Ignore MTDI pin (GPIO12) for VDD_SDIO on reset    = 0 R/W (0x0)
XPD_SDIO_REG           If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset   = 0 R/W (0x0)
XPD_SDIO_TIEH          If XPD_SDIO_FORCE & XPD_SDIO_REG, 1=3.3V 0=1.8V   = 0 R/W (0x0)
SPI_PAD_CONFIG_CLK     Override SD_CLK pad (GPIO6/SPICLK)                = 0 R/W (0x0)
SPI_PAD_CONFIG_Q       Override SD_DATA_0 pad (GPIO7/SPIQ)               = 0 R/W (0x0)
SPI_PAD_CONFIG_D       Override SD_DATA_1 pad (GPIO8/SPID)               = 0 R/W (0x0)
SPI_PAD_CONFIG_HD      Override SD_DATA_2 pad (GPIO9/SPIHD)              = 0 R/W (0x0)
SPI_PAD_CONFIG_CS0     Override SD_CMD pad (GPIO11/SPICS0)               = 0 R/W (0x0)
DISABLE_SDIO_HOST      Disable SDIO host                                 = 0 R/W (0x0)

Identity fuses:
MAC                    MAC Address                                       
  = 84:0d:8e:2d:97:50 (CRC e4 OK) R/W 
CHIP_VERSION           Chip version                                      = 10 R/W (0xa)
CHIP_PACKAGE           Chip package identifier                           = 0 R/W (0x0)

projectgus, 2019-11-11:
Hi @akshar001 ,

In order for us to help you, please provide some more details about what you're doing, and exactly what is happening (ie exact commands which were run, full output from the build process and ESP32 serial, expected behaviour, actual behaviour, etc).

From the efuse summary you've sent it looks like a Secure Boot key has been burned to efuse but secure boot is not yet enabled (ABS_DONE_0 efuse has value 0). Is that what you expect?

Whatever problem you are seeing (please provide some detail about what this problem is), may not be exactly related to secure boot.

akshar001, 2019-11-12:
Yes i know tif i need to run the make bootloader and then upload the bootloader first and then run it again. The problem is i . efuse has been written so the secure boot mode is on but i can compile and run arduino code on it after erasing whole the flash. In theory once secure boot is enabled it will need to run encyrpted bootloader only. (The bootloader with encyrption key). I don't understand it why?

projectgus, 2019-11-21:
Hi @akshar001 ,

Sorry for the delayed reply.

> efuse has been written so the secure boot mode is on 

The efuse dump suggests that the secure boot mode is not on:

> ...
> ABS_DONE_0 secure boot enabled for bootloader = 0 R/W (0x0)
> ...

If this efuse is 0, secure boot is still disabled. This is why you're able to run Arduino sketches.

Is it possible that the secure boot enable process in the IDF bootloader didn't complete successfully?

Without more details of exactly what steps were followed, it's not possible to debug the issue.

akshar001, 2019-11-22:
Hello @projectgus tell me what details you will need i will do it.

projectgus, 2019-11-22:
* What configuration items did you set in the sdkconfig?
* What commands did you run to build/flash the device?
* What output did you see from these commands?
* What output do you see from monitoring the boot process ("make monitor" or "idf.py monitor" or another serial terminal monitor)?

As many details as possible, please.

From the output you've posted so far, it looks like Secure Boot failed to enable for some reason.

Alvin1Zhang, 2020-01-08:
@akshar001 Thanks for reporting. Would you please help share if any updates or more details for this issue? Thanks.

projectgus, 2020-04-30:
Closing due to lack of updates.

closed 2020-04-30 by projectgus