espressif/esp-idf #4311: ESP-NOW Examples crash during runtime (esp_wifi_set_channel failure)

alexanderturner, 2019-11-10:
## Environment

- Development Kit:      ESP32-Wrover-Kit|
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.1-dev-474-g2e6398aff
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     macOS
- Power Supply:        N/A

## Problem Description

ESP-NOW example in master fails at `esp_wifi_set_channel` command in both `station` and `softAp` mode.

### Expected Behavior
ESP-NOW examples to work as per Readme. For esp_wifi_set_channel to execute without error.

### Actual Behavior
```
E (747) wifi: esp_wifi_set_channel 1454 wifi not start or sniffer not enabled, should start wifi or enable sniffer first
ESP_ERROR_CHECK failed: esp_err_t 0x3002 (ESP_ERR_WIFI_NOT_STARTED) at 0x4008b7ac
```

### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue
```
cd $IDF_PATH/examples/wifi/espnow
make flash monitor
```

## Debug Logs

```
I (393) cpu_start: App cpu up.
I (418) heap_init: Initializing. RAM available for dynamic allocation:
I (424) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (430) heap_init: At 3FFB9228 len 00026DD8 (155 KiB): DRAM
I (437) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (443) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (449) heap_init: At 400905C4 len 0000FA3C (62 KiB): IRAM
I (456) cpu_start: Pro cpu start user code
I (474) spi_flash: detected chip: generic
I (475) spi_flash: flash io: dio
I (475) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (577) wifi: wifi driver task: 3ffc0b18, prio:23, stack:3584, core=0
I (577) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (577) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (607) wifi: wifi firmware version: 932678a
I (607) wifi: config NVS flash: enabled
I (607) wifi: config nano formating: disabled
I (607) wifi: Init dynamic tx buffer num: 32
I (607) wifi: Init data frame dynamic rx buffer num: 32
I (617) wifi: Init management frame dynamic rx buffer num: 32
I (617) wifi: Init management short buffer num: 32
I (627) wifi: Init static rx buffer size: 1600
I (627) wifi: Init static rx buffer num: 10
I (637) wifi: Init dynamic rx buffer num: 32
I (727) phy: phy_version: 4102, 2fa7a43, Jul 15 2019, 13:06:06, 0, 0
I (727) wifi: mode : softAP (80:7d:3a:8f:73:d5)
I (727) wifi: Total power save buffer number: 16
I (737) wifi: Init max length of beacon: 752/752
I (737) wifi: Init max length of beacon: 752/752
E (747) wifi: esp_wifi_set_channel 1454 wifi not start or sniffer not enabled, should start wifi or enable sniffer first
ESP_ERROR_CHECK failed: esp_err_t 0x3002 (ESP_ERR_WIFI_NOT_STARTED) at 0x4008b7ac
```

## Other items if possible
[espnow_error.zip](https://github.com/espressif/esp-idf/files/3828338/espnow_error.zip)

```
E (747) wifi: esp_wifi_set_channel 1454 wifi not start or sniffer not enabled, should start wifi or enable sniffer first
ESP_ERROR_CHECK failed: esp_err_t 0x3002 (ESP_ERR_WIFI_NOT_STARTED) at 0x4008b7ac
0x4008b7ac: _esp_error_check_failed at /Users/aturner/Code/Hardware/esp/esp-idf/components/esp32/panic.c:739

file: "/Users/aturner/Code/Hardware/esp/esp-idf/examples/wifi/espnow/main/espnow_example_main.c" line 56
func: example_wifi_init
expression: esp_wifi_set_channel(CONFIG_ESPNOW_CHANNEL, 0)

ELF file SHA256: 2c4833bb6dade5e36082cee79fbc0b4836774d831ee44cef4991977a2e70f0fb

Backtrace: 0x4008b221:0x3ffbaf30 0x4008b7af:0x3ffbaf50 0x400d2acd:0x3ffbaf70 0x400d3236:0x3ffbb030 0x400d0fba:0x3ffbb050 0x4008d759:0x3ffbb070
0x4008b221: invoke_abort at /Users/aturner/Code/Hardware/esp/esp-idf/components/esp32/panic.c:734

0x4008b7af: _esp_error_check_failed at /Users/aturner/Code/Hardware/esp/esp-idf/components/esp32/panic.c:740

0x400d2acd: example_wifi_init at /Users/aturner/Code/Hardware/esp/esp-idf/examples/wifi/espnow/main/espnow_example_main.c:56 (discriminator 1)

0x400d3236: app_main at /Users/aturner/Code/Hardware/esp/esp-idf/examples/wifi/espnow/main/espnow_example_main.c:373

0x400d0fba: main_task at /Users/aturner/Code/Hardware/esp/esp-idf/components/esp32/cpu_start.c:569

0x4008d759: vPortTaskWrapper at /Users/aturner/Code/Hardware/esp/esp-idf/components/freertos/port.c:403

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:7008
```

alexanderturner, 2019-11-11:
Still happening in v3.3 it'd seem `v3.3-127-g0a0f2caa1`
```cpp
I (372) espnow_example: WiFi started
ESP_ERROR_CHECK failed: esp_err_t 0x3002 (ESP_ERR_WIFI_NOT_STARTED) at 0x400876c4
```

liuzfesp, 2019-11-12:
@alexanderturner this is NOT bug, in the latest v3.3, we added the limitation to esp_wifi_set_channel(), it can be called only both of following conditions are met:

1. esp_wifi_start() is returned
2. sniffer is enabled

alexanderturner, 2019-11-12:
Thanks @liuzfesp!

The issue is that the *example* crashes during runtime; the example provided in ESP-IDF. The issue is to correct the example.

Alex

alexanderturner, 2019-11-14:
Really need some help with this one - any follow up from Espressif?

negativekelvin, 2019-11-14:
> this is NOT bug, in the latest v3.3, we added the limitation to esp_wifi_set_channel(), it can be called only both of following conditions are met:
> 
> 1. esp_wifi_start() is returned
> 2. sniffer is enabled

Should always work unless STA mode is connected to an AP or a scan is in progress.  Required for espnow and esp_wifi_80211_tx.

https://github.com/espressif/esp-idf/issues/4005

alexanderturner, 2019-11-16:
Thanks @negativekelvin any chance you can try run the example and see what's happening. I've tried to get espnow running a few times now and I'm not having any luck at all - I'd love to avoid spinning up a BLE stack if possible

alexanderturner, 2019-11-21:
Anything here guys? @negativekelvin @liuzfesp

negativekelvin, 2019-11-21:
For now, fix is to just not set the channel https://github.com/espressif/esp-idf/commit/b9da410cbb037aedcea1deeee8b0e2d72262603c#diff-71d4a3ad5cf2dcb1e3f6bd8eff374134

But hopefully it will be fixed later so that user can set any channel

alexanderturner, 2019-11-21:
@negativekelvin I'm trying to use this to discover ESP32's, is it possible to discover them without setting the channel?

negativekelvin, 2019-11-21:
If they all stay on default channel

alexanderturner, 2019-11-21:
Thanks - not sure if I can guarantee that. Looking to discover ESP's then have them join the 'master's' AP which may be on a different channel. Makes it hard to gaurantee that they'll stay on the same channel. 

Is there another way to change the channel? Even if I bounce it into another mode - this is very valuable to me as it prevents me from enabling the BLE stack

zhangyanjiaoesp, 2019-11-21:
@alexanderturner 

> I'm trying to use this to discover ESP32's

I'm sorry I can't understand you. Do you mean you  want to discover an ESP32 with ESP-NOW ?

alexanderturner, 2019-11-21:
Hi @zhangyanjiaoesp that's right. In a network of ESP32's, I want a "master" to be able to discover a "slave" so it can adopt it.

zhangyanjiaoesp, 2019-11-21:
@alexanderturner 
You mean your "slaves" (or "master")have connected with other APs, and they are in different channels, you want to use 'esp_wifi_set_channel()' to change the channel of the "master" (or "slave") , then "master" can discover them, is that right?

alexanderturner, 2019-11-21:
Correct @zhangyanjiaoesp

zhangyanjiaoesp, 2019-11-22:
@alexanderturner 
Currently, you can only call `esp_wifi_set_channel()` after wifi initialized and sniffer enabled. We are trying to modify this function to let it can set the channel as long as the wifi initialized, thus you can set the channel in STA mode. But we will still don't support setting channel when the STA has connected to an AP, or when STA is scanning, or STA is connecting to an AP.

lgnativs, 2019-11-30:
我碰到了同样的情况
代码如下:
```
 if(!radio_inited) {
        esp_initialize();
        if(!wifi_started) {
            ESP_LOGE("radio", "init and start wifi");
            wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
            ESP_ERROR_CHECK(esp_wifi_init(&cfg));
            ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_RAM));
            ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
            ESP_ERROR_CHECK(esp_wifi_start());
            wifi_started = true;
            ESP_LOGE("radio", "wifi started!");
        }
        ESP_LOGE("radio", "wifi status = %d", wifi_started);
        ESP_ERROR_CHECK(esp_wifi_set_channel(radio_channel, 0));
```
错误信息:
```
>>> radio.on()
E (17260) radio: init and start wifi
I (17290) wifi: wifi driver task: 3ffd31fc, prio:23, stack:3584, core=0
I (17732) wifi: wifi firmware version: 10f4364
I (17742) wifi: config NVS flash: enabled
I (17742) wifi: config nano formating: disabled
I (17742) wifi: Init dynamic tx buffer num: 32
I (17742) wifi: Init data frame dynamic rx buffer num: 32
I (17752) wifi: Init management frame dynamic rx buffer num: 32
I (17752) wifi: Init management short buffer num: 32
I (17762) wifi: Init static rx buffer size: 1600
I (17762) wifi: Init static rx buffer num: 10
I (17762) wifi: Init dynamic rx buffer num: 32
I (18032) wifi: mode : sta (3c:71:bf:9b:be:58)
E (18032) radio: wifi started!
E (18032) radio: wifi status = 1
E (18032) wifi: esp_wifi_set_channel 1464 wifi not start or sniffer not enabled, should start wifi or enable sniffer first
ESP_ERROR_CHECK failed: esp_err_t 0x3002 (ESP_ERR_WIFI_NOT_STARTED) at 0x40091520
file: "builtins/modradio.c" line 65
func: radio_enable
expression: esp_wifi_set_channel(radio_channel, 0)
```

同样的代码用IDF 3.3的版本是没问题的,但是现在的4.0的就报错,这个功能对我的项目很重要,设定channel功能也是必要的.

请问什么时候能修复这个问题呢?

zhangyanjiaoesp, 2019-12-02:
I think it will be merged into IDF this month.

justoke, 2019-12-27:
I have the same error/issue using the ESP32 on PlatformIO under the Arduino framework.

```
 tcpip_adapter_init();
      wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
      ESP_ERROR_CHECK(esp_wifi_init(&cfg));
      ESP_ERROR_CHECK(esp_wifi_set_country(&wifi_country));
      ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_RAM));
      ESP_ERROR_CHECK(esp_wifi_set_mode(ESPNOW_WIFI_MODE));
      ESP_ERROR_CHECK(esp_wifi_start());
      ESP_ERROR_CHECK(esp_wifi_set_channel(_channel, WIFI_SECOND_CHAN_NONE));

      // Initialize ESPNOW and register sending & receiving callback function
      ESP_ERROR_CHECK(esp_now_init());
      ESP_ERROR_CHECK(esp_now_register_send_cb(espnow_send_cb));
      ESP_ERROR_CHECK(esp_now_register_recv_cb(espnow_recv_cb));

      // Set primary master key
      ESP_ERROR_CHECK(esp_now_set_pmk(_esp_pmk));
```
What needs to be done to make sure the conditions are met please? This is code that used to work.

justoke, 2019-12-28:
@alexanderturner I'm not sure exactly where you should change the code in your example, but I had to do this to get it to work as per the documentation.

```
    //These two steps are required BEFORE the channel can be set
    // As per the documentation from Espressif:
    // https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/network/esp_wifi.html#_CPPv420esp_wifi_set_channel7uint8_t18wifi_second_chan_t
    ESP_ERROR_CHECK(esp_wifi_start());
    ESP_ERROR_CHECK(esp_wifi_set_promiscuous(true));

    ESP_ERROR_CHECK(esp_wifi_set_channel(_channel, WIFI_SECOND_CHAN_NONE));
```

closed 2020-01-06 by espressif-bot