espressif/esp-idf #4312: hello_world example has Windows prohibited directory name

DROBNJAK, 2019-11-10:
I installed esp-idf and for some reason I had to remove it. But Windows 7 refused to remove the hello_world directory. After some digging around eventually it was found that there was a sub directory named 'con' which is a reserved/prohibited directory name in Windows environment. 

Can you possibly use different directory name?

chegewara, 2019-11-10:
https://github.com/espressif/esp-idf/issues/3299

projectgus, 2019-11-11:
Hi @DROBNJAK ,

Which ESP-IDF version are you using? If it is v3.3 (the current stable release) then this bug will be fixed in v3.3.1, as mentioned in the linked issue.

Angus

DROBNJAK, 2019-11-11:
That problem happened with v3.3. Since than I replaced it with the latest ESP-IDF version, I don't know which one it is, but it is higher than v3.3. Problem doesn't affect IDF, it only makes hard deleting that directory when you want to re-install everything.

projectgus, 2019-11-21:
Hi @DROBNJAK ,

Sorry for the delayed reply.

If your first install was v3.3 then it's a known issue that this directory was created and caused this problem with deletion. Sorry for the hassle. As stated above, a fix has been merged and will be in the v3.3.1 release.

Are you still seeing the problem of a "con" directory being created from a newer ESP-IDF install?

Angus

DROBNJAK, 2019-11-23:
Yeah, I upgraded to a newest version and I guess it's fixed now.

closed 2020-04-30 by projectgus