espressif/esp-idf #4313: Compilation error for PPP protocol

Arthedian, 2019-11-10:
## Environment

- IDF version (run ``git describe --tags`` to find it): v4.0-beta2-70-g0a03a55c1
- Build System: CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): crosstool-NG esp32-2019r1) 8.2.0
- Operating System: Fedora 31

## Problem Description

When I compile official example code for SIM800L (PPP protocol), than everything is ok, but once I implement it into my code, than there is a error in compilation

### Expected Behavior

Successful compilation. 

### Actual Behavior

`
../components/gsm/lib/esp_modem.c: In function 'on_ppp_status_changed':
../components/gsm/lib/esp_modem.c:484:38: error: 'dns_getserver(0)' is a pointer; did you mean to use '->'?
         ipinfo.ns1 = dns_getserver(0).u_addr.ip4;
                                      ^
                                      ->
../components/gsm/lib/esp_modem.c:485:38: error: 'dns_getserver(1)' is a pointer; did you mean to use '->'?
         ipinfo.ns2 = dns_getserver(1).u_addr.ip4;
                                      ^
                                      ->
`

suda-morris, 2019-11-11:
FYI, https://github.com/espressif/esp-idf/blob/master/examples/protocols/pppos_client/components/modem/src/esp_modem.c#L468

Arthedian, 2019-11-12:
Oh sorry, thanks.

closed 2019-11-12 by Arthedian