espressif/esp-idf #4315: EspBluF蓝牙配网失败

hscxg, 2019-11-11:
下载esp-idf后, 代码未作任何改, 直接编译blufi项目OK, 烧录到ESP32开发板上, 手机上再运行EspBluFi, 进行蓝牙配网, 

手机上EspBluFi通过蓝牙可以成功连接到ESP32开发板, 但是蓝牙配网失败!

ESP32开发板运行后, 输出Log信息: 
......
I (333) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (363) wifi: wifi firmware version: aeed694
I (363) wifi: config NVS flash: enabled
I (363) wifi: config nano formating: disabled
I (363) wifi: Init dynamic tx buffer num: 32
I (373) wifi: Init data frame dynamic rx buffer num: 32
I (373) wifi: Init management frame dynamic rx buffer num: 32
I (383) wifi: Init management short buffer num: 32
I (383) wifi: Init static rx buffer size: 1600
I (393) wifi: Init static rx buffer num: 10
I (393) wifi: Init dynamic rx buffer num: 32
I (493) phy: phy_version: 4102, 2fa7a43, Jul 15 2019, 13:06:06, 0, 0
I (493) wifi: mode : sta (4c:11:ae:71:18:d0)
I (493) BTDM_INIT: BT controller compile version [e736f10]
I (493) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (803) BLUFI_EXAMPLE: BD ADDR: 4c:11:ae:71:18:d2

I (803) BLUFI_EXAMPLE: BLUFI VERSION 0102

I (803) BLUFI_EXAMPLE: BLUFI init finish

I (1363) wifi: new:<6,2>, old:<1,0>, ap:<255,255>, sta:<6,2>, prof:1
I (2343) wifi: state: init -> auth (b0)
I (2353) wifi: state: auth -> assoc (0)
I (2363) wifi: state: assoc -> run (10)
I (5363) wifi: state: run -> init (2c0)
I (5373) wifi: new:<6,0>, old:<6,2>, ap:<255,255>, sta:<6,2>, prof:1
I (5373) wifi: new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,2>, prof:1
I (7793) wifi: new:<6,2>, old:<6,0>, ap:<255,255>, sta:<6,2>, prof:1
I (7793) wifi: state: init -> auth (b0)
I (7803) wifi: state: auth -> assoc (0)
I (7813) wifi: state: assoc -> run (10)
I (7823) wifi: connected with TP-LINK_4589D0_KBS, channel 6, 40D, bssid = e0:05:c5:45:89:d0
I (7823) wifi: pm start, type: 1

I (8813) event: sta ip: 192.168.1.103, mask: 255.255.255.0, gw: 192.168.1.1
W (8813) BT_GATT: gatt_if out of range [ = 0]
E (8813) BT_APPL: Unknown connection ID: 0 fail sending notification
W (8823) BT_GATT: gatt_if out of range [ = 0]
E (8823) BT_APPL: Unknown connection ID: 0 fail sending notification
W (8833) BT_GATT: gatt_if out of range [ = 0]
E (8833) BT_APPL: Unknown connection ID: 0 fail sending notification
I (13723) BLUFI_EXAMPLE: BLUFI ble connect

App显示信息:
Connected    4C:11:AE:71:18:D2
Discover service and Characteristics success
Set mtu complete, mtu=128
Receive error code -1000.

请问一下, 这是什么原因,  如何解决? 谢谢!

gengyuchao, 2019-11-12:
你是否每次链接都出现相同的情况呢？我在你的分支下进行了实验，没有发现相同的问题。是否能提供更具体的复现方法，或者提供你编译的 bin 和 elf 文件提供给我进行测试。谢谢。

hscxg, 2019-11-13:
是的,  已经试了很多次了, 都是这样,  所需文件见下面附件.
[blufi_demo.zip](https://github.com/espressif/esp-idf/files/3838894/blufi_demo.zip)

gengyuchao, 2019-11-13:
> 是的, 已经试了很多次了, 都是这样, 所需文件见下面附件.
> [blufi_demo.zip](https://github.com/espressif/esp-idf/files/3838894/blufi_demo.zip)

我刚刚使用了你提供的 bin 文件、V1.0  和 V1.2 版本的 APP 进行测试，也能够正常的链接我这里的蓝牙和 WiFi，你能否提供一下 ESP32开发板型号、手机型号和APP版本方便我们确认？谢谢。

hscxg, 2019-11-13:
ESP开发板型号: ESP32-DevkitC-32D
手机型号: 华为P20
EspBluFi App版本: 1.5.0  (BluFi库版本: 2.3.0)

gengyuchao, 2019-11-14:
这个问题我们已经得到确认，正在进行解决，在正式修复之前您可以使用下面的任意一种方法解决这个问题：

1、在 app_main 最开始调用  `ESP_ERROR_CHECK(nvs_flash_erase());  ` 擦除 flash 中保存的 wifi 信息 

2、将server_if 的初始值设为0xFF ，在 7 个 `esp_blufi_send_wifi_` 开头的函数之前添加 if(server_if !=0xFF)，防止在蓝牙链接成功之前调用发送函数。

感谢您的反馈。

hscxg, 2019-11-15:
好的, 谢谢!

GYC-Espressif, 2019-12-06:
Hello, 该问题已经由 commit 34083c1463c3e272ae7473427b7c68b95c9d7ff8 解决。

closed 2019-12-06 by Alvin1Zhang