espressif/esp-idf #4317: LEDC Fade functions aren't working and blocking threads

alexanderturner, 2019-11-11:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROOM-32
- IDF version: v4.1-dev-815-ga45e99853 
- Build System:         Make
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:   macOS
- Power Supply:         USB

## Problem Description
ledc_set_fade_time_and_start, when called appears to block multiple threads (no errors, no crash or stack). Other ledc functions appear to work but calling fades stops _some_ threads.

### Expected Behavior
Execution ok, led fades

### Actual Behavior
Led doesn't fade, tried both timers and different channels. Some threads stop - for example, my rest server thread stops but others continue working but I can't make http requests (maybe tcp stack or wifi stack?). set_duty works fine. Function is being called with LEDC_FADE_NO_WAIT which also makes the blocking behaviour strange.

### Steps to reproduce

`ESP_ERROR_CHECK(ledc_set_fade_time_and_start(ledc_channel[0].speed_mode, ledc_channel[0].channel, 4000, fadetime, LEDC_FADE_NO_WAIT));`

### Code to reproduce this issue

```cpp
ledc_timer_config_t ledc_timer = {
    .duty_resolution = LEDC_TIMER_13_BIT, // resolution of PWM duty
    .freq_hz = 5000,                      // frequency of PWM signal
    .speed_mode = LEDC_HS_MODE,           // timer mode
    .timer_num = LEDC_HS_TIMER,            // timer index
    .clk_cfg = LEDC_AUTO_CLK,              // Auto select the source clock
};

ledc_channel_config_t ledc_channel[1] = {
  {
    .channel    = LEDC_HS_CH0_CHANNEL,
    .duty       = 0,
    .gpio_num   = LEDC_HS_CH0_GPIO,
    .speed_mode = LEDC_HS_MODE,
    .hpoint     = 0,
    .timer_sel  = LEDC_HS_TIMER
  },
};

  ledc_timer_config(&ledc_timer);

  ledc_channel_config(&ledc_channel[0]);
  ESP_ERROR_CHECK(ledc_fade_func_install(0));
  int fadetime = 10;
  ESP_ERROR_CHECK(ledc_set_fade_time_and_start(ledc_channel[0].speed_mode, 
ledc_channel[0].channel, 4000, fadetime, LEDC_FADE_NO_WAIT));
 
```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

nil

## Other items if possible

negativekelvin, 2019-11-12:
Does it happen if you use ledc_set_fade_with_time and ledc_fade_start instead of ledc_set_fade_time_and_start?

alexanderturner, 2019-11-12:
It doesn't appear to block but it doesn't cause the light to fade.

Ethan-CYQ, 2020-08-19:
Hi, @alexanderturner 

I had reproduced your problem but I found that the `fadetime` is 10, it means that the led will fade in 10 milliseconds. It is difficuit for us to observe it.  But from the logic analyzer, we can see the fade in 10 milliseconds.
[image]

You can try to set the `fadetime` longer(e.g. 1000ms) so that you can observe the led fade.

Alvin1Zhang, 2021-08-18:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more details, thanks.

closed 2021-08-18 by Alvin1Zhang